Koenkk/zigbee2mqtt #5236: Cannot pair Philips Hue 929001953101

derzico, 2020-12-09:
## What happened

I bought four brand new Hue GU10 Color and White lamps and they are not pairing.
I tried to pair it via bluetooth reset and touchlink, both did not work. At Bluetooth reset nothing was happening at all and via touchlink you only get the started and failed message as you can see in the log file.

## What did you expect to happen

Touchlink factory_reset send, the lamp should start blinking and then should pair.

## How to reproduce it (minimal and precise)

Well everything is at latest update stand. I did not have any devices in the zigbee2mqtt paired / connected.
I thought it has something to do with the zstack home 1.2 firmware version, so I flashed the CC2531 to Z-stack 3.0.x, but that was not the problem.

## Debug info
Log file: https://hastebin.com/raw/wuyidofuqi

Zigbee2MQTT version: 1.16.2
Adapter hardware: CC2531
Adapter firmware version: latest Z-Stack 3.0.x

Koenkk, 2020-12-09:
Make sure to check: https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

derzico, 2020-12-09:
> Make sure to check: https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

already tried everything

closed 2021-01-16 by github-actions[bot]