Koenkk/zigbee2mqtt #5237: IKEA Tradfri flash effect returns brightness to max instead of 1

Morphy99, 2020-12-09:
I've discovered a bug either in Home Assistant or Z2M, I think the latter. 
When switching on a Tradfri [LED1623G12](https://www.zigbee2mqtt.io/devices/LED1623G12.html#ikea-led1623g12) with brightness 1, if I send `effect: okay` for example it will flash and return brightness to what seems like max but still report brightness 1. If I turn on with brightness 2 or above, or just turn on and then set brightness 1 it will return to 1 after the effect.

It should turn on with 1 and return to 1 once the effect has finished.

I opened a thread on [HA forum](https://community.home-assistant.io/t/using-flash-and-brightness-mixed-results/252473) but I think it's Z2M related.

zigbee2mqtt version: 1.16.2-dev commit: 3f2eec8
coordinator type: zStack12 (CC2530)
coordinator revision: 20190619

Koenkk, 2020-12-09:
Looks like this is a bug in the IKEA firmware. Please check if updating them fixes it. Note that currently IKEA ota updates are broken and you need to use their test server. This option has just been added so first updated to the latest dev branch. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html) 

https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/information/ota_updates.md#using-the-ikea-tradfri-test-server

Morphy99, 2020-12-09:
They're on the latest stable firmware afaik, checking for updates doesn't show any.

Morphy99, 2020-12-10:
I've tested with other bulbs, Hue, Innr and even another Tradfri bulb LED1545G12 and they don't seem affected. I guess that confirms a bug with the firmware of this particular bulb. Any idea how to raise this with the Tradfri team? In the meantime I could edit the device page on the wiki with this info?

Koenkk, 2020-12-10:
@Morphy99 maybe you can reach out to them on reddit, to update the docs press the pencil icon here: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/LED1623G12.md

Morphy99, 2020-12-14:
[Added ](https://github.com/Koenkk/zigbee2mqtt.io/pull/503 to docs and opened a [thread](https://www.reddit.com/r/tradfri/comments/kbnrs1/ikea_tradfri_led1623g12_flash_effect_returns/) on reddit.

closed 2020-12-14 by Morphy99

Morphy99, 2021-03-03:
I've noticed there is a firmware update to these bulbs which has changed this behaviour. The light will now return to the correct brightness after the flash effect. BUT, turning on with brightness: 1 does nothing, the bulb does not power on even when the state is showing on.

Koenkk, 2021-03-03:
Can you try adding `meta: {turnsOffAtBrightness1: true},` (https://github.com/Koenkk/zigbee-herdsman-converters/blob/6c3a9beddbbd18d5994075f28e35544931ebc34c/devices.js#L2858) to the device.js definition?