Koenkk/zigbee2mqtt #5241: FLAIR Viyu E27 Smart LED (9/60w)

lvefferen, 2020-12-09:
## Information about the device + link
This is the W only variant of the bulb that was aded lately 
Hornbach FLAIR Viyu Smart GU10/E27 RGBW lamp #5117
https://www.hornbach.nl/shop/FLAIR-Viyu-Smart-LED-lamp-E27-9W-peervorm-instelbaar-wit/10011723/artikel.html
https://www.hornbach.nl/shop/FLAIR-Viyu-Smart-LED-lamp-GU10-5W-reflectorvorm-instelbaar-wit/10011724/artikel.html

I guess this will not be hard to add it is also available as a GU10 bulb.

## data/database.db entry of the device E27 version
{"id":31,"type":"Router","ieeeAddr":"0x00158d0003e3f551","nwkAddr":43348,"manufId":4709,"manufName":"HORNBACH Baumarkt AG","powerSource":"Mains (single phase)","modelId":"VIYU-A60-806-CCT-10011723","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"VIYU-A60-806-CCT-10011723","manufacturerName":"HORNBACH Baumarkt AG","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190127-27","swBuildId":"2.7"}}},"binds":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190127-27","swBuildId":"2.7","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1607537951514}

## data/database.db entry of the device GU10 version
{"id":29,"type":"Router","ieeeAddr":"0x00158d0003a5d32c","nwkAddr":1072,"manufId":4709,"manufName":"HORNBACH Baumarkt AG","powerSource":"Mains (single phase)","modelId":"VIYU-GU10-350-CCT-10011724","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"VIYU-GU10-350-CCT-10011724","manufacturerName":"HORNBACH Baumarkt AG","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190127-27","swBuildId":"2.7"}}},"binds":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190127-27","swBuildId":"2.7","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1607538555667}

Koenkk, 2020-12-09:
Does this only support brightness or also color temperature?

lvefferen, 2020-12-09:
> Does this only support brightness or also color temperature?

Both Brightness and color temperature (2700K - 6500K)

Koenkk, 2020-12-10:
Thanks, added.

Changes will be available in the latest dev branch tomorrow (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-10 by Koenkk

lvefferen, 2020-12-11:
Thanks, it seems te be working, how can we add a picture to the device?

Koenkk, 2020-12-11:
This will be done on the next z2m release (1 january)