Koenkk/zigbee2mqtt #5243: Zemismart Tubular motor error

rodrigorsdev, 2020-12-09:
None of the cover controls work, I have thi on log:

Zigbee2MQTT:error 2020-12-09 16:04:06: Publish 'set' 'state' to 'zemismart-persiana' failed: 'Error: Command 0x5c0272fffe8a5a2b/1 manuSpecificTuya.setData({"status":0,"transid":52,"dp":1,"datatype":4,"length_hi":0,"length_lo":1,"data":[2]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC no resources' (26))'
Zigbee2MQTT:info  2020-12-09 16:04:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'zemismart-persiana' failed: 'Error: Command 0x5c0272fffe8a5a2b/1 manuSpecificTuya.setData({\"status\":0,\"transid\":52,\"dp\":1,\"datatype\":4,\"length_hi\":0,\"length_lo\":1,\"data\":[2]}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC no resources' (26))'","meta":{"friendly_name":"zemismart-persiana"},"type":"zigbee_publish_error"}'

Thanks

Koenkk, 2020-12-10:
Please stick to the template (https://github.com/Koenkk/zigbee2mqtt/issues/new?assignees=&labels=problem&template=problem_report.md&title=) and provide the adapter and firmware version. Also provide the amount of devices in your network.

rodrigorsdev, 2020-12-10:
I´m sorry for that.

I use cc2531, zigbee2mqtt version 1.16.2.2.

I have 25 devices in my network.

Koenkk, 2020-12-10:
And what firmware version?

rodrigorsdev, 2020-12-10:
is this it?

> zigbee2mqtt@1.16.2 start /zigbee2mqtt-1.16.2
> node index.js
Zigbee2MQTT:info  2020-12-10 15:31:22: Logging to console and directory: '/share/zigbee2mqtt/log/2020-12-10.15-31-19' filename: log.txt
Zigbee2MQTT:info  2020-12-10 15:31:22: Starting Zigbee2MQTT version 1.16.2 (commit #1.16.2)
Zigbee2MQTT:info  2020-12-10 15:31:22: Starting zigbee-herdsman (0.13.37)

Koenkk, 2020-12-10:
Nope, check https://www.zigbee2mqtt.io/information/flashing_the_cc2531.html#how-to-check-the-installed-firmware-version

rodrigorsdev, 2020-12-10:
I found

info  2020-12-10 15:55:15: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'

Koenkk, 2020-12-11:
The MAC no resources issue you are getting means that the CC2531 is overloaded. Switching to the source routing firmware might improve that (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing) but if after this you are still getting these errors you have to update to a more powerful adapter. Usually somewhere between 20-30 devices people are getting issues with the CC2531 (https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html#supported-zigbee-adapter).

closed 2020-12-11 by Koenkk