Koenkk/zigbee2mqtt #5244: retain: true automatically added for new devices

Mr-Groch, 2020-12-09:
## What happened
I have noticed that all my new devices added this and previous month has `retain: true` in devices.yaml. I'm using zigbee2mqtt since about half year (from I think 1.14 version), first on CC2531 adapter, now on CC2652R zzh! adapter. Unfortunately I can't say exactly from when this flag is automatically added. I have switched to CC2652R last month and enabled reporting and availability features and after that all new added devices will have `retain: true` added to device configuration...

## What did you expect to happen
According to documentation, retain should be default set to force for new devices, not automatically added.

## How to reproduce it (minimal and precise)
Join new device to network for first time and see that in devices.yaml, newly added device will have `retain: true`

## Debug info
Zigbee2MQTT version: 1.16.2
Adapter hardware: CC2652R zzh!
Adapter firmware version: 20201113

Koenkk, 2020-12-10:
Can you provide the debug logging when pairing the device ending up with retain: true?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Mr-Groch, 2020-12-10:
Yes I can:

http://groch.cloud/misc/z2m_debug_log_20201210.zip

As you  can see, there is:

```
debug 2020-12-10 12:32:35: Received MQTT message on 'zigbee2mqtt/bridge/config/device_options' with data '{"friendly_name":"0x7cb03eaa0a00133c","options":{"retain":true}}'
info  2020-12-10 12:32:36: Changed device specific options of '0x7cb03eaa0a00133c' ({"retain":true})
```

Koenkk, 2020-12-10:
@Mr-Groch `zigbee2mqtt/bridge/config/device_options` indicates that Zigbee2MQTT receives a request to change it, if it only happens for this device this is probably a retained MQTT message (google how to clear it)

Mr-Groch, 2020-12-10:
No - this is happening for ALL NEW devices that I have bought this and last month... All newly added devices to my network willl have retain: true set...

Mr-Groch, 2020-12-10:
OK, I have digged more in my mqtt network, checked all devices and apps that can have something in common with z2m and found that zigbee2mqtt Node-Red addon adds `retain: true` device option on each pairing end mqtt message... Don't know why, but I've opened issue there: https://github.com/andreypopov/node-red-contrib-zigbee2mqtt/issues/25 . I'm closing this issue now, sorry for bothering you and thanks for advice to search on mqtt

closed 2020-12-10 by Mr-Groch

uncle-fed, 2022-01-20:
@Koenkk, I would like to re-open this because I believe I can confirm the original problem description here.
Except that I never had 'retain' set to true. On the contrary, it always has been globally set to 'false'.

I bought a brand new sensor here that was never previously seen in my Zigbee network.
My z2m configuration contains the one and only global setting:
```
device_options:
  retain: false
```
And yet, as soon as the sensor was paired and renamed to a friendly name, the retained MQTT message has appeared and remains in the corresponding device `.../availability` topic. The friendly name that I gave to this device has never previously existed in my configuration either. The device was renamed from the web UI, if this makes any difference.

Also, I would like to mention that after entirely removing all retained messages from mosquitto (by deleting actual mosquitto DB file from disk), I still see two more retained messages produced by z2m in the following topics:

```
zigbee2mqtt/bridge/state      # ('online' or 'offline')
zigbee2mqtt/bridge/devices    # (huge JSON that shows the entire Zigbee infrastructure)
```

I could not find it in the documentation whether these topics are supposed to have messages retained or not, no matter what. I suppose it makes certain sense to retain those at all times, but it would be nice to confirm that this is the intended behaviour and not a bug.

Now as to how I detected the rogue retained availability message, I am going to open another Github issue. The `.../availability` feature actually is nice to have but I would like to discuss the way the MQTT message is structured, because I probably would never realize that the availability message is there, and is retained if it did not break/affect something in my system.

uncle-fed, 2022-01-21:
Well, not at all really. :-))  When I wrote what I wrote, I meant something else, not how you got it in the end.

However, your reply forced me to test a few more things before I'd write crap. So actually it looks like a combination of both what you wrote and what I wrote originally, i.e. there is more than a single issue here.

So, once more, the "retain" is set globally to be disabled in my config file. Does this mean that I should never see any retained messages for my devices coming from z2m? I would imagine so. However, 

Pproblem 1: the availability topic appears and it turns out that it has a message with 'retain' flag set to true., even if my z2m config says otherwise..... Or are the availability messages always 'retained' no matter what the settings are in the configuration.yaml? 

Problem 2: I am NOT using the 'availability' feature, in general, it is not enabled in the config file, so I should NOT be seeing any device availability topics and/or messages, right? Wrong... The availability topic gets created as soon as the device is renamed. (And the message with the 'retain' flag set to true is published there even though the 'retain' is set to false globally).

Also, could you clarify about the 'retain' flag for the messages published to `bridge/state` and `bridge/devices` by z2m? It looks like they are always published with the retain flag set to true. Is this the intended behaviour? It is not documented, I believe...

Koenkk, 2022-01-21:
> Problem 1: the availability topic appears and it turns out that it has a message with 'retain' flag set to true, even if my z2m config says otherwise.... Or are the availability messages always 'retained' no matter what the settings are in the configuration.yaml? If so, then the documentation regarding the availability topics should probably mention this.

Availability messages are always retained indeed. This is to make sure other applications receive this message as soon as they connect to the MQTT server and even when z2m is down. I've updated the docs.

> Problem 2: I am NOT using the 'availability' feature, in general, it is not enabled in the config file, so I should NOT be seeing any device availability topics and/or messages, right? Wrong... The [...]

You should indeed not be seeing any messages if the availability feature is not enabled. This issue has already been fixed in the dev branch. You will need to manually clear any already published retained messages. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

While checking the code I found out that the old availability topic was not cleared upon rename, I fixed this now.

> Also, could you clarify about the 'retain' flag for the messages published to bridge/state and bridge/devices by z2m? It looks like they are always published with the retain flag set to true. Is this the intended behaviour? It is not documented, I believe...

Yes, reason is the same as why availability is retained. I've updated the docs.

uncle-fed, 2022-01-24:
Thanks a lot, @Koenkk for addressing my questions and fixing the issue!
I just have one more note regarding the availability feature, I'd appreciate your commentary:
https://github.com/Koenkk/zigbee2mqtt/issues/10802#issuecomment-1017897080