Koenkk/zigbee2mqtt #5247: Discovery of devices stopped to work if device is not close to coordinator

skumka, 2020-12-10:
What happened

Before the update of z2m last Friday when rediscovering the device it was possible to it joined and interviewed by any router which was close to the device.

From the last weekend after update to  1.16.2, it is possible to make a discovery only in close proximity to the coordinator.
Devices stay not discovered if the device is placed very close to any router in the netweok.

What did you expect to happen?

The device was not able to join the network close to routers. Several attempts were run.
I dismounted my TRV and moved it close to the coordinator. TRV was discovered immediately.
The same for Aqara and sonoff Thermometers as well as with motion sensors.

How to reproduce it (minimal and precise)

1-Delete any device in z2m and in HA.
2-Restart z2m
3-permit to join devices (all) - I have 7 routers including cc2531
4-reset the device to start discovery process - the device is not able to join the network
5-move device close to coordinator - the device is discovered immediately

Used cc2538+cc2592 with antenna additionally cc2531 is used as a router.
Stick and restarted zigbee2mqtt a few times over the last months.

Debug info

Zigbee2MQTT version: 1.16.2 
Adapter hardware:  cc2538+cc2592 with antenna plus cc2531 used as a router.

Koenkk, 2020-12-10:
I'm not sure if this is related to a Zigbee2MQTT update. Zigbee2MQTT has no influence in which devices accept others in the network. Does downgrading to 1.16.1 fix the issue?

If yes, please provide the sniff on both 1.16.1 and 1.16.2: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

skumka, 2020-12-10:
Downgrade to to 1.16.0 removes the issue.

skumka, 2020-12-10:
There is definitely change in behaviors of routers between those versions

Koenkk, 2020-12-10:
@skumka then sniff the traffic as I proposed in https://github.com/Koenkk/zigbee2mqtt/issues/5247#issuecomment-742705921

iamkarlson, 2020-12-22:
Same problem. More importantly, some devices just drop off the network and not be joining back on repairing. Sniffer shows empty log. No requests besides network operation.

iamkarlson, 2021-01-22:
The issue is not stale

skumka, 2021-01-22:
Problem exist

closed 2021-03-01 by github-actions[bot]