Koenkk/zigbee2mqtt #5252: OSRAM AC01353010G (motion sensor) get lost after Z2M restart, need to pull out battery

ufluido, 2020-12-10:
# Bug Report
## What happened
I use Z2M on Raspberry pi 4 and home assistant core.

When I reboot entire raspberry or even just restart the Z2M service, I need to pull out and reinsert battery on my 2 osram motion sensor to get it work again.
If I just restart home assistant core no problem.
If I don't reboot, all works ok for days.
I have also other sensors like Aqara motion, Aqara door sensor that is not affected by this behaviour
I have a one OSRAM+ dimmable zigbee bulbs.
I have no routers, just the CC2531 coordinator plugged on raspberry usb with an extension cable (1m).

I think something happened in the past few days, because before I had no issue like this.

Maybe something happened when I tried to use availability function to check if that osram lamp is still online,
now I removed availability line from config.yaml and I also had to bump my head on mqtt retain problem with message "offline" from battery sensor. To fix this I sent an "online" retained payload in each availability topic of the battery sensors. 
Maybe it can cause my issue?

## What did you expect to happen
Obliviously, I don't want to pull out battery every time :)

## How to reproduce it (minimal and precise)
-Start with raspberry on and all is working
-Restart only Z2M service
-Osram sensors (both) get lost
-Pull out battery, wait a second and reinsert
-Now all is ok

When I say "get lost" I mean that I cannot read any message received from zigbee2mqtt until I pull out and reinsert battery into sensors (I checked it into the log in debug mode).
When I reinsert battery, sensors send a message to zigbee2mqtt, something with "CommandStatusChangeNotification" (sorry for "something" but I'm not at home and I cannot reproduce now). after that message, communication is stable until next restart of the Z2M service.

## Debug Info
Zigbee2MQTT version: 1.16.2
Adapter hardware: CC2531
Adapter firmware version: revision: 20190608, type: zStack12

Koenkk, 2020-12-10:
- When the device is "lost", do you see nothing in the debug log when triggering occupancy on the sensor?
- If not, we need to sniff the traffic of this in the following order:
  - Trigger occupancy on the device
  - Reboot zigbee2mqtt
   - Trigger occupancy on the device (now it should not work)

https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

ufluido, 2020-12-10:
Thanks for your reply and also for your work,

- No, I cannot see nothing in debug log when Z2M restart, just that the past occupancy/contact value from each sensor 
  (included lost osram) is sent on MQTT topics and read from homeassistant.
  If I activate sensors, they respond correctly, just the two motion osram remain in the past state.
  If last state before Z2M restart was "occupancy: true" then homeassistant show them "on" and their state remain "on" always, 
  but it happens only on osram, all the other sensor start to work correctly.
  I can notice that they're not working because link quality index remains 0.
  In front of the debug log nothing happen when I activate osram sensors until I remove battery and then reinsert it.
  In this exactly moment osram sensor send that "CommandStatusChangeNotification" and I think is this command that make 
  system works but I don't know exactly what it means

- Unfortunately I do not have another CC2531 for traffic sniffing, but I just ordered new one for this purpose.
  I will come back to you with sniff information as soon as possible.

Is there any other thing that I can check until I receive CC2531?

What do you think about "availability" function that may caused it?
Can you please tell me what that "CommandStatusChangeNotification" message means?

Koenkk, 2020-12-11:
Since you don't get any events the availability function has nothing to do with it. CommandStatusChangeNotification is used by this sensor to say "hey I'm detecting occupancy". We need to wait until you have your sniffer.

closed 2021-01-18 by github-actions[bot]