Koenkk/zigbee2mqtt #5255: Unable to join new devices

valfr, 2020-12-10:
## What happened
Installed last version of Zigbee2mqtt - 1.16.2
Using a CC2531 flashed with "CC2531_DEFAULT_20190608"
CC2531 is visible on Raspberry
Starting Zigbee2mqtt seems all right
I'm trying pair a Xiaomi door sensor, not working
I'm watching logs and nothing 

I have flashed a second CC2531 with "CC2531_router_2018_09"
Pressing 5 seconds, entering in pairing mode but nothing happens

## What did you expect to happen
message under logs
message under mqtt 
pairing of device

## How to reproduce it (minimal and precise)
raspberry 4 + CC2531 
fresh install 

## Debug info
Zigbee2MQTT version: 1.16.2
Adapter hardware: CC2531
Adapter firmware version:20190608

When starting mosquitto_sub, i have : 
```
zigbee2mqtt/bridge/state offline
zigbee2mqtt/bridge/config {"commit":"0514204","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.16.2"}
```

mechamaches, 2021-01-03:
same problem here!
I've downgrade to 1.15 and nothing...

ivargj, 2021-01-20:
It seems many people have had problems with this sensor (and others from Xiaomi), myself included. It's become better in the later versions of zigbee2mqtt, so make sure to have it upgraded. 
What finally solved it for me was to move around a bit to other locations while trying to pair. This might be because of variations in zigbee network coverage, and noise from other things like wifi. 

Also I recently reduced the transmitting power of my wifi router in order to minimize zigbee interference, as well as making sure the channels didn't interfere (they didn't, so that wasn't the issue). (Check this out: https://support.metageek.com/hc/en-us/articles/203845040-ZigBee-and-WiFi-Coexistence) 
This might also have had an effect.

MattWestb, 2021-01-20:
Sleeping end devices is normally only using 1/10 of the radion power (for saving batteries) so they is needing good router in the near for getting the communication working OK and not leveling the network.
That is one of the 2 lare problem with the TI CC-253X then the radio and antenna its not so good but if having some good routers in the near the end devices it can working very well (and some CC-253X with external antennas is being good but not all).

closed 2021-02-27 by github-actions[bot]