Koenkk/zigbee2mqtt #5256: bitron-902010-28

spijkey87, 2020-12-10:
https://www.bol.com/nl/p/bitron-902010-28-dimmer-wit/9200000023790899/?Referrer=ADVNLGOO002063-G-57343768066-S-484451117271-9200000023790899&gclid=CjwKCAiAq8f-BRBtEiwAGr3DgS_vyu1Bmtqyxx-qWbecy9a0l58bJNbdfb156q1EKNwDvkJxyT9vRxoCG-AQAvD_BwE

I tried to add it true the how to but it would'nt work.

        zigbeeModel: ['902010/28'],
        model: 'AV2010/28',
        vendor: 'Bitron',
        description: 'Video wireless socket',
        fromZigbee: [fz.on_off],
        toZigbee: [tz.on_off],
        exposes: [e.switch()],
        meta: {configureKey: 2},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            await configureReporting.instantaneousDemand(endpoint);
..

spijkey87, 2020-12-28:
is there anyu indication as to when this will be added. Or if this will be added at all

Koenkk, 2020-12-28:
> I tried to add it true the how to but it would'nt work.

If it doesn't work I cannot add it (if it does I will add it of course). Please try to follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

closed 2021-02-05 by github-actions[bot]