Koenkk/zigbee2mqtt #5257: Saswell SEA802 TRV set schedule feature

r-glebov, 2020-12-10:
## Is your feature request related to a problem? Please describe

This feature request is about possibility to program schedule mode with time and temperatures user needs, but not the ones which are prebuilt, e.g. 16 degrees after 10pm each day. 

[image]

## Additional context

And one more thing, not sure if it is possible to change temperature calibration value, at least there is no word in [docs](https://www.zigbee2mqtt.io/devices/SEA802-Zigbee.html#saswell-sea802-zigbee) about that.

dev-2-4-h, 2020-12-28:
Hi 
I have an adapter that works, I have no knowledge how to submit the code to git but here is a zip-file containing the code.

[saswell.zip](https://github.com/Koenkk/zigbee2mqtt/files/5747814/saswell.zip)

Place the folder 'saswell' into the folder 'adapters' and edit '__init__.py' adding '`from adapters.saswell import saswell_adapters`' and '`**saswell_adapters,`'

Once done - pair the TRV - you will find several new devices.

I could not find a way to actually code the TRV in the way the Tuya app does.
This maybe possible for someone with more knowledge, however this works for me.

**Here are the new devices and their function:**
- SWITCH - **Device**, this is the on or off switch for the TRV 
- SWITCH - **Schedule**, this defines the mode or who controls the Schedule. 'Device' means AUTO mode, 'Domoticz' means MANUAL mode.
- SWITCH - **Away**, switches between normal operation and frost protection. 
- UTILITY - **Setpoint**, here you change the room temperature, use the 'Timers' as schedule.
- TEMPERATURE - **Temperature**, the actual (local) temperature of the area the TRV is installed.

In my home I use the 'Away' switch, allowing me to set the whole home into hibernation, the 'Setpoint' to control the temperature and the temperature sensor for information. I found no use for the other two switches. I simply set the TRV manually to the mode 'MANU'. 

Please share any feedback or ask questions about this solution directly, outside this Issue Tracker.
I created a new topic here: https://zigbee2mqtt.discourse.group/t/domoticz-saswell-sea802-trv/1926?u=screwitfix

dev-2-4-h, 2020-12-28:
Hi, 
as I understand the zigbee2mqtt project has no graphical user interface. There is one project ( https://github.com/yllibed/Zigbee2MqttAssistant ) that provides a gUI but even for them, this would be off-topic.

This Blog ( https://medium.com/@dzegarra/zigbee2mqtt-how-to-add-support-for-a-new-tuya-based-device-part-1-b20227251d46 ) may provide some additional information.

However, I use Domoticz to control the SEA802 TRV directly, bypassing the device own programming by changing the 'SETPOINT' with my timer (Schedule) function. More info here: https://zigbee2mqtt.discourse.group/t/domoticz-saswell-sea802-trv/1926

closed 2021-02-05 by github-actions[bot]