Koenkk/zigbee2mqtt #5263: BlitzWolf BW-SS7 is recognized as Zemismart ZM-L03E-Z

rubenOF, 2020-12-11:
## What happened
I just paired a new BlitzWolf BW-SS7 switch module. This is a 2-gang wall module but zigbee2mqtt recognized the module as "Zemismart ZM-L03E-Z", which is a 3-gang switch. I also received an error in the log:

````
ERROR: Failed to configure '0x847127fffe216458', attempt 2 (TypeError: Cannot read property 'bind' of undefined at bind (/app/node_modules/zigbee-herdsman-converters/devices.js:49:24) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices.js:11960:19) at async Configure.configure (/app/lib/extension/configure.js:131:13))
```

## What did you expect to happen
Recognize the module as a 2-gang module, not a 3-gang. and possible with it's own entry.

## How to reproduce it (minimal and precise)
Pair a BlitzWolf BW-SS7 module.

## Debug info
zigbee2mqtt version: 1.16.2 commit: 04c15f7
coordinator type: zStack30x
coordinator revision: 20190425
frontend version: 0.2.100

Module: https://www.blitzwolf.com/BlitzWolf-BW-SS7-ZigBee3.0-Smart-Light-Switch-Module-12Gang-with-ZigBee3.0,-12Gang,-APP-Control,-Voice-Control,-Timing-Switch-and-Devices-Sharing-p-647.html

## Pair log:
```
zigbee2mqtt_1  | Zigbee2MQTT:info  2020-12-11 12:43:13: Successfully interviewed '0x847127fffe216458', device has successfully been paired
zigbee2mqtt_1  | Zigbee2MQTT:info  2020-12-11 12:43:13: Device '0x847127fffe216458' is supported, identified as: Zemismart Smart light switch - 3 gang with neutral wire (ZM-L03E-Z)
zigbee2mqtt_1  | Zigbee2MQTT:info  2020-12-11 12:43:13: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart light switch - 3 gang with neutral wire","exposes":[{"endpoint":"left","features":[{"access":7,"description":"On/off state of the switch","endpoint":"left","name":"state","property":"state_left","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"center","features":[{"access":7,"description":"On/off state of the switch","endpoint":"center","name":"state","property":"state_center","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"right","features":[{"access":7,"description":"On/off state of the switch","endpoint":"right","name":"state","property":"state_right","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ZM-L03E-Z","vendor":"Zemismart"},"friendly_name":"0x847127fffe216458","ieee_address":"0x847127fffe216458","status":"successful","supported":true},"type":"device_interview"}'
```

rubenOF, 2020-12-11:
```json
{
	"id": 35,
	"type": "Router",
	"ieeeAddr": "0x847127fffe216458",
	"nwkAddr": 40766,
	"manufId": 4098,
	"manufName": "_TYZB01_digziiav",
	"powerSource": "Mains (single phase)",
	"modelId": "TS0003",
	"epList": [1, 2],
	"endpoints": {
		"1": {
			"profId": 260,
			"epId": 1,
			"devId": 256,
			"inClusterList": [0, 4, 5, 6, 6],
			"outClusterList": [25, 10],
			"clusters": {
				"genBasic": {
					"attributes": {
						"modelId": "TS0003",
						"manufacturerName": "_TYZB01_digziiav",
						"powerSource": 1,
						"zclVersion": 3,
						"appVersion": 65,
						"stackVersion": 0,
						"hwVersion": 1,
						"dateCode": ""
					}
				},
				"genOnOff": {
					"attributes": {
						"onOff": 0
					}
				}
			},
			"binds": [{
				"cluster": 6,
				"type": "endpoint",
				"deviceIeeeAddress": "0x00124b001936914c",
				"endpointID": 1
			}],
			"meta": {}
		},
		"2": {
			"profId": 260,
			"epId": 2,
			"devId": 256,
			"inClusterList": [4, 5, 6, 6],
			"outClusterList": [],
			"clusters": {},
			"binds": [{
				"cluster": 6,
				"type": "endpoint",
				"deviceIeeeAddress": "0x00124b001936914c",
				"endpointID": 1
			}],
			"meta": {}
		}
	},
	"appVersion": 65,
	"stackVersion": 0,
	"hwVersion": 1,
	"dateCode": "",
	"zclVersion": 3,
	"interviewCompleted": true,
	"meta": {},
	"lastSeen": 1607694295671
}
```

Koenkk, 2020-12-11:
Does it work when adding the following to devices.js

```js
{
    fingerprint: [{modelID: 'TS0003', manufacturerName: '_TYZB01_digziiav'}],
    model: 'BW-SS7',
    vendor: 'BlitzWolf',
    description: 'Zigbee 3.0 smart light switch module 2 gang',
    extend: generic.switch,
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2')],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    meta: {configureKey: 1, multiEndpoint: true},
    configure: async (device, coordinatorEndpoint) => {
        await bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
    },
},
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

g19i85g, 2020-12-19:
I have tested it and it works with my device.

Koenkk, 2020-12-20:
@g19i85g thanks, added it.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-20 by Koenkk