Koenkk/zigbee2mqtt #5264: Zigbee2mqtt very sluggish (with network and publish errors) after power outages

croc1234, 2020-12-11:
'Yesterday i had several power outages in my house, which caused my RPI with C2531 stick to restart.
After that zigbee2mqttt works very troublesome. There are huge delays in executing switch on/off commands. For instance when power on my Tradfri GU10 leds in a group it takes a minute before all eight leds in the domoticz scene/group are turned on. To be sure the stick is not damaged i flashed another one with the latest firmware, but the problem persists.
journalctl -u zigbee2mqtt.service -f provides me with the following information:

Dec 11 16:52:48 raspberrypi npm[460]: Zigbee2MQTT:info  2020-12-11 16:52:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":[],"type":"groups"}'
Dec 11 16:52:48 raspberrypi npm[460]: Zigbee2MQTT:error 2020-12-11 16:52:48: Failed to configure '0x00158d000205c489', attempt 1 (Error: Bind 0x00158d000205c489/1 genPowerCfg from '0x00124b001cce2346/1' failed (AREQ - ZDO - bindRsp after 10000ms)
Dec 11 16:52:48 raspberrypi npm[460]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
Dec 11 16:52:48 raspberrypi npm[460]:     at listOnTimeout (internal/timers.js:554:17)
Dec 11 16:52:48 raspberrypi npm[460]:     at processTimers (internal/timers.js:497:7))
Dec 11 16:54:11 raspberrypi npm[460]: Zigbee2MQTT:info  2020-12-11 16:54:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00158d00031e1b6c"},"type":"device_announced"}'
Dec 11 16:54:47 raspberrypi npm[460]: Zigbee2MQTT:error 2020-12-11 16:54:47: Publish 'set' 'brightness' to '0xec1bbdfffe02477d' failed: 'Error: Command 0xec1bbdfffe02477d/1 genLevelCtrl.moveToLevelWithOnOff({"level":127,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'No network route' (205))'
Dec 11 16:54:47 raspberrypi npm[460]: Zigbee2MQTT:info  2020-12-11 16:54:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to '0xec1bbdfffe02477d' failed: 'Error: Command 0xec1bbdfffe02477d/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":127,\"transtime\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"0xec1bbdfffe02477d"},"type":"zigbee_publish_error"}'
Dec 11 16:54:47 raspberrypi npm[460]: Zigbee2MQTT:error 2020-12-11 16:54:47: Publish 'set' 'brightness' to '0xec1bbdfffe024bd6' failed: 'Error: Command 0xec1bbdfffe024bd6/1 genLevelCtrl.moveToLevelWithOnOff({"level":127,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'No network route' (205))'
Dec 11 16:54:47 raspberrypi npm[460]: Zigbee2MQTT:info  2020-12-11 16:54:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to '0xec1bbdfffe024bd6' failed: 'Error: Command 0xec1bbdfffe024bd6/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":127,\"transtime\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"0xec1bbdfffe024bd6"},"type":"zigbee_publish_error"}'

Hope somebody can help me to solve this problem.

jonbrunt123, 2020-12-27:
I'm not sure if this is relevant to you, however I had the sam issue, it wasn't actually Zigbee, it was my WiFi Mesh rebooting and selecting channel 1 (overlapping with the default Zigbee channel) that caused the issue!

croc1234, 2020-12-28:
Thanks for your reply jonbrunt123. I restarted the wifi and the problem is gone, so it seems to be a channel conflict indeed.

closed 2021-02-05 by github-actions[bot]