Koenkk/zigbee2mqtt #5268: multiple scene_add for same ID merges scene state?

sjorge, 2020-12-11:
## What happened
Calling scene_add multiple times with the same ID seems to not clear the previous payload.

- add scene 42 with state=on, brightness=128, color=#ff0000
- recall scene 42
- <some time later, bulb was changed to blue and 64 brightness without using scenes>
- add scene 42 with state=on, brightness=254
- recall scene 42

Bulbs turns to 100% brightness, but color changes to red again.

## What did you expect to happen
Scene add should wipe the previous scene state, or it should be documented that the state does not get cleaned up.

My current work around is issuing a scene_remove before the scene_add, it usually works fine except few odd cases where I think the add gets executed before the remove because the scene recall sometimes fails if I do remove,add,recall, having zhc do the scene remove and wait for the result before doing the new scene add might be better? Although it looks like scene_remove does not give a response back to the coordinator if it was successful or not?

## How to reproduce it (minimal and precise)
I noticed this issue with Innr  RGB bulbs, I was not able to reproduce with Trådfri WS bulbs... so it might be vendor specific?

But doing a scene_add with for example brightness + color, the recalling, changing the color and brightnes without using scenes and then doing another scene_add without color or brightness, followed by a recall seems to trigger it about 50/50 for me.

There might be more at play here though, I looked at the traffic and the commands go out in the right order. But perhaps the bulbs themselves have not yet complete updated the scene state before it is asked to recall... so many things at play :(

## Debug info
Zigbee2MQTT version: 1.16.2-dev commit: b6c4c2d
Adapter hardware: CC2531
Adapter firmware version: zStack12 20201110

Koenkk, 2020-12-11:
> do the scene remove and wait for the result before doing the new scene add might be better

This sounds like a good solution to me. But please check what happens when the scene remove is done when a scene with that ID was never added before. When using `await` the call will not return before the response is received (e.g. https://github.com/Koenkk/zigbee-herdsman-converters/blob/e220ffb2ab40ab07a6b6058ad3447b218e1bb0b1/converters/toZigbee.js#L3809). This works because a response ID is specified here: https://github.com/Koenkk/zigbee-herdsman/blob/060378e522f3572c4d01661038d1c80bc0a6e8be/src/zcl/definition/cluster.ts#L275 which refers to https://github.com/Koenkk/zigbee-herdsman/blob/060378e522f3572c4d01661038d1c80bc0a6e8be/src/zcl/definition/cluster.ts#L381

sjorge, 2020-12-13:
With the current code I don't see an error removing a scene twice (it never existed, as I just paired the bulb.

```
Zigbee2MQTT:debug 2020-12-13 09:30:57: Received MQTT message on 'zigbee2mqtt/0xec1bbdfffeaf89de/set' with data '{"scene_remove": 22}'
Zigbee2MQTT:debug 2020-12-13 09:30:57: Publishing 'set' 'scene_remove' to '0xec1bbdfffeaf89de'
Zigbee2MQTT:debug 2020-12-13 09:30:59: Received MQTT message on 'zigbee2mqtt/0xec1bbdfffeaf89de/set' with data '{"scene_remove": 22}'
Zigbee2MQTT:debug 2020-12-13 09:30:59: Publishing 'set' 'scene_remove' to '0xec1bbdfffeaf89de'
```

I will start with updating scene_remove to capture the response before updating the meta state, which should have been like this from the start. And try to work up from there.

sjorge, 2020-12-13:
I've also done more testing... 

```bash
DEV="0xec1bbdfffeaf89de"
CP="color_temp"
CPV=("250" "454")
#CP="color"
#CPV=("{\"x\":0.3818,\"y\":0.3797}" "{\"x\":0.5018,\"y\":0.4153}")

set -x
mosquitto_pub -t zigbee2mqtt/${DEV}/set  -m "{\"scene_remove\": 22}"
sleep 3
mosquitto_pub -t zigbee2mqtt/${DEV}/set  -m "{\"scene_add\": {\"ID\": 22, \"state\": \"ON\", \"${CP}\": ${CPV[1]}, \"brightness\": 254}}"
sleep 3
mosquitto_pub -t zigbee2mqtt/${DEV}/set  -m "{\"scene_recall\": 22}"
sleep 3
mosquitto_pub -t zigbee2mqtt/${DEV}/set  -m "{\"${CP}\": ${CPV[0]}}"
sleep 3
mosquitto_pub -t zigbee2mqtt/${DEV}/set  -m "{\"scene_add\": {\"ID\": 22, \"state\": \"ON\", \"brightness\": 32}}"
sleep 3
mosquitto_pub -t zigbee2mqtt/${DEV}/set  -m "{\"scene_recall\": 22}"
```

This is a simple reproducible case (at least on a Trådfri GU10)

Expected result: bulb on at a brightness of ~12% in cool white color (250)
Actual result: bulb on at a brightness of ~12% in warm white color (454)

sjorge, 2020-12-13:
I spend some more time reading the ZCL for scene add...

```
4136 The format of each extension field set is a 16 bit field carrying the cluster ID, followed by an 8 bit length field and the
4137 set of scene extension fields specified in the relevant cluster. The length field holds the length in octets of that extension
4136 The format of each extension field set is a 16 bit field carrying the cluster ID, followed by an 8 bit length field and the
4137 set of scene extension fields specified in the relevant cluster. The length field holds the length in octets of that extension
4138 field set.
4139 Extension field sets =
4140 {{clusterId 1, length 1, {extension field set 1}}, {clusterId 2, length 2, {extension field set 2}} ...}.
4141 The attributes included in the extension field set for each cluster are defined in the specification for that cluster in this
4142 document (the Cluster Library). The field set consists of values for these attributes concatenated together, in the order
4143 given in the cluster specification, with no attribute identifiers or data type indicators.
4144 For forward compatibility, reception of this command SHALL allow for the possible future addition of other attributes
4145 to the trailing ends of the lists given in the cluster specifications (by ignoring them). Similarly, it SHALL allow for
4146 one or more attributes to be omitted from the trailing ends of these lists (see 3.7.2.4.7.2).
4147 It is not mandatory for a field set to be included in the command for every cluster on that endpoint
```

@Koenkk I read this a few times and I'm still confused, but does 4141 indicate that we should always send all possible cluster we know of in each scene_add? 

Further down it also clearly describes the behavior we saw with ColorTemperatureMires so thats good too.

Perhaps instead of calling scene_remove, we should always send all know clusters in the extensionsFieldSets, meaning the 2nd scene_add call above is missing cluster id 768 currently. Which probably means the bulb writes fields we give it over the old ones and leaving cluster id  768 untouched... which would explain the behavior. If we send it but with all the fields set to null, it should replace it, making the extra call to scene_remove unneeded... (less traffic == good)

But I find this whole section to be somewhat confusion.

So i think either the fix is:
- do a remove call at the beginning of the add call (extra traffic + call that is not really mentioned in ZCL)
- just call add but specify all know fields, if not provided by the user assume they are zeroed but do send them.

Based on my current reading of the ZCL, the 2nd one seems correct. (I've tested neither so far though)

sjorge, 2020-12-13:
I've got the checking of the response working for `scene_remove` like what is done in `scene_add`, however I noticed the response is undefined when calling `scene_remove` (or `scene_add` for that matter) on a group, is there a way to get these too? I guess we need to wait for the response of each group member? But how do I do that?

I couldn't find any existing examples that do this (or not one that was obvious anyway), is that currently even possible?

sjorge, 2020-12-13:
I've made a PR of what I have, there is still improvement possible w.r.t. groups, see above.

closed 2020-12-14 by sjorge