Koenkk/zigbee2mqtt #5273: Repairing of devices not working

SpatialJ, 2020-12-12:
Hello,

i have a raspberry pi 4 with an zig-a-zig-ah! CC2652 Stick and running zigbee2mqtt on IOTstack.
Before i used a CC2531 stick and had around 6 aqara temperature/humidity sensors paired.
After the update to the CC2652  i changed the pan_id as documented ( i randomly used 6754).
Since my existing sensors did not work  afterwards i changed the panid back to the default in my configuration.yml.

New (never paired, fresh out of the box) devices are able to join with the new coordinator, the existing ones just dont.
I sniffed the traffic using the old CC2532 stick and i see a lot of beaocn requests when i press the aqara sensors button to repair but nothing else:
[image]

I tried repairing the device standing next to the raspberry so distance should not be the problem.

What aslo strikes me odd is that zigbee2mqtt  still logs 6754 as the panid although  i changed it in the config, deleted the database and reflashed the coordinator.

What else can i try/troubleshoot to get my sensors to pair again?

Thanks,
Johannes

thrasheri, 2020-12-12:
I have the same exact issue. Moved from CC2351 to CC2652. I was able to pair Ikea bulbs, but the xiaomi temperature sensors show nothing in the logs when resetting. I tried to change battery to one of the sensor, but it didnt help.

skumka, 2020-12-16:
I have very similar issue described here: https://github.com/Koenkk/zigbee2mqtt/issues/5247

SpatialJ, 2020-12-16:
@skumka thanks for your reply. 
I saw your issue and downgraded to 1.16.0  but no luck :-/

Gollam, 2020-12-24:
I had exactly the same issue, and what I did was to press the sensor button for 5 seconds, then when the 3 long led blinks came, I started quickly and repeatedly pressing the button, 5-6 times, then it reconnected. It worked with both my sensors.

thrasheri, 2020-12-24:
I also got mine working by changing pan_id and the batteries for the sensors. Some sensors needed multiple tries to pair. Also some Ikea zigbee devices required zigbee2mqtt restart to pair.

SpatialJ, 2020-12-25:
@thrasheri  what value did you use for the new pan_id?

SpatialJ, 2020-12-25:
> I had exactly the same issue, and what I did was to press the sensor button for 5 seconds, then when the 3 long led blinks came, I started quickly and repeatedly pressing the button, 5-6 times, then it reconnected. It worked with both my sensors.

no luck for me so far unfortunately :-/

kpetremann, 2020-12-25:
I had a similar issue: after changing the coordinator, I was unable to re-pair the Xiaomi sensor. The coordinator received some messages for sure, but pairing was not done.

What worked for me: replacing the battery by a new one (the previous was at 35% according the sensor). As soon as I replaced the battery, it paired instantaneously.

thrasheri, 2020-12-27:
> @thrasheri what value did you use for the new pan_id?

I used 6755

SpatialJ, 2021-01-27:
is still have no solution to this problem. It seems all my devices are just unusable now :-/

Gollam, 2021-01-27:
What solved it for me in the end was an USB extender. Initially I plugged the zzh stick directly into the pi, and the pi's wifi causes so much interference, that the stick is unable to communicate with barely any device. I plugged in the stick with a ~1m USB extender, and it was truly magical, every device joined when I pressed their buttons.

SpatialJ, 2021-02-14:
Tried it with an extender, no luck :-/
what i dont understand is why the log shows 6754 as the panid:

> Zigbee2MQTT:info  2021-02-07 14:01:13: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"9ed84e1","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20201026,"transportrev":2},"type":"zStack3x0"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.16.0"}'

i did not set the panid in the configuration. Is there a way to "reset" it somehow?

closed 2021-03-24 by github-actions[bot]