Koenkk/zigbee2mqtt #5284: Add Support for Lidl ZigBee Plugs (UK)

bedfelllow, 2020-12-13:
https://www.lidl.co.uk/our-products/smart-home
https://i.ibb.co/HqgdC3R/IMG-20201213-095206.jpg
..
WARNING:  `Device 
'0x842e14fffecd080c' with Zigbee model 'TS011F' and manufacturer name 
'_TZ3000_00mk2xzy' is NOT supported, please follow 
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html`
..

Koenkk, 2020-12-13:
Already supported in the dev branch. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-13 by Koenkk

danstreeter, 2020-12-13:
Thanks @Koenkk - Is there a commit I can track for when it merges so I can shift back onto stable?

Koenkk, 2020-12-14:
You can switch back to stable when a new version of z2m gets released (1 january)

emarcin, 2021-01-04:
any news about the support?

Koenkk, 2021-01-04:
Supported in Zigbee2MQTT 1.17.0

bedfelllow, 2021-01-04:
> any news about the support?

An update to HomeAssistant plugin has been released. I installed it yesterday and it works fine with these now.

@Koenkk, thanks for the update.

emarcin, 2021-01-04:
Now it works! Thanks a lot!

ximon, 2021-01-13:
Does this device report its state when the button on the device is pressed? I.e are we able to get updates when the socket is turned off via its built in button?

bedfelllow, 2021-01-13:
> Does this device report its state when the button on the device is pressed? I.e are we able to get updates when the socket is turned off via its built in button?

Zigbee2mqtt reports the state correcting when the button is pressed.

ximon, 2021-01-26:
> > Does this device report its state when the button on the device is pressed? I.e are we able to get updates when the socket is turned off via its built in button?
> 
> Zigbee2mqtt reports the state correcting when the button is pressed.

Is there something I need to change / do for this to work? I dont get state updates if I change the state of the socket via its built in button - I dont see any traffic in the logs when pressing the button after enabling logging.

bedfelllow, 2021-01-26:
> Is there something I need to change / do for this to work? I dont get state updates if I change the state of the socket via its built in button - I dont see any traffic in the logs when pressing the [...]

I never had to change any configuration, it just worked as default.

If you open the device in Zigbee2MQTT web UI, in the 'Exposes' tab do you see the switch toggle when you press the socket button?

[image]
[image]

ximon, 2021-01-26:
It doesn't toggle automatically, but it does when i click the refresh button next to 'state'
I have four of these devices and its the same for all four.

RobvanSuilen, 2022-03-11:
I have the same problem, using a raspberry pi, homeassistant, mosquitto and zigbee2mqtt