Koenkk/zigbee2mqtt #5285: OSRAM AC01353010G (Smart+ Motion Sensor) - Temperature update fail

Jogobo, 2020-12-13:
# Bug Report
## What happened
After switching to zigbee2mqtt V 1.16.2 the temperature send by the motion sensor stays static. It stays the same value once found after pairing the device. Removing the device and re-pairing updates the temperature once.

## What did you expect to happen
Current temperature should be updated every time the device communicates with zigbee2mqtt.

## Debug Info
Zigbee2MQTT version: 1.16.2
Adapter hardware: CC2652RB
Adapter firmware version: zStack3x0 20201026

database entry of a device:
{"id":41,"type":"EndDevice","ieeeAddr":"0x000d6f000e059937","nwkAddr":51658,"man
ufId":4174,"manufName":"CentraLite","powerSource":"Battery","modelId":"Motion Se
nsor-A","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inC
lusterList":[0,1,3,32,1026,1280,2821],"outClusterList":[25],"clusters":{"genBasi
c":{"attributes":{"modelId":"Motion Sensor-A","manufacturerName":"CentraLite","p
owerSource":3,"zclVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"��������
��������"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0021cc3cbd","zoneSt
ate":1}},"msTemperatureMeasurement":{"attributes":{"measuredValue":1826}}},"bind
s":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc3cbd","
endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0021c
c3cbd","endpointID":1}],"meta":{}},"2":{"profId":260,"epId":2,"devId":263,"inClu
sterList":[0,1,3,1030,2821],"outClusterList":[3],"clusters":{"genBasic":{"attrib
utes":{}}},"binds":[],"meta":{}}},"stackVersion":2,"hwVersion":1,"dateCode":"���
�������������","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},
"lastSeen":1607859586936}

gstrotmann, 2020-12-22:
I have the same problem. 
I had switched to an Electrolama zig-a-zig-ah! (zzh!) Adapter. Additional I upgraded Zigbee2MQTT to version 1.16.2 an re-paired every item.
Thince than, I do not get periodically 'health' messages from the OSRAM Smart+ Motion Sensors anymore and also no temperature value.

gstrotmann, 2021-01-22:
After removing the devices complete from the zigbee network (with zigbee2mqtt/bridge/request/device/remove) and re-pairing them again, I get from every of my 4 devices temperatur values.
Also missed occupancy values "false" messages from 1 device are there now.

closed 2021-03-03 by github-actions[bot]