Koenkk/zigbee2mqtt #5288: MQTT LWT goes Offline when Z2M Still running

marc-gist, 2020-12-13:
## What happened
MQTT LWT offline, but Z2M is running fine. causes issues with HomeAssitant or other automations that rely on accurate LWT.

## What did you expect to happen
Z2M to keep online payload alive (periodic setting online status?)

## How to reproduce it (minimal and precise)
I can't reproduce, seems random. run for a month, or a few days it can occur.

## Debug info
Adapter hardware:  CC1352P-2
Version of Zigbee2Mqtt: 1.16.2
Coordinator version: 20201026

marc-gist, 2020-12-13:
Update: I believe the issue is when my docker machine running z2m losses network connection (switch firmware is updated), it connects back to mqtt fine, but for some reason doesn't issue a LWT again (perhaps the mqtt client didn't realize it was not connected for long enough for the mqtt server to issue the offline payload).

Perhaps updating LWT every few minutes could avoid this issue?

Koenkk, 2020-12-13:
Can you try to catch a debug log when this happens?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

closed 2021-01-13 by Koenkk