Koenkk/zigbee2mqtt #5291: Addon will not start with conbee II

eric-zimmer, 2020-12-13:
## What happened
Zigbee2mqtt HA addon will not start, Zigbee stick is the dresden electronic conbee II

## What did you expect to happen
the addon to start

## How to reproduce it (minimal and precise)
use conbee II stick and try to start HA zigbee1mqtt add on

## Debug info
Zigbee2MQTT version: Zigbee2mqtt Edge, Current version: test (installed yesteray, so I think it must be the latest version)
Adapter hardware: Conbee II
Adapter firmware version: latest (check on ohoscon for windows today)

Configuration:
https://pastebin.com/JhHmSntZ

Log:
https://pastebin.com/f7rQLS13

I tried a sport "null", cannot save, also  - /dev/ttyAMA0 and - /dev/ttyACM0 does not work.

Any idea what else I can change? Also the non edge version of the add-on does not workt but as I read in the guthub of add-on it is more likely a problem with zugbee2mqtt itself.

Koenkk, 2020-12-13:
Can you provide the herdsman debug logging so @chrishae can check this?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

eric-zimmer, 2020-12-13:
https://pastebin.com/cKEfFHjZ

Hope this is the right one.

closed 2021-01-20 by github-actions[bot]