Koenkk/zigbee2mqtt #5295: Müller Licht 404037 flickers irregularily - looks similiar to effect "channel change"

XRyu, 2020-12-13:
## What happened
I own several Müller Licht Lights - most of them are of the Warm+Cold-white Retro LED Bulbs 404037 - they irregularily flicker - the effect looks veeeery similiar to the integrated effect "channel change". This happens in every state, even when the lights state is off. In the logs i dont see what this might be. I just gathered the log after it happened - attached below. This happened to the light "Licht - Wohnzimmer Essbereich - Esstisch - Filament Q" = 0x00158d00013b1dce

## What did you expect to happen
no flickering :)
## How to reproduce it (minimal and precise)
Use Müller Licht - 404037
## Debug info
Zigbee2MQTT version: 1.16.2 commit: 04c15f7
Adapter hardware: CC1352P-2
Adapter firmware version: 20201113

´´´
Zigbee2MQTT:info  2020-12-13 23:44:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to 'Licht - Flur' failed: 'Error: Command 0x000b57fffe98737e/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":13,\"transtime\":10}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 9746 - 1 - 202 - 8 - 11 after 10000ms)'","meta":{"friendly_name":"Licht - Flur"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:debug 2020-12-13 23:44:46: Publishing 'set' 'transition' to 'Licht - Flur'
Zigbee2MQTT:debug 2020-12-13 23:44:53: Received Zigbee message from 'Licht - Arbeitszimmer Thomas - Schreibtisch - Filament O', type 'attributeReport', cluster 'lightingColorCtrl', data '{"currentY":21396}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-12-13 23:44:53: MQTT publish: topic 'zigbee2mqtt/Licht - Arbeitszimmer Thomas - Schreibtisch - Filament O', payload '{"brightness":24,"color":{"x":0.3151,"y":0.3265},"color_temp":153,"device":{"applicationVersion":1,"dateCode":"20190601","friendlyName":"Licht - Arbeitszimmer Thomas - Schreibtisch - Filament O","hardwareVersion":1,"ieeeAddr":"0x00158d00013dcc45","manufacturerID":4137,"manufacturerName":"MLI","model":"404037","networkAddress":16243,"powerSource":"Mains (single phase)","softwareBuildID":"TDU18650B01","stackVersion":1,"type":"Router","zclVersion":2},"elapsed":32031,"last_seen":1607899493128,"linkquality":60,"state":"OFF"}'
Zigbee2MQTT:debug 2020-12-13 23:44:53: Received MQTT message on 'zigbee2mqtt/Licht - Flur/set' with data '{"state": "OFF", "transition": 1.0}'
Zigbee2MQTT:debug 2020-12-13 23:44:53: Publishing 'set' 'transition' to 'Licht - Flur'
Zigbee2MQTT:debug 2020-12-13 23:44:53: Publishing 'set' 'state' to 'Licht - Flur'
Zigbee2MQTT:debug 2020-12-13 23:44:56: Received Zigbee message from 'Licht - Arbeitszimmer Thomas - Decke - Filament N', type 'attributeReport', cluster 'lightingColorCtrl', data '{"currentY":21396}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-12-13 23:44:56: MQTT publish: topic 'zigbee2mqtt/Licht - Arbeitszimmer Thomas - Decke - Filament N', payload '{"brightness":24,"color":{"x":0.3151,"y":0.3265},"color_temp":153,"device":{"applicationVersion":1,"dateCode":"20190601","friendlyName":"Licht - Arbeitszimmer Thomas - Decke - Filament N","hardwareVersion":1,"ieeeAddr":"0x00158d00013b1ecc","manufacturerID":4137,"manufacturerName":"MLI","model":"404037","networkAddress":41365,"powerSource":"Mains (single phase)","softwareBuildID":"TDU18650B01","stackVersion":1,"type":"Router","zclVersion":2},"elapsed":32257,"last_seen":1607899496035,"linkquality":57,"state":"ON"}'
Zigbee2MQTT:debug 2020-12-13 23:44:58: Received Zigbee message from 'Licht - Wohnzimmer Essbereich - Esstisch - Filament J', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":26}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-12-13 23:44:58: MQTT publish: topic 'zigbee2mqtt/Licht - Wohnzimmer Essbereich - Esstisch - Filament J', payload '{"brightness":26,"color":{"x":0.5267,"y":0.4173},"color_temp":500,"device":{"applicationVersion":1,"dateCode":"20190601","friendlyName":"Licht - Wohnzimmer Essbereich - Esstisch - Filament J","hardwareVersion":1,"ieeeAddr":"0x00158d00013dc5c8","manufacturerID":4137,"manufacturerName":"MLI","model":"404037","networkAddress":23660,"powerSource":"Mains (single phase)","softwareBuildID":"TDU18650B01","stackVersion":1,"type":"Router","zclVersion":2},"elapsed":13253,"last_seen":1607899498156,"linkquality":42,"state":"ON"}'
Zigbee2MQTT:debug 2020-12-13 23:45:01: Received Zigbee message from 'Licht - Wohnzimmer Essbereich - Esstisch - Filament Q', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-12-13 23:45:01: MQTT publish: topic 'zigbee2mqtt/Licht - Wohnzimmer Essbereich - Esstisch - Filament Q', payload '{"brightness":26,"color":{"x":0.5267,"y":0.4173},"color_temp":500,"device":{"applicationVersion":1,"dateCode":"20190601","friendlyName":"Licht - Wohnzimmer Essbereich - Esstisch - Filament Q","hardwareVersion":1,"ieeeAddr":"0x00158d00013b1dce","manufacturerID":4137,"manufacturerName":"MLI","model":"404037","networkAddress":6004,"powerSource":"Mains (single phase)","softwareBuildID":"TDU18650B01","stackVersion":1,"type":"Router","zclVersion":2},"elapsed":98638,"last_seen":1607899501889,"linkquality":42,"state":"ON"}'
Zigbee2MQTT:debug 2020-12-13 23:45:06: Received Zigbee message from 'Licht - Wohnzimmer Essbereich - Esstisch - Filament Q', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":500}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-12-13 23:45:06: MQTT publish: topic 'zigbee2mqtt/Licht - Wohnzimmer Essbereich - Esstisch - Filament Q', payload '{"brightness":26,"color":{"x":0.5267,"y":0.4173},"color_temp":500,"device":{"applicationVersion":1,"dateCode":"20190601","friendlyName":"Licht - Wohnzimmer Essbereich - Esstisch - Filament Q","hardwareVersion":1,"ieeeAddr":"0x00158d00013b1dce","manufacturerID":4137,"manufacturerName":"MLI","model":"404037","networkAddress":6004,"powerSource":"Mains (single phase)","softwareBuildID":"TDU18650B01","stackVersion":1,"type":"Router","zclVersion":2},"elapsed":5015,"last_seen":1607899506904,"linkquality":48,"state":"ON"}'
Zigbee2MQTT:debug 2020-12-13 23:45:09: Received Zigbee message from 'Licht - Outdoor Lightstrip A', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":655368546,"imageType":1793,"manufacturerCode":4454}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-12-13 23:45:14: Received Zigbee message from 'Licht - Filament I', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":500}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-12-13 23:45:14: MQTT publish: topic 'zigbee2mqtt/Licht - Filament I', payload '{"brightness":26,"color":{"x":0.5267,"y":0.4174},"color_temp":500,"device":{"applicationVersion":1,"dateCode":"20190601","friendlyName":"Licht - Filament I","hardwareVersion":1,"ieeeAddr":"0x00158d00013dbf4b","manufacturerID":4137,"manufacturerName":"MLI","model":"404037","networkAddress":43697,"powerSource":"Mains (single phase)","softwareBuildID":"TDU18650B01","stackVersion":1,"type":"Router","zclVersion":2},"elapsed":142845,"last_seen":1607899514534,"linkquality":51,"state":"ON"}'
Zigbee2MQTT:debug 2020-12-13 23:45:14: Received Zigbee message from 'Licht - Wohnzimmer Essbereich - Esstisch - Filament Q', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":26}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-12-13 23:45:14: MQTT publish: topic 'zigbee2mqtt/Licht - Wohnzimmer Essbereich - Esstisch - Filament Q', payload '{"brightness":26,"color":{"x":0.5267,"y":0.4173},"color_temp":500,"device":{"applicationVersion":1,"dateCode":"20190601","friendlyName":"Licht - Wohnzimmer Essbereich - Esstisch - Filament Q","hardwareVersion":1,"ieeeAddr":"0x00158d00013b1dce","manufacturerID":4137,"manufacturerName":"MLI","model":"404037","networkAddress":6004,"powerSource":"Mains (single phase)","softwareBuildID":"TDU18650B01","stackVersion":1,"type":"Router","zclVersion":2},"elapsed":8090,"last_seen":1607899514995,"linkquality":39,"state":"ON"}'
Zigbee2MQTT:error 2020-12-13 23:45:16: Publish 'set' 'state' to 'Licht - Flur' failed: 'Error: Command 0x000b57fffe98737e/1 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":10}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reserve [... cut ...]
Zigbee2MQTT:debug 2020-12-13 23:46:05: Received Zigbee message from 'Licht - Terrasse - Filament F', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16782596,"imageType":274,"manufacturerCode":4107}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2020-12-13 23:46:15: Received Zigbee message from 'Licht - Terrasse - Filament F', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16782596,"imageType":274,"manufacturerCode":4107}' from endpoint 11 with groupID 0

thunder-62, 2020-12-16:
Hi, I had a similar behaviour when my Paulmann Amaris LED Panel was only paired with the coordinator but not bound to a Switch. Might be an idea to bind a wwitch directly to your device....

XRyu, 2020-12-16:
Thank you for the tip @thunder-62  :)
But currently i dont want to bind any switches directly (primary using the IKEA 1810 with HA-appdeamon-configs).
But when Paulman has this too, there might be an issue in some kind of Communication with Z2M :)

XRyu, 2021-01-19:
still happening

Koenkk, 2021-01-22:
I don't see much activity from this device from the log, don't think it comes from Zigbee2MQTT itself. Did it crash at this time?

```js

Zigbee2MQTT:debug 2021-01-21 23:24:23: Received Zigbee message from 'Licht - Schlafzimmer - Decke - Filament K', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0

2021-01-21T22:24:23.816Z zigbee-herdsman:controller:endpoint DefaultResponse 0x00158d00013d3a1d/1 6(10, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})

2021-01-21T22:24:23.816Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x00158d00013d3a1d:47011/1 (0,0,1)
```

I previously saw flickers with IKEA bulb which turned out to be a crashing/rebooting bulb (issue in the firmware). Hard to judge if that is also the case here.

XRyu, 2021-01-23:
Yes, this was the time when it happened. (also to mention - but definitely minor bug -  the time-discrepancy between the Zigbee2Mqtt and Herdsman Debug log :D)

Hmmm, i am not sure. When i had these at the "old" Hue Bridge it worked without this. And some versions ago it did not happen. So not sure about this.

i also seem to have this happen irregulary (sadly pretty often), and after restarting the Coordinator everything is fine for around a half up to two days when i have to restart the container again.

Is there anything i could enable in docker, to get a even better trace where you might see more of the communications of the problematic bulbs?

Koenkk, 2021-01-24:
@XRyu sniffing the traffic of the moment it flickers is the best way to debug this I think: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

XRyu, 2021-01-25:
@Koenkk Thank you! I will try to capture when it happens next time :)
I have setuo the sniffer right now, i can see the Zigbee Data flowing - but do you have any idea why so many of the lines show "Malformed Packet" ? :)

[image]

Koenkk, 2021-01-25:
This happens when you are out of range. Moving closer to the device sending it should fix it.

closed 2021-02-25 by Koenkk

DerMega, 2021-12-15:
> Hi, I had a similar behaviour when my Paulmann Amaris LED Panel was only paired with the coordinator but not bound to a Switch. Might be an idea to bind a switch directly to your device....

@thunder-62 what do you mean by "switch"? Do I have to buy a physical switch to get rid of that problem?
I recently bought a Paulmann Amaris and it flashes while beeing switched off like every 10 Minutes :(

tbrtje, 2022-10-03:
Hi, 

> > Hi, I had a similar behaviour when my Paulmann Amaris LED Panel was only paired with the coordinator but not bound to a Switch. Might be an idea to bind a switch directly to your device....
> 
> @thunder-62 what do you mean by "switch"? Do I have to buy a physical switch to get rid of that problem? I recently bought a Paulmann Amaris and it flashes while beeing switched off like every 10 Minutes :(
> 
> Link Quality is 60 what is strange because its 1.5m to the next zigbee device.

Hi,
I am having the same issue with my Paulmann LED Panel. Have you been able to resolve this issue?

Best regards

thunder-62, 2022-10-03:
> Hi,
>
>
> Hi, I am having the same issue with my Paulmann LED Panel. Have you been able to resolve this issue?
>
> Best regards

I paired it to a Zigbee Switch and it worked. Today, I use the Amaris with the Paulman Remote as I had repetitive issues with it :-(

tbrtje, 2022-10-04:
> I paired it to a Zigbee Switch and it worked. Today, I use the Amaris with the Paulman Remote as I had repetitive issues with it :-(

What were the other issues you had?
I fixed the flickering by disabling auto_join.
The flickering only occurs when new devices are allowed to join the network.

GregWoods, 2022-10-18:
> 

PaulMann Amaris RGBW. Flickers every 6-7 minutes, as regular as clockwork!
I thought I'd fixed by having the "Disable Join (All)" option selected. 
I also had the Home Assistnant Integration "Enable newly added entities." set to DISABLED
I was convinced it was this second setting that had fixed it.

To my disappointment, after a reboot of my docker host, the lamp is back to flicking on/off/on every 6-minutes.
My settings are as they were above.

Regarding pairing the remote....
I do have the PaulMann remote somewhere. Can I pair it after it has already been paired to the coordinator?