Koenkk/zigbee2mqtt #5298: Tuya/Moes radiator thermostat gives incorrect schedule information

groenmarsmannetje, 2020-12-14:
## What happened
The device returns incorrect hours in the workdays or holidays schedule. See below example, the second schedule timeslot should be 8.30 instead of 136.30. Overtime this issue shows up in different timeslots.

`    "workdays": [
        {
            "hour": 6,
            "minute": 30,
            "temperature": 19
        },
        {
            "hour": 136,
            "minute": 30,
            "temperature": 21
        },
        {
            "hour": 18,
            "minute": 30,
            "temperature": 20
        },
        {
            "hour": 20,
            "minute": 0,
            "temperature": 19
        },
        {
            "hour": 23,
            "minute": 0,
            "temperature": 18
        },
        {
            "hour": 23,
            "minute": 30,
            "temperature": 15
        }
    ] 
`

## What did you expect to happen

The schedule should reflect correct timeslots all the time.

## How to reproduce it (minimal and precise)

Set schedule for workdays and holidays and watch how these are reflected on Json response overtime.

## Debug info
Zigbee2MQTT version: 1.16.2-dev commit: dd68d1d
Adapter hardware: ZZH
Adapter firmware version: 20201026

drzony, 2020-12-14:
Do the values stay incorrect all the time or are they just incorrect sometimes?

Westcott1, 2020-12-14:
Same here-
,"holidays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":**139**,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],

drzony, 2020-12-14:
OK guys I think I have some idea why the values might be wrong. Can you verify that the "incorrect" schedule is the active one? It seems that 128 is added for current schedule.
@groenmarsmannetje Can you versify that the readout was between 8:30 and 18:30
@Westcott1 Can you verify that readout was between 11:30 and 12:30?

Westcott1, 2020-12-14:
Here's the full status message -
Zigbee2MQTT:info  2020-12-13 23:43:32: MQTT publish: topic 'zigbee2mqtt/Radiator', payload '{"auto_lock":"MANUAL","away_mode":"OFF","away_preset_temperature":15,"battery_low":false,"boost_time":300,"child_lock":"UNLOCKED","comfort_temperature":20,"current_heating_setpoint":15,"eco_temperature":15,"force":"normal","holidays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":139,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"linkquality":88,"local_temperature":10.5,"local_temperature_calibration":"-1.0","max_temperature":25,"min_temperature":15,"position":100,"preset":"manual","preset_temperature":15,"system_mode":"heat","week":"5+2","window_detection_params":{"minutes":10,"temperature":5,"valve":"OFF"},"workdays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}]}'

Westcott1, 2020-12-14:
And one from this morning -
Zigbee2MQTT:info  2020-12-14 12:18:20: MQTT publish: topic 'zigbee2mqtt/Radiator', payload '{"auto_lock":"MANUAL","away_mode":"OFF","away_preset_temperature":15,"battery_low":false,"boost_time":300,"child_lock":"UNLOCKED","comfort_temperature":20,"current_heating_setpoint":15,"eco_temperature":15,"force":"normal","holidays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":139,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"linkquality":93,"local_temperature":11.5,"local_temperature_calibration":"-1.0","max_temperature":25,"min_temperature":15,"position":100,"preset":"manual","preset_temperature":15,"system_mode":"heat","week":"5+2","window_detection_params":{"minutes":10,"temperature":5,"valve":"OFF"},"workdays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}]}'

groenmarsmannetje, 2020-12-14:
> @groenmarsmannetje Can you versify that the readout was between 8:30 and 18:30

I think you might be correct. This was indeed the active timeslot this morning. And during the weekend I always saw one incorrect timeslot in "holidays" schedule.

drzony, 2020-12-14:
@Westcott1 @groenmarsmannetje I made a PR for this fix, are any of you able to pull it and test it?

groenmarsmannetje, 2020-12-14:
@drzony I can update edge version and check when it is released

closed 2020-12-14 by Koenkk

groenmarsmannetje, 2020-12-14:
@drzony I just installed latest edge version (1.16.2-dev commit 6198460), but the issue still exists. 

Not sure if it is related, but I noticed that "weeks" value changed from "5+2" to "6+1" after reinstall. While I am pretty sure that I did not do that myself with MQTT or on device.

drzony, 2020-12-14:
@groenmarsmannetje The state will change when device sends the data, try setting schedule once again and/or wait for new time period

groenmarsmannetje, 2020-12-14:
@drzony yes that was the trick. I can confirm that issue is solved. Thanks again.

groenmarsmannetje, 2020-12-23:
@drzony I think this change has an unexpected side effect. 

When I try to change the schedule with zigbee2mqtt/<device_name>/set/schedule and that schedule is active for that day. It sometimes doesn't accept the values or it does accept it but then the values for the active timeslot are incorrect. When I do it for a day which is not active (e.g. holiday) then there is no problem.