Koenkk/zigbee2mqtt #5299: "availability', payload 'offline'" => unavailable device

kevincaradant, 2020-12-14:
Hi

Each 3 or 4 hours I got this issue

```
Zigbee2MQTT:info  2020-12-14 10:21:23: MQTT publish: topic 'zigbee2mqtt/gledopto_desktop/availability', payload 'offline'
Zigbee2MQTT:error 2020-12-14 10:21:23: Failed to ping 'gledopto_desktop'
Zigbee2MQTT:info  2020-12-14 10:22:53: MQTT publish: topic 'zigbee2mqtt/gledopto_desktop', payload '{"brightness":254,"color":{"x":0.1904,"y":0.7201},"color_temp":201,"linkquality":87,"state":"ON"}'
Zigbee2MQTT:info  2020-12-14 10:22:53: MQTT publish: topic 'zigbee2mqtt/gledopto_desktop', payload '{"brightness":254,"color":{"x":0.1904,"y":0.7201},"color_temp":201,"linkquality":84,"state":"ON"}'
Zigbee2MQTT:info  2020-12-14 10:22:53: MQTT publish: topic 'zigbee2mqtt/gledopto_desktop', payload '{"brightness":254,"color":{"x":0.1904,"y":0.7201},"color_temp":201,"linkquality":84,"state":"ON"}'
Zigbee2MQTT:info  2020-12-14 10:22:53: MQTT publish: topic 'zigbee2mqtt/gledopto_desktop', payload '{"brightness":254,"color":{"x":0.1904,"y":0.7201},"color_temp":201,"linkquality":84,"state":"ON"}'
```

I detected it because I got also an unavailable state on Home Assistant.

As you can see between the offline time and the answer which appears 4 times on exactly the same time "10:22:53", you can see 90 seconds. It's the time that I have in my configuration.yaml. => `availability_timeout: 90`

Any idea why on my 80 devices (35 AC power) I got 2 devices (one bulb and one controller leds gledopto) which have this issue about unavailable state just during the time in seconds defined in the `configuration.yaml` file

On the picture it's unavailable during 60 seconds because yesterday it was 60s for the availability_timeout and since this morning I edited at 90 seconds to see if it happens again with 90s of unavailability and yes it's the case (see logs)
[image]
[image]

Indisponible = unavailable.

Koenkk, 2020-12-14:
What is the other device that fails?

kevincaradant, 2020-12-14:
Devices are:

[image]

And:

[image]

But at home each bulbs are gledopto bulbs and I have 3 gledopto bulbs (same version and model as gledopto_desktop) and 3 (same version and model as gledopto_leds_tv)

Koenkk, 2020-12-14:
I don't see how the availability feature could fix this. The unresponsive 5 hours just means no message was received for 5 hours. I also see you are using Zigbee2mqtt assistant, I would recommend the native frontend instead (https://www.zigbee2mqtt.io/information/frontend.html)

kevincaradant, 2020-12-14:
Not sure why it affects them but I keep always a delay (green line) before 60 minutes like that:
[image]

Without `availability_timeout` I would have  `unresponsive since 2 hours`

And yes I use both apps (yours and Z2M assistant) :)

But it's not really the aim of this issue.
I will create an other one if you think it's weird that this option help me to keep devices available but it's seems to be the case.

About the initial issue do you have any idea what's happens or what could be done to help these 2 AC devices :) ?

Koenkk, 2020-12-14:
When not using the availability feature it is expected that these device send no message. When no messages are received Zigbee2MQTT assistant marks them as "unresponsive" which is misleading I feel. This does not mean they are not connected! This should go away when you control the device EXCEPT if this device has disableDefaultResponse: true (https://github.com/Koenkk/zigbee-herdsman-converters/blob/cb1232a10e64f053a85e1f7bf07f3146d08172eb/devices.js#L7030). In this case the devices will never send messages. 

My gut feeling why the fail to pings happen: these device are not the best, in fact these are using the CC2530 chip (which is the same as the CC2531 which we know sucks). Also their meshing capabilities (thus their connectivity) isn't the best. Gledopto confirmed this with the release of their Pro line (which uses a much better chip). If this issue only happens with 2 devices I recommend to add those the the `availability_blocklist`: https://www.zigbee2mqtt.io/information/configuration.html

[image]

kevincaradant, 2020-12-14:
> When not using the availability feature it is expected that these device send no message. When no messages are received Zigbee2MQTT assistant marks them as "unresponsive" which is misleading I feel. [...]

Ok I understand the idea, but how could explain it about my 40 end-devices also ? It's often them which left every day my network. The first days, I got so many issues with Xiaomi ( temperature sensor, water-leak sensor, motion-sensor ect)
Since (coincidence ???) I added the `timeout_available: 60` I didn't have any end-devices which left the network.

> My gut feeling why the fail to pings happen: these device are not the best, in fact these are using the CC2530 chip (which is the same as the CC2531 which we know sucks)

Indeed, not the best chip. that's explain a little bit the price with them :p

>If this issue only happens with 2 devices I recommend to add those the the availability_blocklist: https://www.zigbee2mqtt.io/information/configuration.html
Yep I thought about this option if it's not possible to fix that.

But I was wondering, the device concerned by the ping error, is it really unavailable or it's just because it fails and by consequence it showed in unavailable in HA for nothing ? I mean by that, in reality, the device is still available but for Z2M it's appears "offline"

Koenkk, 2020-12-15:
Yes I agree, if I understand you correctly you want something like `availability_attempts` when this is configured the device will only be marked unavailable when ping has failed for X attempts. Let me know if this is what you mean, then I will create a new feature request for it.

kevincaradant, 2020-12-15:
You have exactly understood the idea :).

You can create the ticket about this feature request.

Thank you

Koenkk, 2020-12-17:
I've created a fr for this: https://github.com/Koenkk/zigbee2mqtt/issues/5330 . Let's see how many people are interested in it or if any additional feedback is given.

closed 2020-12-17 by Koenkk

kevincaradant, 2020-12-17:
> I've created a fr for this: #5330 . Let's see how many people are interested in it or if any additional feedback is given.

Thank you :) Perfect
I cheated and I added a thumb it :P