Koenkk/zigbee2mqtt #5302: Viessmann ZK03840 no success message in Zigbee2MQTT

elbracht, 2020-12-14:
Hello,

I'm trying to connect the Viessmann thermostat (ZK03840) with Zigbee2MQTT but unfortunately, I don't get any success message. I have the problem at the fifth step of the instructions (https://www.zigbee2mqtt.io/devices/ZK03840.html). After waiting several minutes, I only get the message "E7" on the display of the thermostat.

So far, I have tried resetting and reconnecting the thermostat several times. I also updated Zigbee2MQTT to the latest version and restarted the service and the whole server.

Does anyone have any ideas what else I can try to connect the thermostat?

## Debug info
Zigbee2MQTT version: 1.16.2
Adapter hardware: CC2531
Adapter firmware version: Z-Stack_Home_1.2 (20190608)

Koenkk, 2020-12-14:
@houssemsioud  @sjorge @ChrisChoke do you maybe know what the error is here?

ChrisChoke, 2020-12-14:
E7 is failure on wireless connection. Retry the connection process. 
I hope my translation is okay. Still have german version of the document.
I use the cc2531 with zigbee 3.0 from this repo. maybe issue cause if the 1.2 stack firmware?

sjorge, 2020-12-14:
Yeah E7 is connection error, I got those a lot too. Usually after 2 minutes the device gives up and you get the E7.

elbracht, 2020-12-15:
@sjorge Do you have an idea what helps against this connection error?

I can try to flash the zigbee 3.0 firmware on the adapter. Can I just flash the stick without getting problems with my connected Zigbee devices?

sjorge, 2020-12-15:
Having a router near by and only allowing joining through that router, seems to help but not by a lot.

~ sjorge

ChrisChoke, 2020-12-15:
@elbracht 
look at: https://www.zigbee2mqtt.io/information/FAQ.html#doesnt-require-repairing

for my understanding there is no re-pairng required with your version of zstack firmware.

@sjorge which adapter and zstack do you use?

sjorge, 2020-12-15:
1.2 source routing 

~ sjorge

ChrisChoke, 2020-12-15:
today on afternoon i paired my Viessmann TRV with the manufacturer bridge for collecting a trace. after re-pairing with z2m i saw the E7 error on the display, too. i dont know why, but just resetting the TRV solved it and it paired immediately. 
so if you see E7, press >3sec on the pairing button below the display and wait what happen. maybe press one time the button to start pairing process again.

elbracht, 2020-12-16:
The re-pairing (press >3sec on the pairing button) with z2m unfortunately has no success for me.

I plan to test the CC26X2R1 adapter over the holidays and will report if the thermostat works with it.

closed 2021-01-24 by github-actions[bot]