Koenkk/zigbee2mqtt #5306: Tuya ZigBee 3.0 Smart Light Switch Relay Module 2 Gang Zigbee RF Switch Module

Minutus, 2020-12-14:
Hi,
I nead help with support for new device. It is 2 gang Smart light switch module. I try to do it myself, but I'm begginer in programing and my experiment with scripts didn't work.

This switch module is simmilar to this:
https://www.zigbee2mqtt.io/devices/TS0121_switch.html

Basic information about the module is here:
https://www.aliexpress.com/item/1005001831813560.html?spm=a2g0o.productlist.0.0.3ed691b6ZudzYS&algo_pvid=873ef32c-ca5f-4628-8093-a33dc12b8f3d&algo_expid=873ef32c-ca5f-4628-8093-a33dc12b8f3d-1&btsid=2100bdca16079794517372193e9e72&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

## Information about the device:
model: 'MS-104BZ',
vendor: 'Moes',
description: 'Smart light switch module (2 gang)',

## data/database.db entry of the device:
{"id":8,"type":"Router","ieeeAddr":"0x842e14fffe05b313","nwkAddr":49008,"manufId":4098,"manufName":"_TZ3000_pmz6mjyu","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":9,"inClusterList":[0,4,5,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_pmz6mjyu","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"32770":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1}}},"binds":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":9,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1607968346260}

## logs from zigbee2mqtt for switching swich 1:
debug 2020-12-14 18:34:18: Received Zigbee message from '0x842e14fffe05b313', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
warn  2020-12-14 18:34:18: Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_pmz6mjyu'
warn  2020-12-14 18:34:18: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-12-14 18:34:21: Received Zigbee message from '0x842e14fffe05b313', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
warn  2020-12-14 18:34:21: Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_pmz6mjyu'
warn  2020-12-14 18:34:21: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

## logs from zigbee2mqtt for switching swich 2:
debug 2020-12-14 18:34:22: Received Zigbee message from '0x842e14fffe05b313', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
warn  2020-12-14 18:34:22: Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_pmz6mjyu'
warn  2020-12-14 18:34:22: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-12-14 18:34:24: Received Zigbee message from '0x842e14fffe05b313', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 2 with groupID 0
warn  2020-12-14 18:34:24: Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_pmz6mjyu'
warn  2020-12-14 18:34:21: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

Koenkk, 2020-12-15:
Try with:

```js
{
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_pmz6mjyu'}],
    model: 'MS-104BZ',
    description: 'Smart light switch module (2 gang)',
    vendor: 'Moes',
    extend: generic.switch,
    meta: {configureKey: 1, multiEndpoint: true},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint1 = device.getEndpoint(1);
        await bind(endpoint1, coordinatorEndpoint, ['genOnOff']);
        await configureReporting.onOff(endpoint1);
        const endpoint2 = device.getEndpoint(2);
        await bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
        await configureReporting.onOff(endpoint2);
    },
},
```

Minutus, 2020-12-15:
Hi,
sorry Im real "user" and I try push it trough External converters, but I dont uderstand what is necessary to write in this file to make it wokrs. It ends with:
SyntaxError: Missing initializer in const declaration

Sorry for bothering with the basic questions. Examples dont help me. Can you write my exct code for External convertert file?

When I tray push it trough devices. it ends with error:
 cp: can't create './node_modules/zigbee-herdsman-converters/devices.js': No such file or directory

I use HassOS 4.17 with zigbee2mqtt version 1.16.2 and Home Assistant

Koenkk, 2020-12-17:
Since I have good confidence that this config works, I've added it.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Minutus, 2020-12-24:
Thanks for your solution. I test it and it works very well. It is possible to forward it to stable version... 
Merry Xmas 😁

Koenkk, 2020-12-24:
Good, will be added to stable with next z2m release (1 january).

closed 2020-12-24 by Koenkk