Koenkk/zigbee2mqtt #5314: Immax Remote Type:07007L

cubertt, 2020-12-15:
I tried, but i am using Hass.io and the Node.js Windows-way is full of errors:

`c:\users\test\node_modules\unix-dgram\src\unix_dgram.cc(9):` fatal error C1083: Datei (Include) kann nicht geöffne
t werden: "unistd.h": No such file or directory `[C:\Users\test\node_modules\unix-dgram\build\unix_dgram.vcxproj]`

## Information about the device + link

I bought 2 Immax remotes, Type  07007L with Colours, dimming, etc functions. Cheap, 6 Euro each.
[https://www.immaxneo.cz/en/products/immax-neo-remote-control-rgbcct-zigbee-3-0/] (https://www.immaxneo.cz/en/products/immax-neo-remote-control-rgbcct-zigbee-3-0/)

Other products from Immax are supported, perhabs somebody has this remote and can follow the guide?  https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

## data/database.db entry of the device
?
Can i do something else to help?

cubertt, 2020-12-21:
Had time to install raspbian on a raspberry 3 and followed the steps of the guide.
I have introduced a new device on basis of an Illumize remote in devices.js:

`{
        zigbeeModel: ['IM-Z30-Remote'],
        model: '07007L',
        vendor: 'Immax',
        description: 'Zigbee handheld touch remote RGBW 3 channels',
        fromZigbee: [
            fz.battery, fz.command_move_to_color, fz.command_move_to_color_temp, fz.command_move_hue,
            fz.command_step, fz.command_recall,
            fz.ZG2819S_command_on, fz.ZG2819S_command_off,
        ],
        exposes: [e.battery(), e.action([
            'color_move', 'color_temperature_move', 'hue_move', 'brightness_step_up', 'brightness_step_down', 'recall_*', 'on', 'off',
        ])],
        toZigbee: [],
        meta: {multiEndpoint: true},
        endpoint: (device) => {
            return {ep1: 1, ep2: 2, ep3: 3, ep4: 4};
        },
    },`

Now i have the attached logs ( i pressed the following buttons: on/off, +,-, w, colour-ring in two directions, n, 3, 2, 1):
[Zigbee2mqtt-Logs_07007l_Immax.txt](https://github.com/Koenkk/zigbee2mqtt/files/5724909/Zigbee2mqtt-Logs_07007l_Immax.txt)

I need now the converter, has anybody an fitting converter from another device?
Thanks for help,
cubertt

Koenkk, 2020-12-21:
Only use the `fz.command_*` converters, not e.g. `fz.ZG2819S_command_on` (you can use `fz.command_on` for this one).

For: `Zigbee2MQTT:debug 2020-12-21 14:43:16: No converter available for '07007L' with cluster 'lightingColorCtrl' and type 'commandColorLoopSet' and data '{"action":0,"direction":0,"starthue":0,"time":0,"updateflags":15}'`  you can use `fz.command_color_loop_set`

closed 2021-01-21 by Koenkk

kisdaniel, 2021-03-16:
@Koenkk 
Do you mean this?
```
	{
        zigbeeModel: ['IM-Z30-Remote'],
        model: '07007L',
        vendor: 'Immax',
        description: 'Zigbee handheld touch remote RGBW 3 channels',
        fromZigbee: [
            fz.battery, 
            fz.command_move_to_color, 
            fz.command_move_to_color_temp, 
            fz.command_move_hue,
            fz.command_step,
            fz.command_recall,
            fz.command_on,
            fz.command_off,
            fz.command_color_loop_set
        ],
        exposes: [e.battery(), e.action([
            'color_move',
            'color_temperature_move',
            'hue_move',
            'brightness_step_up',
            'brightness_step_down',
            'recall_*',
            'on',
            'off',
        ])],
        toZigbee: [],
        meta: {multiEndpoint: true},
        endpoint: (device) => {
            return {ep1: 1, ep2: 2, ep3: 3, ep4: 4};
        },
    },
```

Koenkk, 2021-03-16:
Yes, if it works correctly with that config I can add it.

kisdaniel, 2021-03-16:
I am testing it.

When I press "+" or "-" nothing happens in mqtt, 

When i press power button I receive an on and off message, but It should be better if a toggle message or button press would arrive.
```
Mar 16 19:46:04 home-automation-1 npm[3216]: Zigbee2MQTT:info  2021-03-16 19:46:04: MQTT publish: topic 'zigbee2mqtt/COLORREMOTE1', payload '{"action":"off_ep1","action_group":1,"linkquality":47}'
Mar 16 19:46:06 home-automation-1 npm[3216]: Zigbee2MQTT:info  2021-03-16 19:46:06: MQTT publish: topic 'zigbee2mqtt/COLORREMOTE1', payload '{"action":"on_ep1","action_group":1,"linkquality":47}'
```

kisdaniel, 2021-03-16:
I also get a `No converter available for '07007L' with cluster 'lightingColorCtrl'`:

```
Mar 16 20:13:45 home-automation-1 npm[3365]: Zigbee2MQTT:debug 2021-03-16 20:13:45: Received Zigbee message from 'COLORREMOTE1', type 'commandColorLoopSet', cluster 'lightingColorCtrl', data '{"action":0,"direction":0,"starthue":0,"time":0,"updateflags":15}' from endpoint 1 with groupID 3
Mar 16 20:13:45 home-automation-1 npm[3365]: Zigbee2MQTT:debug 2021-03-16 20:13:45: Received Zigbee message from 'COLORREMOTE1', type 'commandEnhancedMoveToHueAndSaturation', cluster 'lightingColorCtrl', data '{"enhancehue":46517,"saturation":254,"transtime":3}' from endpoint 1 with groupID 3
Mar 16 20:13:45 home-automation-1 npm[3365]: Zigbee2MQTT:debug 2021-03-16 20:13:45: No converter available for '07007L' with cluster 'lightingColorCtrl' and type 'commandEnhancedMoveToHueAndSaturation' and data '{"enhancehue":46517,"saturation":254,"transtime":3}'
Mar 16 20:13:45 home-automation-1 npm[3365]: Zigbee2MQTT:debug 2021-03-16 20:13:45: Received Zigbee message from 'COLORREMOTE1', type 'commandColorLoopSet', cluster 'lightingColorCtrl', data '{"action":0,"direction":0,"starthue":0,"time":0,"updateflags":15}' from endpoint 1 with groupID 3
Mar 16 20:13:45 home-automation-1 npm[3365]: Zigbee2MQTT:debug 2021-03-16 20:13:45: Received Zigbee message from 'COLORREMOTE1', type 'commandEnhancedMoveToHueAndSaturation', cluster 'lightingColorCtrl', data '{"enhancehue":39321,"saturation":254,"transtime":3}' from endpoint 1 with groupID 3
Mar 16 20:13:45 home-automation-1 npm[3365]: Zigbee2MQTT:debug 2021-03-16 20:13:45: No converter available for '07007L' with cluster 'lightingColorCtrl' and type 'commandEnhancedMoveToHueAndSaturation' and data '{"enhancehue":39321,"saturation":254,"transtime":3}'
```

kisdaniel, 2021-03-16:
@Koenkk this seems to be working, I have commented out some lines, you should remove. I don't really know what it can be.
Battery information is also missing, but I don't really know when it can be sent by device.

```
    // Immax Neo
    {
        zigbeeModel: ['IM-Z30-Remote'],
        model: '07007L',
        vendor: 'Immax',
        description: 'Zigbee handheld touch remote RGBW 3 channels',

        fromZigbee: [
            fz.battery, 
            fz.command_move,
//            fz.command_move_to_color, 
            fz.command_move_to_color_temp, 
            fz.command_move_hue,
            fz.command_ehanced_move_to_hue_and_saturation,
            fz.command_step,
//            fz.command_recall,
            fz.command_on,
            fz.command_off
//            fz.command_color_loop_set
        ],
        exposes: [e.battery(), e.action([
            'color_move',
            'color_temperature_move',
            'hue_move',
            'brightness_step_up',
            'brightness_step_down',
            'recall_*',
            'on',
            'off',
        ])],
        toZigbee: [],
        meta: {multiEndpoint: true},
        endpoint: (device) => {
            return {ep1: 1, ep2: 2, ep3: 3, ep4: 4};
        },
    },
```

Some messages from mqtt with the above configuration:

```
{"action":"enhanced_move_to_hue_and_saturation_ep1","action_enhanced_hue":21588,"action_group":1,"action_hue":118.58642578125,"action_saturation":254,"action_transition_time":3,"linkquality":13}
```

`{"action":"off_ep1","action_group":1,"linkquality":13}`

`{"action":"on_ep1","action_group":1,"linkquality":13}`

`{"action":"color_temperature_move_ep1","action_color_temperature":285,"action_group":1,"action_transition_time":1,"linkquality":13}`

Koenkk, 2021-03-17:
Try with the config below and repair the device, then battery might work:

```js
{
    zigbeeModel: ['IM-Z30-Remote'],
    model: '07007L',
    vendor: 'Immax',
    description: 'Zigbee handheld touch remote RGBW 3 channels',
    fromZigbee: [fz.battery, fz.command_move, fz.command_move_to_color_temp, fz.command_move_hue,
        fz.command_ehanced_move_to_hue_and_saturation, fz.command_step, fz.command_on, fz.command_off],
    exposes: [e.battery(), e.action(['color_move', 'color_temperature_move', 'hue_move', 'brightness_step_up',
        'brightness_step_down','recall_*', 'on', 'off'])],
    toZigbee: [],
    meta: {multiEndpoint: true, configureKey: 1},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint);
    },
},
```

cubertt, 2021-08-02:
I started another Experiment, with the config og koenk from 17. march, but zigbee2 mqtt does not start: (on hass.io)

device.js file:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const definition = {
    zigbeeModel: ['IM-Z30-Remote'],
    model: '07007L',
    vendor: 'Immax',
    description: 'Zigbee handheld touch remote RGBW 3 channels',
    fromZigbee: [fz.battery, fz.command_move, fz.command_move_to_color_temp, fz.command_move_hue,
        fz.command_ehanced_move_to_hue_and_saturation, fz.command_step, fz.command_on, fz.command_off],
    exposes: [e.battery(), e.action(['color_move', 'color_temperature_move', 'hue_move', 'brightness_step_up',
        'brightness_step_down','recall_*', 'on', 'off'])],
    toZigbee: [],
    meta: {multiEndpoint: true, configureKey: 1},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint);
},

module.exports = definition;

error:

23:14:00] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.18.1 start /app
> node index.js
(node:773) UnhandledPromiseRejectionWarning: SyntaxError: Unexpected token '.'
(Use `node --trace-warnings ...` to show where the warning was created)
(node:773) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:773) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
[23:14:04] INFO: Handing over control to Zigbee2mqtt Core ...

Help would be appreciated

cubertt, 2021-08-03:
Hi there, i have tested the above suggestion from Koenkk , 17.March and it works, but some stuff is missing:

Battery not working

+ and + Buttons give the following message in the log:

```
Debug Received Zigbee message from '0x00158d00024f70e7', type 'commandStop', cluster 'genLevelCtrl', data '{}' from endpoint 1 with groupID 3
Debug No converter available for '07007L' with cluster 'genLevelCtrl' and type 'commandStop' and data '{}'
Debug Received Zigbee message from '0x00158d00024f70e7', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":1,"rate":80}' from endpoint 1 with groupID 3
```
and "N" Button:

 No converter available for '07007L' with cluster 'lightingColorCtrl' and type 'commandColorLoopSet' and data '{"action":0,"direction":0,"starthue":0,"time":0,"updateflags":15}'
Debug Received Zigbee message from '0x00158d00024f70e7', type 'commandEnhancedMoveToHueAndSaturation', cluster 'lightingColorCtrl', data '{"enhancehue":43851,"saturation":254,"transtime":0}' from endpoint 1 with groupID 3
Info MQTT publish: topic 'zigbee2mqtt/0x00158d00024f70e7', payload '{"action":"enhanced_move_to_hue_and_saturation_1","action_enhanced_hue":43851,"action_group":3,"action_hue":240.8807373046875,"action_saturation":254,"action_transition_time":0,"linkquality":78}'
Info MQTT publish: topic 'zigbee2mqtt/0x00158d00024f70e7', payload '{"action":"","linkquality":78}'
Info MQTT publish: topic 'zigbee2mqtt/0x00158d00024f70e7/action', payload 'enhanced_move_to_hue_and_saturation_1'
Debug Received Zigbee message from '0x00158d00024f70e7', type 'commandQueryNextImageRequest', cluster 'genOta', data

cubertt, 2022-12-06:
Hi there old problem, no solution? no:
I got it to work with:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['IM-Z30-Remote'],
    model: '07007L',
    vendor: 'Immax',
    description: 'Zigbee handheld touch remote RGBW 3 channels',
    fromZigbee: [fz.battery, fz.command_move, fz.command_move_to_color_temp, fz.command_move_hue,
        fz.command_ehanced_move_to_hue_and_saturation, fz.command_step, fz.command_on, fz.command_off, fz.command_color_loop_set],
    exposes: [e.battery(), e.action(['color_move', 'color_temperature_move', 'hue_move', 'brightness_step_up',
        'brightness_step_down','recall_*', 'on', 'off'])],
    toZigbee: [],
    meta: {multiEndpoint: true, configureKey: 1},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint);
    },		
};

module.exports = definition;
```

One error appeared sometimes:

#No converter available for '07007L' with cluster 'genLevelCtrl' and type 'commandStop' and data '{}'

but i cann't reproduce it...

Picture:

https://interlink-static1.tsbohemia.cz/immax-neo-dalkovy-ovladac-rgb-cct-pro-zigbee_ien319653.jpg

salopette, 2024-02-22:
Können Sie bitte die Fernbedienung hinzufügen?