Koenkk/zigbee2mqtt #5319: BlitzWolf BW-IS4 doesn't work anymore with Z2M 1.16.2

lucipacurar, 2020-12-16:
## What happened
BlitzWolf BW-IS4 doesn't work anymore with Z2M 1.16.2
## What did you expect to happen
BlitzWolf BW-IS4 should work with Z2M 1.16.2 as it did with previous versions
## How to reproduce it (minimal and precise)
Upgrade from Z2M 1.15.0 to Z2M 1.16.2

## Debug info
Zigbee2MQTT version: 1.16.2
Adapter hardware: CC2531
Adapter firmware version: {"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}

lucipacurar, 2020-12-16:
I don't understand it anymore: RPi 4, Ubuntu 20.04.1 LTS on SDCard + Z2M 1.15/1.14.3 running inside docker and the device is fine. And I have the same thing, plain Ubuntu 20.04.1 LTS on a USB SSD drive + Z2M 1.15 isn't working. Any clues on where to look?

Koenkk, 2020-12-17:
Can you provide the data/database.db entry of this device?

lucipacurar, 2020-12-17:
This is the entry from version 1.14.3 that works and is in use right now: 

```
{"id":7,"type":"EndDevice","ieeeAddr":"0xec1bbdfffe70b235","nwkAddr":62216,"manufId":4098,"modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,1026,1029],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65506":19,"modelId":"TS0201","zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2110}},"msRelativeHumidity":{"attributes":{"measuredValue":5430}},"genPowerCfg":{"attributes":{"batteryVoltage":28,"batteryPercentageRemaining":200}}},"binds":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1608207322924}
```

Probably it has been removed in newer versions and when creating a new network it doesn't get matched to anything anymore. Let me know what else do you need. Thank you!

Koenkk, 2020-12-17:
I've checked but it's still there: https://github.com/Koenkk/zigbee-herdsman-converters/blob/7623a5855922893e9d47cbad755fc3236fccd794/devices.js#L1612

Can you provide your debug startup log? (first 30 seconds after starting z2m)

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

lucipacurar, 2020-12-18:
Here it is:
https://pastebin.com/hs3Pypg7

Koenkk, 2020-12-18:
Where is it reported as unsupported? Assuming it is `Blitzwolf (0xec1bbdfffe70b235): TS0201 - TuYa Temperature & humidity sensor with display (EndDevice)` it's recognised correctly.

```
Zigbee2MQTT:info  2020-12-18 11:58:07: Currently 8 devices are joined:
Zigbee2MQTT:info  2020-12-18 11:58:07: Aqara garden (0x00158d00047b8e3a): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2020-12-18 11:58:07: Aqara living (0x00158d0004866f75): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2020-12-18 11:58:07: Aqara water leak bathroom (0x00158d00042d092c): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
Zigbee2MQTT:info  2020-12-18 11:58:07: Power outlet living (0x588e81fffe5be90f): E1603/E1702 - IKEA TRADFRI control outlet (Router)
Zigbee2MQTT:info  2020-12-18 11:58:07: Aqara water leak kitchen (0x00158d00042ad954): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
Zigbee2MQTT:info  2020-12-18 11:58:07: Blitzwolf (0xec1bbdfffe70b235): TS0201 - TuYa Temperature & humidity sensor with display (EndDevice)
Zigbee2MQTT:info  2020-12-18 11:58:07: Power outlet lobby (0x680ae2fffef2f903): E1603/E1702 - IKEA TRADFRI control outlet (Router)
Zigbee2MQTT:info  2020-12-18 11:58:07: Repeater (0x680ae2fffe9c84b0): E1746 - IKEA TRADFRI signal repeater (Router)
Zigbee2MQTT:warn  2020-12-18 11:58:07: `permit_join` set to  `true` in configuration.yaml.
```

lucipacurar, 2020-12-18:
OK, let me delete it from the database file and let Z2M add it again. On a brand new setup, the device is recognized as BlitzWolf BW-IS4, and not TS0201, and it's reported as an unsupported device.

Koenkk, 2020-12-18:
> and it's reported as an unsupported device.

Can you provide the logging of that?

lucipacurar, 2020-12-18:
I deleted the database file, the stats file, and the devices from configuration, and it doesn't report it as an unsupported device. All I get now is: 

```
Zigbee2MQTT:debug 2020-12-18 21:09:31: No converter available for 'TS0201' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3}'
```

and it works fine. 

Maybe the network was in some inconsistent state. Sorry for wasting your time on this.

Koenkk, 2020-12-18:
Good it's working now, those messages are normal indeed.

closed 2020-12-18 by Koenkk

lucipacurar, 2020-12-18:
@Koenkk: I've restored the backup before deleting the files from the previous comment, and now I get the error:

https://pastebin.com/6h9VgD8J

database.db:
```
{"id":7,"type":"EndDevice","ieeeAddr":"0xec1bbdfffe70b235","nwkAddr":65296,"manufId":4098,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genPowerCfg":{"attributes":{"batteryVoltage":28,"batteryPercentageRemaining":200}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2430}},"msRelativeHumidity":{"attributes":{"measuredValue":5730}}},"binds":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1608320263782}
```

lucipacurar, 2020-12-18:
It looks like some inconsistent state. I restored the database.db file again from the backup, restarted Z2M, and the error is gone.