Koenkk/zigbee2mqtt #5320: Ikea E1524/E1810 toggle button stopped working

fnfid, 2020-12-16:
Forwarded two commits to 
 1.16.2-dev commit: eba8849
unfortunally the toggle button of all Ikea E1524/E1810 remotes stopped working, no message, apart from this, all other stuff including all other buttons are working.

fnfid, 2020-12-16:
returning to master fixed it for the moment

BugsonPL, 2020-12-16:
The same here. With DEV docker Ikea stuff  (LED1623G12,E1524/E1810, L1529) Stopped working.

But updated 15 min. ago and lights are OK but in remote the main/middle button is not working.

fnfid, 2020-12-17:
some of the changes in the fromzigbee.js in the converter must cause this... should be somewhere around here
https://github.com/Koenkk/zigbee-herdsman-converters/commit/80f164134acd7188c3d035ad4e8d4f55cec495f2#diff-b676ab0df4d49b8572967cb4c92a2dc6264091131f6b1828ca13d0f9656e367a
but haven't found it so far

fnfid, 2020-12-17:
It should be fixed in the near future by drzony and his pull :-)
https://github.com/Koenkk/zigbee-herdsman-converters/pull/1927
Fix IKEA Controller toggle button after refactor #1927
Thanks!!!!!

closed 2020-12-17 by fnfid