Koenkk/zigbee2mqtt #5324: Z2M does not start as a service, OK in console

Westcott1, 2020-12-16:
Running directly on an Rpi4b. node v14.15.1
When I first installed Z2M some months ago the service ran correctly.
Then after a Z2M update (forgot which) the service errors on start.
Running directly from the console is OK
It's due to serialport NODE_MODULE_VERSION differences.
I've tried npm rebuild, npm install, and $(npm bin)/electron-rebuild, after removing all node_modules
The error is always the same, but the NODE_MODULE_VERSIONs can be different

Dec 16 11:05:29 RaspberryPi-4b zigbee2mqtt[13374]: Zigbee2MQTT:info  2020-12-16 11:05:29: Starting Zigbee2MQTT version 1.16.2-dev (commit #dd68d1d)
Dec 16 11:05:29 RaspberryPi-4b zigbee2mqtt[13374]: Zigbee2MQTT:info  2020-12-16 11:05:29: Starting zigbee-herdsman (0.13.41)
Dec 16 11:05:29 RaspberryPi-4b zigbee2mqtt[13374]: Zigbee2MQTT:error 2020-12-16 11:05:29: Error while starting zigbee-herdsman
Dec 16 11:05:29 RaspberryPi-4b zigbee2mqtt[13374]: Zigbee2MQTT:error 2020-12-16 11:05:29: Failed to start zigbee
Dec 16 11:05:29 RaspberryPi-4b zigbee2mqtt[13374]: Zigbee2MQTT:error 2020-12-16 11:05:29: Exiting...
Dec 16 11:05:29 RaspberryPi-4b zigbee2mqtt[13374]: Zigbee2MQTT:error 2020-12-16 11:05:29: Error: The module '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/bindings.node'
Dec 16 11:05:29 RaspberryPi-4b zigbee2mqtt[13374]: was compiled against a different Node.js version using
Dec 16 11:05:29 RaspberryPi-4b zigbee2mqtt[13374]: NODE_MODULE_VERSION 83. This version of Node.js requires
Dec 16 11:05:29 RaspberryPi-4b zigbee2mqtt[13374]: NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
Dec 16 11:05:29 RaspberryPi-4b zigbee2mqtt[13374]: the module (for instance, using `npm rebuild` or `npm install`).

Westcott1, 2021-01-03:
I somehow seem to have overcome this.
Changed the service config file (that used to work!) to be the one shown in "Running Zigbee2MQTT"
Updated Z2M to latest dev 1.17.0
Service starts correctly.

Koenkk, 2021-01-03:
```
Dec 16 11:05:29 RaspberryPi-4b zigbee2mqtt[13374]: NODE_MODULE_VERSION 83. This version of Node.js requires
Dec 16 11:05:29 RaspberryPi-4b zigbee2mqtt[13374]: NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
```

This error can be solved by executing `npm ci` in the zigbee2mqtt install directory (`/opt/zigbee2mqtt`). 

Since errors is fixed I guess this can be closed now.

closed 2021-01-03 by Koenkk