Koenkk/zigbee2mqtt #5325: Z2M Admin does not work correctly in dev-branch

VincentSC, 2020-12-16:
Seems the messages have breaking changes in the dev-branch, as Z2M Admin does not work correctly anymore.
- all info messages (status updates) result in a message in the upper-right corner.
- "last messages" of the selected device is empty

Koenkk, 2020-12-17:
- Do you mean https://github.com/yllibed/Zigbee2MqttAssistant ?
- all info messages (status updates) result in a message in the upper-right corner.  -> maybe this happens because the new api is now enabled by default and I guess Zigbee2MqttAssistant cannot deal with it.
- "last messages" of the selected device is empty -> need more info what is used here underwater.

Note that Zigbee2MQTT also comes with a frontend nowadays: https://www.zigbee2mqtt.io/information/frontend.html

VincentSC, 2021-01-17:
Great! I've recently moved from [Z2MAdmin](https://flows.nodered.org/flow/1f1c0dde4ff193e5c14d5b99442a7d00) to the built-in devices admin. It covers my needs.

Z2MAdmin is a bit more compact, so maybe some design ideas can come from that.
[image]

closed 2021-01-17 by VincentSC