Koenkk/zigbee2mqtt #5327: Adding a description to devices

pepp86, 2020-12-17:
Hi,

I would like to introduce a description field for all devices. I know to rename devices, but I would like to stay with a unique ID in MQTT. Do you think that is easy to implement?

pepp86, 2021-01-17:
Unstale

pepp86, 2021-03-23:
Hi,

any update on this one? Would really love to have this feature in place.

Koenkk, 2021-03-23:
@pepp86 I see but currently it doesn't seem many people are interested in it (given that there are only 2 thumbs up on the OP). It's also unclear to me where you expect this `description` field to be exposes. It might be worth to better describe your request to also attract other people.

pepp86, 2021-04-06:
Hi @Koenkk, I understand.

But found other issues/request where people basically want the same:

https://github.com/Koenkk/zigbee2mqtt/issues/6958
https://github.com/Koenkk/zigbee2mqtt/issues/6886

Thought it will be easy to add a description of a device to the system, but seems to be an bigger effort to deliver. Will think about other solutions then.

DivanX10, 2021-04-07:
Is it possible to make the name of the device entered in a separate file, without affecting the unique number of the device? It turns out that in the web interface itself it will be clear what kind of device, and in the home assistant only unique device numbers will arrive and this will be correct. You can add a choice of whether to include device names in topics so that they are displayed in home assistant or not. Everyone will decide for themselves how it is convenient for them to use it.

------------------------

Можно ли сделать так, чтобы название устройства вносилось в отдельный файлик, не затрагивая уникальный номер устройства? Получится, что в самом веб интерфейсе будет понятно, что за устройство, а в home assistant будут прилетать только уникальные номера устройств и это будет правильно. Можно добавить выбор, включать ли название устройств в топики, чтобы они отображались в home assistant или не включать. Каждый сам для себя решит, как ему удобно использовать.

timoline, 2021-05-03:
I also would like to see this feature, also a description for groups would be nice

closed 2021-06-11 by github-actions[bot]

GHolli, 2025-03-03:
I also like the idea.
Why: Dual-use - when using via openHAB, for instance, I would prefer the unique ID over the friendly name in some cases.
But when using also the z2m dashboard in parallel, it would be handy to have a description as well (e. g. location of the sensor).

GHolli, 2025-04-27:
This feature is implemented now, not sure when it was introduced.
A description can be added to every device.
So, for completeness, the issue was closed due to inactivity, but now the requested feature is available.