Koenkk/zigbee2mqtt #5330: Availability attempts

Koenkk, 2020-12-17:
## Is your feature request related to a problem? Please describe
When using the availability feature, pings can sometime fail due to e.g. the Zigbee network being too busy at that time. When a ping fails the device is marked as unavailable in Home Assistant until a ping succeeds. When a device is unavailable in Home Assistant it cannot be controlled from it.

## Describe the solution you'd like
Introduce a `availability_attempts` option, when e.g. `availability_attempts: 2` the device will only be marked as offline when the ping failed for two times in a row.

## Describe alternatives you've considered
In case only a certain set of devices show this problem it would be possible to disable availability checking from them through the `availability_blocklist`. 

## Additional context
Discussion in: https://github.com/Koenkk/zigbee2mqtt/issues/5299

copystring, 2020-12-18:
How does pinging a device work with zigbee? Sounds like it works different than traditional pinging known from ip network devices.
I do find knowing when a device is offline is important to troubleshoot connectivity problems. The past few days I have been watching my zigbee network carefully to see which devices go offline. It feels like they do not actually go offline but instead do not respond in time to the ping send by z2m.
Is it possible that a ping request actually never returns pong or is it possible it just takes longer? Depending on that possible solution could be:
Scenario 1: Ping could possible never return a pong - availability_attempts
Scenario 2: Ping always returns a pong but could take longer: increase the timeout of that response if that is possible. Does zigbee allow to increase the timeout of a response to ping or is that "hardcoded" in the zigbee protocol?

zhapwey, 2020-12-19:
I think that an option to evaluate would be to be able to ping devices when a device has not sent for more time than that established in the availability_timeout.
So the devices were not pinged all at once, which I think is what makes some devices take so long to answer.
In fact, if the messages that we ignore or that do not have a converter could also be counted, it would be more real, since the device sends data and is connected. This could reduce the time the end devices are disconnected as well.

There are times when a device that sent 2 seconds ago gives you disconnected, because it does not answer the ping.

Best Regards,

kevincaradant, 2020-12-19:
> How does pinging a device work with zigbee? Sounds like it works different than traditional pinging known from ip network devices.
> I do find knowing when a device is offline is important to [...]

I agree with your thought. My case will be satisfied by the scenario 1 (I'm the guy which is concerned by the #5299 issue :))

But maybe it could be because my devices doesn't send ? doesn't receive the ping ask ? Doesn't have time to answer (some interferences, lags) ? So the scenerio 2 could fix also my problem but it's not sure. Scenerio 1 for sure will fix the issue and can also take ina ccount the use case for scenario 2.

For me, the priority is to develop the scenerio 1 which concerne and can solve more use cases than the scenario 2.
But in the second time, I agree also to have the feature described in the scenario 2 :)

sjorge, 2020-12-19:
I need to look at the code again, but if a device has send a message on its own less than the availability_timeout ago, we could skip the ping when the timer fires. 

But I’m pretty sure we’re already doing that, I’d not or could safe a bit of traffic allow other devices a less busy media to reply. If I don’t forget I’ll have a look tomorrow.

Koenkk, 2020-12-20:
Zigbee has no pining mechanism, what we do is read the `zclVersion` property from the `genBasic` cluster. Response means successful ping here.

As @sjorge mentioned, when a message is received from a device the timer is reset. So if you e.g. have `availability_timeout: 60` and receive a message at 10 seconds, the device will not be pinged at 60 seconds but at 70.

sjorge, 2020-12-20:
> As @sjorge mentioned, when a message is received from a device the timer is reset. So if you e.g. have `availability_timeout: 60` and receive a message at 10 seconds, the device will not be pinged at 60 seconds but at 70.

I was just about to look it up :) saved me the effort.

I guess a cleaner implementation would ideally have an `availability_attempts` and an `availability_backoff` (=`availability_timeout` by default)

```
availability_timeout=300
availability_attempts=1 (default)
availability_backoff=availability_timeout
```

Would work as it does now, try every 5 minutes unless the device has sent a message and go offline when it fails to respond.

```
availability_timeout=300
availability_attempts=3
availability_backoff=availability_timeout
```

Same as above, but it needs to have missed 3 consecutive timeouts, so ~15min

```
availability_timeout=300
availability_attempts=3
availability_backoff=60
```

Same as above, but timer after the first failure would be 60 seconds instead of 300 on a success or message from the device it goes back to 300... so that would mean 5+1+1 = 7 minutes before the device goes offline.

To encourage people not to bump attempts and decrease the timeout, as that will probably results in configs likes

timeout=15, attempts=4 which would wreck havoc on the mesh...

Seems like a big rework though (not something I have time for atm :()

Koenkk, 2020-12-20:
@sjorge yes was already planning some availabilty rework and will take this into account.

io53, 2021-01-11:
This would be great!
Could you on the same time make reporting/successful control of a devices turn it available? :)
It seems like devices will stay unavailable even tho they are reporting the state successfully.

Morphy99, 2021-01-11:
Thanks for the heads-up Koenkk. As mentioned in  #4703 I think it would be very useful to be able to adjust the default timeout for battery powered devices from the current default 25 hours to a user set interval for each device. I think this is important with critical sensors such as smoke detectors or leak detectors. Even though the chances are small I'd like to know within minutes if there is an issue with one of them, not hours.

This is especially pertinent now we can adjust the reporting interval on certain devices.

sjorge, 2021-01-11:
It's a bit hard to do though, the timeout is 24... because most of them only check in 1x ~24h with there battery status if all is well.

Having it at 1 or 2 hours will just show them all offline.

Morphy99, 2021-01-11:
@sjorge I thought this could be adjusted with the reporting interval topic? I don't mean to change the global reporting timeout either, I mean to change it per device. 24 hours is a very generalised figure, I find they vary quite a lot per device or sensor type. Could leave that as the default and then specify it for those devices that report more often.

closed 2021-02-18 by github-actions[bot]

kevincaradant, 2021-03-17:
Why this issue is closed ? I think this should be still open :)

Koenkk, 2021-03-17:
@kevincaradant continued in #6281