Koenkk/zigbee2mqtt #5332: Request for support new Tuya/Moes Radiator Valve _TZE200_i48qyn9s

tinker2code, 2020-12-17:
Hello,
first at all: I am a complete newbie in zigbee2mqtt and also in the GitHub world (just created the GitHub-account now). And unfortunately English is not one of my preferences. But I will try my very best... in both - to understand how zigbee2mqtt works and to write an understandable English. :-) In addition, I'm not really a programmer, I learned Basic and Turbo Pascal at school over 30 years ago, never learnd java, c++ or python. So, everything I do here is more tinker with code.

I bought a Zigbee Radiator Thermostat and was hoping that with the help of the instructions "How to support new devices" I could get everything working. But for my first steps with zigbee2mqtt it doesn't seem it is the best project to start with, it is not as easy as I was hoping before. So I would like to take advantage of the nice offer "_In case you require any help feel free to create an issue._"

## Information about the device + link
model 'TS0601'
manufacturer name '_TZE200_i48qyn9s'
vendor: 'Essentials', description: 'TRV', model: '120144'
[https://www.wesmartify.de/heizung/heizkoerperradiator/333/essentials-smart-home-heizkoerperthermostat/](https://www.wesmartify.de/heizung/heizkoerperradiator/333/essentials-smart-home-heizkoerperthermostat/)
[https://www.pollin.de/p/heizkoerper-thermostat-essentials-zigbee-590612](https://www.pollin.de/p/heizkoerper-thermostat-essentials-zigbee-590612)

## data/database.db entry of the device
`{"id":7,"type":"EndDevice","ieeeAddr":"0xec1bbdfffe70b445","nwkAddr":42862,"manufId":4098,"manufName":"_TZE200_i48qyn9s","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\\�\u0004\u0000\u0011�\u0010\u0005\u0000\u0011","65506":27,"modelId":"TS0601","manufacturerName":"_TZE200_i48qyn9s","powerSource":3,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b000be5ad57","endpointID":1}],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1608224290539}`

I made the following changes in devices.js:
`{

       fingerprint: [   

              {
                     modelID: 'TS0601', 
                     manufacturerName: '_TZE200_i48qyn9s' 
              },
       ],
       model: 'TS0601_TRV',
       vendor: 'essentials',
       description: 'Thermostatic radiator valve',
       supports: 'thermostat, temperature',
       fromZigbee: [
              fz.ignore_basic_report,
              fz.tuya_data_point_dump,
       ],
       toZigbee: [
              tz.tuya_data_point_test,
       ],
       onEvent: tuya.setTime,
       meta: {
              configureKey: 1,
       },
       configure: async (device, coordinatorEndpoint) => {
              const endpoint = device.getEndpoint(1);
              await bind(endpoint, coordinatorEndpoint, ['genBasic']);
       },
       exposes: [
              // Here you should put all functionality that your device exposes
       ],
}`

I receive messages in the `tuya.dump.txt` file and at first view these look identical to those from the request #https://github.com/Koenkk/zigbee-herdsman-converters/issues/1803

I also can change tempreture at the thermostat via MQTT
`zigbee2mqtt/friendly_name/set/tuya_data_point_test 'value,16,44'`

And now I don't understand how to continue. What are the next steps?

I understand I have to edit `commen.js` -> `fromZigbee.js` -> `toZigbee.js` -> `devices.js` but in the moment I do not understand what I have to enter there, especially in `fromZigbee.js` and `toZigbee.js`.

Or can I use the same like in the request from serrj-sv and just change the names "moesAltThermostat"?

Thanks a lot in advance, best regards
Sebastian

serrj-sv, 2020-12-30:
> I receive messages in the `tuya.dump.txt` file and at first view these look identical to those from the request #[Koenkk/zigbee-herdsman-converters#1803](https://github.com/Koenkk/zigbee-herdsman-converters/issues/1803)

You don't have to re-invent the wheel then and can try my config I posted in 1st message

userxn10, 2021-02-03:
Would be great to include support for this TRV "Heizkörper-Thermostat ESSENTIALS, Zigbee"
[https://www.amazon.de/gp/product/B08GN52SWS](Amazon Link)

Warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_i48qyn9s'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_i48qyn9s'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

Thank you

insipiens, 2021-03-06:
You should look up the entry for the TS0601 tuya radiator valve and basically copy that into your set up above using/replacing the manufacturer code you want to change. That will probably work. 

if it does then all you need do is raise a PR under the zigbee2mqttconverters section on github to have it added to the devices.js file

closed 2021-04-14 by github-actions[bot]