Koenkk/zigbee2mqtt #5334: GL-C-008-1ID interview successful but device cannot be used

Pollegazon, 2020-12-17:
## What happened
After performing the reset procedure on my Gledopto Zigbee LED controller RGB + CCT (1 ID) (GL-C-008-1ID), the device is visible in the log file:
```
Zigbee2MQTT:info  2020-12-17 21:04:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b002114bf84"},"type":"device_announced"}'Zigbee2MQTT:debug 2020-12-17 21:04:38: Received Zigbee message from '0x00124b002114bf84', type 'readResponse', cluster 'genBasic', data '{"modelId":"GL-C-008"}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2020-12-17 21:04:38: Received Zigbee message from '0x00124b002114bf84', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"GLEDOPTO"}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2020-12-17 21:04:39: Received Zigbee message from '0x00124b002114bf84', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2020-12-17 21:04:39: Received Zigbee message from '0x00124b002114bf84', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2020-12-17 21:04:39: Received Zigbee message from '0x00124b002114bf84', type 'readResponse', cluster 'genBasic', data '{"appVersion":1}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2020-12-17 21:04:39: Received Zigbee message from '0x00124b002114bf84', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2020-12-17 21:04:39: Received Zigbee message from '0x00124b002114bf84', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2020-12-17 21:04:39: Received Zigbee message from '0x00124b002114bf84', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20131206"}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2020-12-17 21:04:39: Received Zigbee message from '0x00124b002114bf84', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"3.0.0"}' from endpoint 11 with groupID 0
```
Zigbee2mqtt log files confirm the interview was successful: 

```Zigbee2MQTT:info  2020-12-17 21:04:40: Successfully interviewed '0x00124b002114bf84', device has successfully been paired
Zigbee2MQTT:info  2020-12-17 21:04:40: Device '0x00124b002114bf84' is supported, identified as: Gledopto Zigbee LED controller RGB + CCT (1 ID) (GL-C-008-1ID)
Zigbee2MQTT:info  2020-12-17 21:04:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Zigbee LED controller RGB + CCT (1 ID)","friendly_name":"0x00124b002114bf84","model":"GL-C-008-1ID","supported":true,"vendor":"Gledopto"},"type":"pairing"}'
```

The device seems to get stuck in "pairing" mode and never becomes available as a topic on MQTT.

When powering off and on the device it shows again as "annouced"

## What did you expect to happen
The device should complet its setup and be available as a topic on MQTT ready to perform actions.
## How to reproduce it (minimal and precise)
Pair a Gledopto Zigbee LED controller RGB + CCT (1 ID) (GL-C-008-1ID).
I'm using a pi 4
## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC26X2R1
Adapter firmware version: CC26X2R1_20201026
```
{"commit":"0514204","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20201026,"transportrev":2},"type":"zStack3x0"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6755},"permit_join":true,"version":"1.16.2"}
```

Koenkk, 2020-12-18:
> The device seems to get stuck in "pairing" mode and never becomes available as a topic on MQTT.

What makes you think this?

Have you tried controlling the device? (e.g. change state via MQTT: https://www.zigbee2mqtt.io/devices/GL-C-008-1ID.html#light)

Pollegazon, 2020-12-18:
@Koenkk Thanks for your reply. Indeed, although the topic is not visible I'm able to send commands. I was misled by the fact that all my other devices publish something after pairing resulting in adding their address to the list op topic. This device is silent until you ask it to do something. Great that is good news. I'll close this ticket. Thanks for your help.

closed 2020-12-18 by Pollegazon