Koenkk/zigbee2mqtt #5335: Heiman COsensor-EF-3.0 unsupported

patrickarkenbout, 2020-12-17:
Hi,

I use more heiman devices without a problem, with my last order i ordered 1 COsensor without looking in the list of support devices. Now i turn out that this devices is not supported

Device '0xbc33acfffe01087e' with Zigbee model 'COSensor-EF-3.0' and manufacturer name 'HEIMAN' is NOT supported

## Information about the device + link
According the site it is a model M414

https://www.aliexpress.com/item/4000209418201.html?spm=a2g0o.productlist.0.0.30f23bf3fNxbm2&algo_pvid=d0b6bf95-e364-4c04-950e-9845f6a7a770&algo_expid=d0b6bf95-e364-4c04-950e-9845f6a7a770-0&btsid=2100bdca16082402888164765ecb88&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

sjorge, 2020-12-19:
Try updating this block in devices.js:

```
    {
        zigbeeModel: ['COSensor-EM', 'COSensor-N'],
        model: 'HS1CA-E',
        vendor: 'HEIMAN',
        description: 'Smart carbon monoxide sensor',
        fromZigbee: [fz.ias_carbon_monoxide_alarm_1, fz.battery],
        toZigbee: [],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
        exposes: [e.carbon_monoxide(), e.battery_low(), e.tamper(), e.battery()],
    },
```

To the following

```
    {
        zigbeeModel: ['COSensor-EM', 'COSensor-N', 'COSensor-EF-3.0'],
        model: 'HS1CA-E',
        vendor: 'HEIMAN',
        description: 'Smart carbon monoxide sensor',
        fromZigbee: [fz.ias_carbon_monoxide_alarm_1, fz.battery],
        toZigbee: [],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
        exposes: [e.carbon_monoxide(), e.battery_low(), e.tamper(), e.battery()],
    },
```

patrickarkenbout, 2020-12-21:
It took me a while to figure out how to edit devices.js. The above suggestion did not seem to work. Finally I copied and modified the above config to and add this to devices.js
{
         zigbeeModel: ['COSensor-EF-3.0'],
         model: 'M414',
         vendor: 'HEIMAN',
         description: 'Smart carbon monoxide sensor',
         fromZigbee: [fz.ias_carbon_monoxide_alarm_1, fz.battery],
         toZigbee: [],
         meta: {configureKey: 1},
         configure: async (device, coordinatorEndpoint) => {
             const endpoint = device.getEndpoint (1);
             await bind (endpoint, coordinatorEndpoint, ['genPowerCfg']);
             await configureReporting.batteryPercentageRemaining (endpoint);
         },
         exposes: [e.carbon_monoxide (), e.battery_low (), e.tamper (), e.battery ()],
     },
This turned out to work.

sjorge, 2020-12-21:
Odd, as there doesn't seem to be any difference in fromZigbee and configure step between the one I suggested and the one that ended up working.

To get this in the next release you can do a PR against https://github.com/Koenkk/zigbee-herdsman-converters to get it added. Or wait, koenkk usually picks them up form here when he has time.

patrickarkenbout, 2020-12-21:
I will try again, maybe i made a mistake.

sjorge, 2020-12-21:
> I will try again, maybe i made a mistake.

The only difference is that the one I listed was grouped with 2 other Heiman CO sensors that had a similar model name:
`COSensor-EM`, `COSensor-N`. 

But if it is your first time messing with devices.js, things are indeed confusing :)

patrickarkenbout, 2020-12-21:
Now it works, I made a typo or something else. It worked with your suggestion. Thank you for your help

closed 2020-12-21 by Koenkk

szaman-89, 2021-01-20:
@Koenkk Did you include this fix in current version of z2m? I paired this device with zigbee2mqtt but still "Unsupported".

Koenkk, 2021-01-20:
@szaman-89 it is in 1.17.0, if you are on this version share your data/database.db entry of this device

szaman-89, 2021-01-20:
> @szaman-89 it is in 1.17.0, if you are on this version share your data/database.db entry of this device

Entry of this device:

{"id":9,"type":"Unknown","ieeeAddr":"0x000d6f0014f90f70","nwkAddr":22552,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1611167575864}

Koenkk, 2021-01-20:
@szaman-89 the interview failed, force remove first through https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgerequestdeviceremove and pair again (https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails)

szaman-89, 2021-01-20:
Sensor can be paired only with combining Touchlink scan. In permit join option only z2m cant get any pairing information. After launching Touchlink scan device can be paired but became unspported.

sjorge, 2021-01-20:
It paired pretty easy for me:
1. enable pairing
2. pull the battery pull tab
3. hold the button to enter pairing mode

It showed up a few seconds later, like maybe 10 or so.

szaman-89, 2021-01-20:
> It paired pretty easy for me:
> 
> 1. enable pairing
> 2. pull the battery pull tab
> 3. hold the button to enter pairing mode
> 
> It showed up a few seconds later, like maybe 10 or so.
> 
> [image]

I did exactly how you described but z2m won't find any device. Last new device was added couple days ago and everything works like a charm.

jorismathijssen, 2021-01-29:
Did not want to make a new issue for just a question, maybe @szaman-89 or @sjorge can answer it. 

My CO sensor will not report any information. When I press the test button a payload message is send but no battery level is included.

MQTT publish: topic 'zigbee2mqtt/0x000d6f00156570ce', payload '{"battery_low":false,"carbon_monoxide":true,"linkquality":57,"tamper":false}'

[image]

Any suggestions?

szaman-89, 2021-01-29:
> Any suggestions?

When did you pair co sensor? Maybe you should wait for this informations. If you did it long time ago I would try to re-pair it.

jorismathijssen, 2021-01-29:
> > Any suggestions?
> 
> When did you pair co sensor? Maybe you should wait for this informations. If you did it long time ago I would try to re-pair it.

I just re-paired the device but last time i have waited 2 weeks :) 

Maybe this repair fixed it, lets wait a couple of hours/days