Koenkk/zigbee2mqtt #5336: Innr SP120 energy wrong by factor 100

WeeGee77, 2020-12-18:
My two Innr SP120 plugs work fine, almost all reported values are as expected, but the “energy” value is wrong by the factor 100. 
The documentation https://www.zigbee2mqtt.io/devices/SP_120.html says, the number is the energy in kWh, but the actual used energy is 100 times lower. 
I calculated the difference of the values from now (101) and 24h before (67) of almost constant power usage, and the difference is 34 - my room would be a lot warmer if the connected LED light actually wasted 34kWh ;-) Based on the measured power the correct energy value would be 340Wh. 
My second SP120 shows the same error.

Koenkk, 2020-12-18:
@WeeGee77 is the reported `power` correct? (should be in W).

WeeGee77, 2020-12-18:
Yes. My Blitzwolf plugs report the same power value +/- 2W as the SP120s. 
And the Blitzwolf report reasonable energy values (but I didn’t check them exactly).

Koenkk, 2020-12-20:
Can you update the devices.js definition to below and check if the `power` (is still) and `energy` is correct now?

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

```js
{
    zigbeeModel: ['SP 120'],
    model: 'SP 120',
    vendor: 'Innr',
    description: 'Smart plug',
    fromZigbee: [fz.electrical_measurement, fz.on_off, fz.ignore_genLevelCtrl_report, fz.metering],
    toZigbee: [tz.on_off],
    meta: {configureKey: 6},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await configureReporting.onOff(endpoint);
        // Gives UNSUPPORTED_ATTRIBUTE on readEletricalMeasurementPowerConverterAttributes.
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
            acCurrentDivisor: 1000,
            acCurrentMultiplier: 1,
        });
        await configureReporting.activePower(endpoint);
        await configureReporting.rmsCurrent(endpoint);
        await configureReporting.rmsVoltage(endpoint);
        // Gives UNSUPPORTED_ATTRIBUTE on readMeteringPowerConverterAttributes.
        endpoint.saveClusterAttributeKeyValue('seMetering', {
            multiplier: 1,
            divisor: 100,
        });
        await configureReporting.currentSummDelivered(endpoint);
    },
    exposes: [e.power(), e.current(), e.voltage(), e.switch(), e.energy()],
},
```

WeeGee77, 2020-12-21:
Even with today's (Dockers latest-dev) of zigbee2mqtt the divisor seem
ed to be ignored.
After some hours of  experimenting and reading into the converters and
almost despairing, I found out that the "energy" value is only updated once
each hour or even longer. Maybe only when the measured value in the plug
actually changes. And as long as this value doesn't change, all editing and
restarting and even manipulating the converters changes nothing...

tl;dr: Changing the divisor to 100 gives the expected result and doesn't
affect other values. If you wait long enough...

notifications@github.com <github.weegee.8533362379.notifications#
reply@reply.github.com>:

Koenkk, 2020-12-21:
Thanks, added the changes.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-21 by Koenkk