Koenkk/zigbee2mqtt #5338: Zigbee2MQTT crashed with a JS error handling issue

esbenr, 2020-12-18:
## What happened
After updating to latest version yesterday, Z2M crashed from no reason to night. 
It looks like missing error handling.

## What did you expect to happen
Z2M not to crash...

## How to reproduce it (minimal and precise)
Don't know. Upgrade to latest version?

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC26X2R1
Adapter firmware version: CC26X2R1_20200925

## Log
I can get this log if I open the HA add-on and select log. But the log files in /share/... don't show this error.
So maybe it's in the add-on?

´´´
at Timeout._onTimeout (/zigbee2mqtt-1.16.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:248) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 138)
(node:248) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.16.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:248) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 139)
(node:248) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.16.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:248) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 140)
(node:248) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.16.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:248) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 141)
(node:248) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.16.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:248) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 142)
(node:248) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.16.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:248) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 143)
(node:248) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.16.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:248) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 144)
(node:248) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.16.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:248) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 145)
(node:248) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.16.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:248) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 146)
(node:248) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.16.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:248) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 147)
(node:248) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
[... 26 lines cut ...]
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:248) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 151)
(node:248) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
[... cut ...]
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:248) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 154)
´´´

esbenr, 2020-12-18:
When restarting Z2M it fails with unable to connect to the adapter.
Unplugging and re-plugging and then restarting Z2M fixes it.

This seem like the same instability I have reported numerous times where Z2M is loosing the connection to the ZZH dongle.

Can you please advise me how to debug the ZZH connection in order to verify that?

Koenkk, 2020-12-18:
Can you provide me the herdsman debug log from starting z2m till this error appears?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

esbenr, 2020-12-19:
Thanks, I have enabled herdsman logging. Will post here with the full log.

esbenr, 2020-12-20:
So. Here is the log from the crash today. It seems to come out of the blue.

At:
```debug 2020-12-20 13:23:08: Publishing 'set' 'state' to 'Carport Wall light shed door right'```
Response:
```
info  2020-12-20 13:23:08: MQTT publish: topic 'zigbee2mqtt/Carport Wall light shed door right', payload '{"brightness":26,"linkquality":72,"state":"OFF","update":{"state":"available"},"update_available":true}'
```

A bit later:
```debug 2020-12-20 13:24:33: Publishing 'set' 'state' to 'Carport Wall light shed door left'```
Fails with:
```
error 2020-12-20 13:24:39: Publish 'set' 'state' to 'Carport Wall light shed door right' failed: 'Error: Command 0xbc33acfffea36824/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SRSP - AF - dataRequest after 6000ms)'
debug 2020-12-20 13:24:39: Error: Command 0xbc33acfffea36824/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SRSP - AF - dataRequest after 6000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.16.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
info  2020-12-20 13:24:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Carport Wall light shed door right' failed: 'Error: Command 0xbc33acfffea36824/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"Carport Wall light shed door right"},"type":"zigbee_publish_error"}'
error 2020-12-20 13:24:43: Publish 'set' 'state' to 'Carport Wall light entrance door left' failed: 'Error: Command 0x086bd7fffe088de1/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'Timeout' (9999))'
debug 2020-12-20 13:24:43: Error: Command 0x086bd7fffe088de1/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'Timeout' (9999))
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.16.2/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:339:27)
    at Generator.next (<anonymous>)
    at fulfilled (/zigbee2mqtt-1.16.2/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:24:58)
```

And then from there everything stats to fail, so it's no longer tied to the specific bulb.

Full log is here: https://gist.github.com/esbenr/f45e2038154cc5debfd765da9a7b3e84

esbenr, 2020-12-20:
Trying to restart Z2M just fails with "cant find the ZZH adapter" as usual. 
Here is the log with debug enabled:
```
2020-12-20T14:27:27: PM2 log: Launching in no daemon mode
2020-12-20T14:27:27: PM2 log: App [npm:0] starting in -fork mode-
2020-12-20T14:27:27: PM2 log: App [npm:0] online
> zigbee2mqtt@1.16.2 start /zigbee2mqtt-1.16.2
> node index.js
Zigbee2MQTT:debug 2020-12-20 14:27:31: Loaded state from file /share/zigbee2mqtt/state.json
Zigbee2MQTT:info  2020-12-20 14:27:31: Logging to console and directory: '/share/zigbee2mqtt/log/2020-12-20.14-27-30' filename: log.txt
Zigbee2MQTT:debug 2020-12-20 14:27:31: Removing old log directory '/share/zigbee2mqtt/log/2020-12-16.17-16-35'
Zigbee2MQTT:info  2020-12-20 14:27:31: Starting Zigbee2MQTT version 1.16.2 (commit #1.16.2)
Zigbee2MQTT:info  2020-12-20 14:27:31: Starting zigbee-herdsman (0.13.37)
Zigbee2MQTT:debug 2020-12-20 14:27:31: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/share/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","databasePath":"/share/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0","rtscts":false}}'
2020-12-20T13:27:31.884Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2020-12-20T13:27:31.888Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6755,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"rtscts":false,"path":"/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","adapter":{"concurrent":null,"delay":null}}'
2020-12-20T13:27:31.901Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2020-12-20T13:27:31.914Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2020-12-20T13:27:32.381Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload
2020-12-20T13:27:32.383Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2020-12-20T13:27:33.396Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-12-20T13:27:33.401Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2020-12-20T13:27:39.412Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-12-20T13:27:39.415Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2020-12-20T13:27:45.424Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-12-20T13:27:45.426Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
Zigbee2MQTT:error 2020-12-20 14:27:51: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-12-20 14:27:51: Failed to start zigbee
Zigbee2MQTT:error 2020-12-20 14:27:51: Exiting...
Zigbee2MQTT:error 2020-12-20 14:27:51: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.16.2/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.16.2/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm
 ERR! code ELIFECYCLE
npm ERR! 
errno 1
npm 
ERR! zigbee2mqtt@1.16.2 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.16.2 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-12-20T13_27_51_487Z-debug.log
2020-12-20T14:27:52: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-12-20T14:27:52: PM2 log: App [npm:0] starting in -fork mode-
2020-12-20T14:27:52: PM2 log: App [npm:0] online
> zigbee2mqtt@1.16.2 start /zigbee2mqtt-1.16.2
> node index.js
Zigbee2MQTT:debug 2020-12-20 14:27:55: Loaded state from file /share/zigbee2mqtt/state.json
Zigbee2MQTT:info  2020-12-20 14:27:55: Logging to console and directory: '/share/zigbee2mqtt/log/2020-12-20.14-27-54' filename: log.txt
Zigbee2MQTT:debug 2020-12-20 14:27:55: Removing old log directory '/share/zigbee2mqtt/log/2020-12-16.17-19-06'
Zigbee2MQTT:info  2020-12-20 14:27:55: Starting Zigbee2MQTT version 1.16.2 (commit #1.16.2)
Zigbee2MQTT:info  2020-12-20 14:27:55: Starting zigbee-herdsman (0.13.37)
Zigbee2MQTT:debug 2020-12-20 14:27:55: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/share/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","databasePath":"/share/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0","rtscts":false}}'
2020-12-20T13:27:56.282Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
[... cut ...]
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-12-20T13_28_15_879Z-debug.log
2020-12-20T14:28:16: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-12-20T14:28:16: PM2 log: App [npm:0] starting in -fork mode-
2020-12-20T14:28:16: PM2 log: App [npm:0] online
```

And it just keep looping...

Unplugging and re--plugging the ZZH-stick will fix this.

Koenkk, 2020-12-20:
This is not the herdsman debug log, it's logged to the stderr, not stoudout or log files.

esbenr, 2020-12-20:
Sorry. I missed this line: Important: this is not logged to the log files and is only available on the STDOUT. ;-)

How do I get the STDOUT?

Koenkk, 2020-12-21:
How are you running z2m?

esbenr, 2020-12-21:
As add-on in Home Assistant.
I have enabled it as per the instructions. I just don't know where to find it.

Koenkk, 2020-12-21:
I see, are you able to do a `docker logs -f [ZIGBEE2MQTT_CONTAINER_NAME]` when logging in through ssh?

closed 2021-01-21 by Koenkk

Zeddy-, 2021-02-02:
Also got the Slaesh, and this keeps happening after running for a while, for me just restarting Z2M helps, no need to reboot raspberry. seems to always start with /bridge/log topic, then it crashes with

 **zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq** 

Zigbee2MQTT:info  2021-02-02 12:10:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":[{"dateCode":"20210120","friendly_name":"Coordinator","ieeeAddr":"0x00124b0021cc3f07","lastSeen":161226065                                                4166,"networkAddress":0,"softwareBuildID":"zStack3x0","type":"Coordinator"},{"dateCode":"20200416","description":"Ally thermostat","friendly_name":"eTRV0100-24","hardwareVersion":69,"ieeeAddr":"0xccccccfffe7c923                                                1","lastSeen":1612260577024,"manufacturerID":4678,"manufacturerName":"Danfoss","model":"014G2461","modelID":"eTRV0100","networkAddress":45562,"powerSource":"Battery","softwareBuildID":"01.02.0008 01.02","type":"                                                EndDevice","vendor":"Danfoss"},{"dateCode":"09-06-2019","description":"Mi power plug ZigBee EU","friendly_name":"lumi.plug.mmeu01-1","hardwareVersion":1,"ieeeAddr":"0x04cf8cdf3c759a75","lastSeen":1612260362337,"                                                manufacturerID":4447,"manufacturerName":"LUMI","model":"ZNCZ04LM","modelID":"lumi.plug.mmeu01","networkAddress":32847,"powerSource":"Mains (single phase)","type":"Router","vendor":"Xiaomi"},{"dateCode":"20200902                                                ","description":"Ally thermostat","friendly_name":"eTRV0100-30","hardwareVersion":69,"ieeeAddr":"0x14b457fffed56132","lastSeen":1612260530433,"manufacturerID":4678,"manufacturerName":"Danfoss","model":"014G2461"                                                ,"modelID":"eTRV0100","networkAddress":40026,"powerSource":"Battery","softwareBuildID":"01.02.0008 01.02","type":"EndDevice","vendor":"Danfoss"},{"description":"-","friendly_name":"eTRV0100-66","ieeeAddr":"0x14b                                                457fffed551b1","lastSeen":1612125915603,"manufacturerID":4678,"networkAddress":2838,"type":"EndDevice","vendor":"-"},{"dateCode":"20200902","description":"Ally thermostat","friendly_name":"eTRV0100-42","hardware                                                Version":69,"ieeeAddr":"0x14b457fffed63621","lastSeen":1612260610322,"manufacturerID":4678,"manufacturerName":"Danfoss","model":"014G2461","modelID":"eTRV0100","networkAddress":15929,"powerSource":"Battery","sof                                                twareBuildID":"01.02.0008 01.02","type":"EndDevice","vendor":"Danfoss"},{"dateCode":"20200902","description":"Ally thermostat","friendly_name":"eTRV0100-39","hardwareVersion":69,"ieeeAddr":"0x14b457fffed56079","                                                lastSeen":1612260606395,"manufacturerID":4678,"manufacturerName":"Danfoss","model":"014G2461","modelID":"eTRV0100","networkAddress":47814,"powerSource":"Battery","softwareBuildID":"01.02.0008 01.02","type":"EndD                                                evice","vendor":"Danfoss"},{"dateCode":"09-06-2019","description":"Mi power plug ZigBee EU","friendly_name":"lumi.plug.mmeu01-2","hardwareVersion":1,"ieeeAddr":"0x04cf8cdf3c75a576","lastSeen":1612260488712,"manu                                                facturerID":4447,"manufacturerName":"LUMI","model":"ZNCZ04LM","modelID":"lumi.plug.mmeu01","networkAddress":3025,"powerSource":"Mains (single phase)","type":"Router","vendor":"Xiaomi"},{"dateCode":"20200902","de                                                scription":"Ally thermostat","friendly_name":"eTRV0100-43","hardwareVersion":69,"ieeeAddr":"0x14b457fffed551a9","lastSeen":1612260511619,"manufacturerID":4678,"manufacturerName":"Danfoss","model":"014G2461","mod                                                elID":"eTRV0100","networkAddress":11858,"powerSource":"Battery","softwareBuildID":"01.02.0008 01.02","type":"EndDevice","vendor":"Danfoss"},{"dateCode":"20200902","description":"Ally thermostat","friendly_name":                                                "eTRV0100-44","hardwareVersion":69,"ieeeAddr":"0x14b457fffed3e3d8","lastSeen":1612260641765,"manufacturerID":4678,"manufacturerName":"Danfoss","model":"014G2461","modelID":"eTRV0100","networkAddress":42194,"powe                                                rSource":"Battery","softwareBuildID":"01.02.0008 01.02","type":"EndDevice","vendor":"Danfoss"},{"dateCode":"09-06-2019","description":"Mi power plug ZigBee EU","friendly_name":"lumi.plug.mmeu01-3","hardwareVersi                                                on":1,"ieeeAddr":"0x04cf8cdf3c75a443","lastSeen":1611928043946,"manufacturerID":4447,"manufacturerName":"LUMI","model":"ZNCZ04LM","modelID":"lumi.plug.mmeu01","networkAddress":63522,"powerSource":"Mains (single                                                 phase)","type":"Router","vendor":"Xiaomi"},{"dateCode":"20200902","description":"Ally thermostat","friendly_name":"eTRV0100-36","hardwareVersion":69,"ieeeAddr":"0x14b457fffed365d9","lastSeen":1612260529437,"manu                                                facturerID":4678,"manufacturerName":"Danfoss","model":"014G2461","modelID":"eTRV0100","networkAddress":26019,"powerSource":"Battery","softwareBuildID":"01.02.0008 01.02","type":"EndDevice","vendor":"Danfoss"},{"                                                dateCode":"20200902","description":"Ally  [... cut ...]
(node:26866) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not han                                                dled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection                                                 id: 2)

Zeddy-, 2021-02-02:
Also when i stop Z2M i get this: 

^CZigbee2MQTT:debug 2021-02-02 13:45:49: Saving state to file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2021-02-02 13:45:49: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
Zigbee2MQTT:info  2021-02-02 13:45:49: Disconnecting from MQTT server
^[[A  zigbee-herdsman:controller:log Disable joining +2h
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":234,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,106,2,11,0,0]}} +2m
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,234,0,30,6,0,25,106,2,11,0,0,149] +2m
Zigbee2MQTT:error 2021-02-02 13:45:55: Failed to stop zigbee