Koenkk/zigbee2mqtt #5340: danfoss ally problems

zhapwey, 2020-12-18:
## What happened
I've tried integrating danffos ally trv, thanks for the work. It has already been discussed in other posts but has not been given a solution.The system for setting the temperature setpoint works in a special way, And since this on z2m it doesn't work.
## What did you expect to happen
Mounting the valve on a radiator sets the setpoint and will take a long time to respond, it can take hours to change the setpoint. However, if you move it by hand from the TRV it changes instantly.
## How to reproduce it (minimal and precise)
change setpoint con {"occupied_heating_setpoint":21}

Danffos differentiates between how the setpoint needs to change through the cluster thermostat command 0x40, to make a normal change or a change in the time it takes to answer a lot, without this it is impossible to control the valve correctly.
I have a little more info about this command if anyone has any idea how it works.

## Debug info
Zigbee2MQTT version: 1.16.2
Adapter hardware: CC26X2R1
Adapter firmware version:

closed 2021-01-25 by github-actions[bot]

barthelemyguillaume, 2021-02-03:
Same problem here!

wormiedk, 2021-02-14:
It works better in zha

zhapwey, 2021-02-14:
What is zha?

wormiedk, 2021-02-22:
zigbee home automation which is integrated in home assistant

robertalexa, 2021-05-22:
This has now been resolved alongside a significant amount of improvements to this TRV.

The changes are now available on the dev branch of z2m