Koenkk/zigbee2mqtt #5345: Failed to connect to the adapter - Ubuntu-Mate on pi-4

mingynaut, 2020-12-19:
I am running ubuntu-mate on a raspberry-pi 4. I have done the 'exact' install steps defined for getting the CC2531 running. 
I included one log at the end so as not to add to the clutter below: 

I have done full shutdown's on the system between changes. 
Also, I am running using DEBUG=zigbee-herdsman* npm start  

I did a successful install (as noted) and edited the configuration.yaml and changed the entry for port to /dev/ttyAMA0 to reflect where the serial port is mapped under ubuntu. 

npm --version
6.14.8
pisys1@pisys1-34:~$ node --version
v12.20.0

[LAUNCH FAILED]

I ran the test:
test -w /dev/ttyAMA0 && echo success || echo failure --> failure
sudo test -w /dev/ttyAMA0 && echo success || echo failure --> success
So for each run and post reboot I did the chown dance
sudo chown pisys1 /dev/ttyAMA0

sudo ls -la /dev/ttyAMA0
crw-rw---- 1 root dialout 204, 64 Dec 17 18:50 /dev/ttyAMA0

sudo chown pisys1 /dev/ttyAMA0

sudo ls -la /dev/ttyAMA0
crw-rw---- 1 pisys1 dialout 204, 64 Dec 17 18:50 /dev/ttyAMA0

[LAUNCH FAILED]

I also tried adding a new udev rule as defined in the 99-usb.serial.rules. 
For that attempt I did change the idProduct as follows:
SUBSYSTEM=="tty", ATTRS{idVendor}=="0451", ATTRS{idProduct}=="16ae", ATTRS{serial}=="fe201000.serial", SYMLINK="ttyUSB.CC2531-01", OWNER="pi"

[LAUNCH FAILED]
Log exceprt shows this:
 zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyUSB.CC2531-01 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
Zigbee2MQTT:error 2020-12-18 16:01:19: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-12-18 16:01:19: Failed to start zigbee
Zigbee2MQTT:error 2020-12-18 16:01:19: Exiting...
Zigbee2MQTT:error 2020-12-18 16:01:19: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyUSB.CC2531-01'

So I don't think the udev rule fired. I did try listing the udev rules and there were no issues with syntax for the 99- rule 

The reason for the idProduct change was after running 'lsusb -v' I saw this... 

Bus 001 Device 003: ID 0451:16ae Texas Instruments, Inc. CC2531 Dongle
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        32
  idVendor           0x0451 Texas Instruments, Inc.
  idProduct          **0x16ae** CC2531 Dongle

In addition, I thought this info might be helpful?

udevadm info --name /dev/ttyAMA0 --query all
P: /devices/platform/soc/fe201000.serial/tty/ttyAMA0
N: ttyAMA0
L: 0
S: serial1
E: DEVPATH=/devices/platform/soc/fe201000.serial/tty/ttyAMA0
E: DEVNAME=/dev/ttyAMA0
E: MAJOR=204
E: MINOR=64
E: SUBSYSTEM=tty
E: USEC_INITIALIZED=2237105
E: ID_MM_CANDIDATE=1
E: DEVLINKS=/dev/serial1
E: TAGS=:systemd:

So, after reading many of the posts here (hours invested), I am unclear as to how to proceed?

This is basically bare metal linux running on a pi-4 board (not raspbian OS) but ubuntu-mate.
There are no other configurations running on this system like openhab or home assitant, it is pure stock ubuntu. There are no special udev rules or scripts used, this is as close to bare metal as I know. 

What do you suggest? 

============================================================
LOG: 
DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.16.2 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:debug 2020-12-18 16:38:27: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2020-12-18 16:38:27: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-12-18.16-38-26' filename: log.txt
Zigbee2MQTT:info  2020-12-18 16:38:27: Starting Zigbee2MQTT version 1.16.2 (commit #0514204)
Zigbee2MQTT:info  2020-12-18 16:38:27: Starting zigbee-herdsman (0.13.37)
Zigbee2MQTT:debug 2020-12-18 16:38:27: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyAMA0"}}'
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyAMA0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyAMA0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +7ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +455ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
Zigbee2MQTT:error 2020-12-18 16:38:48: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-12-18 16:38:48: Failed to start zigbee
Zigbee2MQTT:error 2020-12-18 16:38:48: Exiting...
Zigbee2MQTT:error 2020-12-18 16:38:48: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.16.2 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.16.2 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pisys1/.npm/_logs/2020-12-19T00_38_48_842Z-debug.log

Smiggel, 2020-12-19:
Same happens on Raspberry Pi with HomeAssistant on Home Assistant OS 5.8. When I unplug my CC2351 stick and put it back in, it works.

mingynaut, 2020-12-19:
Thanks, I have tried the unplug/plug approach and no joy for each of the four USB ports.

Smiggel, 2020-12-20:
I was able to fix the issue with the "ha su repair" command in the terminal.

mingynaut, 2020-12-21:
From my above post, I should see a different tty entry after plugging in the CC2531. That 'new' path is what I should be able to add to the configuration.yaml file. The new path is not showing up after the dongle is plugged in... 

For example when plugging in the CC2531 dongle one sees this from dmesg:

[10716.492316] usb 1-1.2: USB disconnect, device number 9
[10774.223275] usb 1-1.3: new full-speed USB device number 10 using xhci_hcd
[10774.330292] usb 1-1.3: New USB device found, idVendor=0451, idProduct=16ae, bcdDevice=39.62
[10774.330308] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[10774.330318] usb 1-1.3: Product: CC2531 USB Dongle
[10774.330329] usb 1-1.3: Manufacturer: Texas Instruments

Notice that is it! no other entry... where as if I plug in a mouse I get additional information and a path is set up.

[ 8484.416709] usb 1-1.3: new full-speed USB device number 8 using xhci_hcd
[ 8484.523920] usb 1-1.3: New USB device found, idVendor=0451, idProduct=16ae, bcdDevice=39.62
[ 8484.523936] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 8484.523947] usb 1-1.3: Product: CC2531 USB Dongle
[ 8484.523957] usb 1-1.3: Manufacturer: Texas Instruments
[ 9148.217336] usb 1-1.2: new low-speed USB device number 9 using xhci_hcd
[ 9148.324957] usb 1-1.2: New USB device found, idVendor=093a, idProduct=2510, bcdDevice= 1.00
[ 9148.324973] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 9148.324984] usb 1-1.2: Product: USB Optical Mouse
[ 9148.324993] usb 1-1.2: Manufacturer: PixArt
[ 9148.333953] input: PixArt USB Optical Mouse as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2:1.0/0003:093A:2510.0002/input/input1

On a prior plug in of the dongle where it was assigned device number 6 I did see the following:

ll /dev/bus/usb/001/006
crw-rw-r-- 1 root root 189, 5 Dec 21 12:00 /dev/bus/usb/001/006

ll /sys/bus/usb/devices/

... 
lrwxrwxrwx 1 root root 0 Apr  1  2020 1-1 -> ../../../devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/

Maybe /dev/bus/usb/001/006 is the path to use in configuration.yaml?

mingynaut, 2020-12-22:
One more thing: 
The driver must NOT be loading as this appears to confirm:

usb-devices
T:  Bus=01 Lev=02 Prnt=02 Port=01 Cnt=01 Dev#= 12 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=32 #Cfgs=  1
P:  Vendor=0451 ProdID=16ae Rev=39.62
S:  Manufacturer=Texas Instruments
S:  Product=CC2531 USB Dongle
C:  #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=100mA
I:  If#=0x0 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)

adi-dev, 2021-01-12:
It seams that I have the same issue on my desktop Ubuntu 20 x64. The USB is there somewhere as `dmesg` shows is but no serial ports nor `TTYAMA0` in `/dev/`.
When I plug it into RPI, it is recognized alright.

adi-dev, 2021-01-13:
> ## It seams that I have the same issue on my desktop Ubuntu 20 x64. The USB is there somewhere as `dmesg` shows is but no serial ports nor `TTYAMA0` in `/dev/`.
> When I plug it into RPI, it is recognized alright.
> Edit:
> I thought RPI recognizes it but this was my silly mistake, the `ttyAMA0` is not for this device, and actually, I have the same problem what I have on desktop.

I just found that, regardless my seller promised to flash it with coordinator firmware, it was not. Flashing the adapter with correct firmware solved my problem, `/dev/ttyACM0` appeared on the list.

mingynaut, 2021-02-13:
The final resolution... The CC2531 USB dongle had a flash error. The provider was gracious and sent another out for free. 
After using a re-flashed CC2531 dongle, there are no more issues. As above, dev/ttyACM0 now appears as a legitimate device.

closed 2021-02-13 by mingynaut

DuckSounds, 2023-05-05:
> I just found that, regardless my seller promised to flash it with coordinator firmware, it was not. Flashing the adapter with correct firmware solved my problem, `/dev/ttyACM0` appeared on the list.

I am sorry to be so late but what firmware was used? I am currently experiencing the same issue.