Koenkk/zigbee2mqtt #5346: MQTT Integration Broken

AshtonKem, 2020-12-19:
## What happened

After a HA upgrade, the integration between Zigbee2mqtt and HA is broken. The following things cannot be accomplished:

1. Device discovery
2. Controlling devices from HA
3. Propagation of state from Zigbee2mqtt to HA (sensors, etc.)

I can still pair devices with Zigbee2mqtt, and I can control them from the Zigbee2mqtt web UI. The logs indicate that it is publishing to MQTT, but apparently the MQTT integration is not receiving/parsing them properly. 

## What did you expect to happen

Regular behavior.

## How to reproduce it (minimal and precise)

Installing a new version of HA with the latest version of Zigbee2mqtt and the MQTT add on appears to be sufficient.

## Debug info
Zigbee2MQTT version: 1.16.2.2
HA version(s): 0.118.5, 2020.12.7
Adapter hardware: CC2531
Adapter firmware version: ??

AshtonKem, 2020-12-19:
Perhaps the issue is in how HA discovery now works? Here are the relevant log sections from the MQTT component. I force deleted and successfully re-paired a Xiaomi motion sensor during this time frame, and nothing showed up in the logs.

[image]

paul12ster, 2020-12-20:
I have the same issue.  I have been tinkering with home assistant and had everything working. Today i was ready to make my final setup and i started with a full new installation. (new intel NUC, proxmox, HA). 
MQTT broker is working, MQTT explorer sees the zigbee messages. (discovery is set to on) Zigbee2mqtt addon is working. devices are recognized and correctly interviewed. But no devices are added to HA and the MQTT add on doesn't show any found devices. (the mqtt add-on reconfigure interface is showing a good connection and discovery tick-box is enabled)

I did start over from scratch two times with the same result.

Koenkk, 2020-12-20:
Looks similar to https://github.com/Koenkk/zigbee2mqtt/issues/5150 , @emontnemery  what would be needed to debug this from the HA perspective?

paul12ster, 2020-12-20:
Like AshtoncKem metnioned, Installing a new version of HA with the latest version of Zigbee2mqtt and the MQTT add on appears to be sufficient.

My system info. Freshly installed on new NUC (proxmox)  yesterday.
Host Operating System | Home Assistant OS 5.8
-- | --
Update Channel | stable
Supervisor Version | 2020.12.7
Docker Version | 19.03.13
Disk Total | 78.0 GB
Disk Used | 4.4 GB
Healthy | true
Supported | true
Board | ova
Supervisor API | ok
Version API | ok
Installed Add-ons | Terminal & SSH (8.10.0), File editor (5.2.0), MariaDB (2.2.1), Samba share (9.3.0), OpenZWave (0.7.1), Zigbee2mqtt (1.16.2.2), ESPHome (1.15.3), Mosquitto broker (5.1)

Zigbee is publisching information:
Zigbee2MQTT:info  2020-12-20 10:13:04: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffe123e31', payload '{"linkquality":75,"occupancy":true,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-12-20 10:13:16: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffe123e31', payload '{"linkquality":63,"occupancy":true,"update":{"state":"idle"},"update_available":false}'

`data_path: /share/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://core-mosquitto:1883'
serial:
  port: /dev/ttyUSB0
advanced:
  log_level: info
`
mosquitto config
logins: []
anonymous: true
customize:
  active: false
  folder: mosquitto
certfile: fullchain.pem
keyfile: privkey.pem
require_certificate: false

[image]

[image]

Integration screen show no new devices/entities

[image]

emontnemery, 2020-12-20:
Please provide a full HA log with these settings:
https://github.com/Koenkk/zigbee2mqtt/issues/5150#issuecomment-740884825

You can drag and drop the full log as zip or txt to a comment in GitHub.

paul12ster, 2020-12-20:
[log_ha_202012201420.txt](https://github.com/Koenkk/zigbee2mqtt/files/5720971/log_ha_202012201420.txt)

paul12ster, 2020-12-21:
Just for testing, i created an additional ubuntu instance with IOStack docker and mosquitto containter. In HA i un-installed the mosquitto addon and added the mqqt: settings in the HA configuration.yaml and pointed the zigbee2mqtt addon to the new mqqt server. 

The devices are now correctly discovered. 

Not the preferred solution but maybe useful information while debugging this issue.

AshtonKem, 2020-12-27:
@paul12ster What version of MQTT did you run locally? Maybe it's different than tho Mosquitto HA addon?

AshtonKem, 2020-12-27:
@Koenkk I was also experiencing #5150 before updating as well.

emontnemery, 2020-12-28:
In the log from @paul12ster, HA successfully connects to the broker and subscribes to topics, but it does not receive any MQTT messages. @paul12ster Are there any logs from the non working mosquitto addon? Can you send and receive messages from other clients, e.g. command line?

@AshtonKem Can you please provide a log as well? It's impossible to debug this from a HA perspective otherwise.

paul12ster, 2020-12-28:
@AshtonKem The local version of mosquitto running correctly in a docker containter is 1.6.12

@emontnemery, I have observed other strange behavior on the mosquitto addon. I have some old sonoff relay;s with old versions of espeasy/letscontrolit firmware. I reconfigured them to use mqtt but it resulted in a lot of warnings in the HA addon mosquitto log. like device with IP connected followed immediately with device with IP lost connection. No MQTT messages were observed from these devices in the MQTT explorer. My guess was that i needed to upgrade the old firmware.

When i replaced the mosquitto addon with the external mosquitto broker the messages from these old devices came in correctly. This make me think to believe that there is something wrong with the HA internal Mosquitto addon.

emontnemery, 2020-12-28:
@paul12ster yes, from the description it sounds like the addon is enforcing authentication or a protocol version not supported by the devices.

AshtonKem, 2020-12-29:
First, I did the following steps:

1. Uninstall MQTT integration
2. Restart HA core
3. Reinstall MQTT integration

Here are the logs from that:

```
2020-12-28 18:07:25 INFO (MainThread) [homeassistant.core] Starting Home Assistant
2020-12-28 18:07:25 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event core_config_updated[L]>
2020-12-28 18:07:25 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event homeassistant_start[L]>
2020-12-28 18:07:25 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=person.ashton_kemerling, old_state=<state person.ashton_kemerling=not_home; editable=True, id=ashton_kemerling, latitude=$LAT, longitude=$LONG, gps_accuracy=65, source=person.ashton_kemerling, user_id=35f45fbc996142e38c8d4818930e4421, friendly_name=Ashton Kemerling @ 2020-12-28T18:07:21.453150-07:00>, new_state=<state person.ashton_kemerling=not_home; editable=True, id=ashton_kemerling, latitude=$LAT, longitude=$LONG, gps_accuracy=65, source=device_tracker.ipad_2, user_id=35f45fbc996142e38c8d4818930e4421, friendly_name=Ashton Kemerling @ 2020-12-28T18:07:21.453150-07:00>>
2020-12-28 18:07:25 INFO (SyncWorker_4) [homeassistant.components.zeroconf] Starting Zeroconf broadcast
2020-12-28 18:07:26 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event core_config_updated[L]>
2020-12-28 18:07:26 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event homeassistant_started[L]>
2020-12-28 18:07:26 INFO (MainThread) [homeassistant.core] Timer:starting
2020-12-28 18:07:26 INFO (MainThread) [homeassistant.components.automation.sunset] Initialized trigger Sunset
2020-12-28 18:07:26 INFO (MainThread) [homeassistant.components.automation.midnight] Initialized trigger Midnight
2020-12-28 18:07:26 INFO (MainThread) [homeassistant.components.automation.sunrise] Initialized trigger Sunrise
2020-12-28 18:07:27 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=persistent_notification, service=dismiss, service_data=notification_id=config_entry_discovery>
2020-12-28 18:07:32 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=persistent_notification, service=dismiss, service_data=notification_id=config_entry_discovery>
2020-12-28 18:07:35 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=persistent_notification, service=dismiss, service_data=notification_id=config_entry_discovery>
2020-12-28 18:07:36 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=persistent_notification, service=dismiss, service_data=notification_id=config_entry_discovery>
2020-12-28 18:13:11 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=persistent_notification, service=dismiss, service_data=notification_id=config_entry_discovery>
2020-12-28 18:13:11 INFO (MainThread) [homeassistant.setup] Setting up mqtt
2020-12-28 18:13:11 INFO (MainThread) [homeassistant.setup] Setup of domain mqtt took 0.0 seconds
2020-12-28 18:13:11 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event service_registered[L]: domain=mqtt, service=publish>
2020-12-28 18:13:11 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event service_registered[L]: domain=mqtt, service=dump>
2020-12-28 18:13:11 INFO (Thread-7) [homeassistant.components.mqtt] Connected to MQTT server core-mosquitto:1883 (0)
2020-12-28 18:13:11 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event component_loaded[L]: component=mqtt>
2020-12-28 18:13:11 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to homeassistant/#, mid: 1
2020-12-28 18:13:11 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to tasmota/discovery/#, mid: 2
2020-12-28 18:13:13 DEBUG (MainThread) [homeassistant.components.mqtt] Transmitting message on homeassistant/status: 'online', mid: 3
2020-12-28 18:13:13 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on homeassistant/status: b'online'
2020-12-28 18:13:17 INFO (MainThread) [homeassistant.components.websocket_api.http.connection] [2888387192] Connection closed by client
```

Based off of the MQTT error listed on the last line, I presume that there's an authentication issue. My next steps were:

1. Copy username/password from Mosquitto addon
2. Manually set username/password for MQTT integration.

Here are the next logs:

```
2020-12-28 18:16:29 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to homeassistant/#, mid: 1
2020-12-28 18:16:29 INFO (Thread-9) [homeassistant.components.mqtt] Connected to MQTT server core-mosquitto:1883 (0)
2020-12-28 18:16:29 DEBUG (MainThread) [homeassistant.components.mqtt] Unsubscribing from homeassistant/#, mid: 2
2020-12-28 18:16:29 DEBUG (MainThread) [homeassistant.components.mqtt] Unsubscribing from tasmota/discovery/#, mid: 3
2020-12-28 18:16:29 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to homeassistant/#, mid: 4
2020-12-28 18:16:29 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to tasmota/discovery/#, mid: 5
2020-12-28 18:16:31 DEBUG (MainThread) [homeassistant.components.mqtt] Transmitting message on homeassistant/status: 'online', mid: 6
2020-12-28 18:16:31 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on homeassistant/status: b'online'
```

Next steps:

1. Force rename device in Zigbee2MQTT UI, including HA name.

[... cut ...]
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/rename', payload '{"data":{"from":"Garage One Foo","homeassistant_rename":true,"to":"Garage Heater One"},"status":"ok"}'
INFO: MQTT publish: topic 'zigbee2mqtt/Motion Sensor', payload '{"battery":91,"illuminance":38,"illuminance_lux":38,"linkquality":36,"occupancy":true,"voltage":2985}'
INFO: MQTT publish: topic 'zigbee2mqtt/Motion Sensor', payload '{"battery":91,"illuminance":38,"illuminance_lux":38,"linkquality":36,"occupancy":true,"voltage":2985}'
```

AshtonKem, 2020-12-29:
I figured it out! 

It looks like something is wrong with authentication between Zigbee2mqtt and the Mosquitto HA addon. Manually creating a new username/password for Zigbee2mqtt and configuring it in both addons made the service work.

For some reason, "anonymous" login does not work, nor does the automatic HA powered authentication that I believe should happen.

AshtonKem, 2020-12-29:
Still having issues with #5150, for the record. Those issues appear to be largely separate.

AshtonKem, 2020-12-29:
To expand my comment, this configuration works:

```yaml
logins:
  - username: homeassistant-discovery
    password: $PASSWORD
  - username: zigbee2mqtt
    password: $PASSWORD
anonymous: true
customize:
  active: false
  folder: mosquitto
certfile: fullchain.pem
keyfile: privkey.pem
require_certificate: false
```

This one does not:

```yaml
logins: []
anonymous: true
customize:
  active: false
  folder: mosquitto
certfile: fullchain.pem
keyfile: privkey.pem
require_certificate: false
```

Koenkk, 2020-12-29:
Since this is neither a Zigbee2MQTT nor HA issue (but a setup one?) I will close this.

#5150 is a HA bug and will be fixed (https://github.com/Koenkk/zigbee2mqtt/issues/5150#issuecomment-752022929)

closed 2020-12-29 by Koenkk

AshtonKem, 2020-12-29:
Probably worth either making some change to the documentation, or filing a bug with the mosquitto addon if authentication should “just work” out of the box. 

Oh, and I would argue that there’s probably a logging bug in zigbee2mqtt; I would’ve expected all kinds of error logs if the MQTT in the above situation. 

—
Ashton Kemerling

Sent from my iPhone.

Koenkk, 2020-12-29:
> Probably worth either making some change to the documentation, or filing a bug with the mosquitto addon if authentication should “just work” out of the box.

Can you make a pr to update: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/integration/home_assistant.md ?

> Oh, and I would argue that there’s probably a logging bug in zigbee2mqtt; I would’ve expected all kinds of error logs if the MQTT in the above situation.

When using an MQTT client and subscribing to `zigbee2mqtt/bridge/state`, do you see any messages being logged?