Koenkk/zigbee2mqtt #5347: TS0601 / PiZiGate : (Cannot read property 'length' of undefined)

Kalypox, 2020-12-19:
Hello

## What happened
I can't control my TS0601 valve (https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html).
It worked during 1 week, and yesterday it decided to stop.

## What did you expect to happen
I would like to control temperature in my room.

## How to reproduce it (minimal and precise)
- Add TS0601 to zigbee2mqtt
- Try to communicate with the valve

## Debug info

https://pastebin.com/raw/Y3bjtSJ6
The device name is "chambre_jonathan_radiateur"

Zigbee2MQTT version: ??? (how can i check ?)
Adapter hardware: PiZiGate
Adapter firmware version: 3.1d

Thanks !

Kalypox, 2020-12-19:
My problem is solved, i deleted my Zigbee2MQTT container and i rebuilt it...
But really strange...

I kept these files : (in persistent volume)
[image]

I don't close the issue, maybe it's a bug.

Kalypox, 2020-12-19:
The problem is back !
```
Zigbee2MQTT:error 2020-12-19 20:56:29: Publish 'set' 'local_temperature_calibration' to 'chambre_jonathan_radiateur' failed: 'TypeError: Command 0xbc33acfffe63fb1e/1 manuSpecificTuya.setData({"status":0,"transid":1,"dp":44,"datatype":2,"length_hi":0,"length_lo":4,"data":[0,0,0,0]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Cannot read property 'length' of undefined)'
```

insipiens, 2020-12-22:
Try set, not get
Pour example

`zigbee2mqtt/chambre_jonathan_radiateur/set`

Kalypox, 2020-12-22:
hi @insipiens,

When i try `zigbee2mqtt/chambre_jonathan_radiateur/child_lock/set`:
I have `Zigbee2MQTT:error 2020-12-23 00:42:24: Entity 'chambre_jonathan_radiateur/child_lock' is unknown`
I think it's because the format is "zigbee2mqtt/entityname/***

If I try: `zigbee2mqtt/chambre_jonathan_radiateur/set`
with: `{"child_lock": "LOCK"}`

I get this:
`Zigbee2MQTT:error 2020-12-23 00:45:48: Publish 'set' 'child_lock' to 'chambre_jonathan_radiateur' failed: 'TypeError: Command 0xbc33acfffe63fb1e/1 manuSpecificTuya.setData({"status":0,"transid":32,"dp":7,"datatype":1,"length_hi":0,"length_lo":1,"data":[1]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Cannot read property 'length' of undefined)'
`

insipiens, 2020-12-23:
I’d say there is no conversion then, 5004 isn’t a DP command I recognise from my devices, but I haven’t updated to the latest dev code version.

perhaps @drzony can shed more light - for comment only?

Kalypox, 2020-12-23:
After a lot of tries i found the problem !!!!

The valve stop working when i set a **friendly name** in the configuration.yaml !
The only friendly name which works is:
```
  '0xbc33acfffe63fb1e':
    friendly_name: '0xbc33acfffe63fb1e'
```

If i set this:
```
  '0xbc33acfffe63fb1e':
    friendly_name: 'chambre_jonathan_radiateur'
```
or this:
```
  '0xbc33acfffe63fb1e':
    friendly_name: chambre_jonathan_radiateur
```
It doesn't works.
It's perhaps a bug in Zigbee2mqtt because I cleared my docker (removed persistent volume and pulled new image), so my installation is pure.

This problem made me search during 1 week ^^
I don't close the issue, maybe it's a bug.
I wait expert opinion !

drzony, 2020-12-23:
```txt
zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #13 with data {"status":0,"transid":15,"dp":13,"datatype":5,"fn":0,"data":{"type":"Buffer","data":[0]}}
zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #115 with data {"status":0,"transid":56,"dp":115,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}}
```
This warnings are OK, since we don't know what the corresponding DPs do, so until someone sniffs this information when connected to TuYa gateway and confirms what they do they will stay that way.

As for `undefined` error, try renaming the device via interface, not via configuration.yaml

Kalypox, 2020-12-23:
Hi !

Okay i see, thanks for the informations !

But that's not what worries me, the problem is the `(Cannot read property 'length' of undefined)'`.
I didn't get this before. It's a really strange message, do you have an idea ?

Kalypox, 2020-12-23:
Hi,

Okay thanks for the informations.
What do you mean by "interface" ?
I know 3 ways to rename it:
- The configuration.yaml
- The Zigbee2MQTTassistant
- The `zigbee2mqtt/bridge/config/rename` with `{"old": "OLD_FRIENDLY_NAME", "new": "NEW_FRIENDLY_NAME"}` (but how it works if there isn't OLD_FRIENDLY_NAME ?)
- Or this: `zigbee2mqtt/bridge/config/rename_last`

drzony, 2020-12-23:
Newer versions of z2m have builtin web interface.
You can use `zigbee2mqtt/bridge/config/rename` with `{"old": "0xbc33acfffe63fb1e", "new": "chambre_jonathan_radiateur"}`.
If you change the name in configuration.yaml, you probably need to modify/delete `state.json` and make sure that z2m is stopped before changing configuration.yaml

Kalypox, 2020-12-23:
An interface ? I don't find anything about that, how can i access it ?
I have 1.16.2.

I tried to remove the valve, repair the valve and modify the name with your command, and it works perfectly ! Thanks you a lot !
I tried to set a target temperature, **it works**, but i still have the error:
```
Zigbee2MQTT:error 2020-12-23 14:29:45: Publish 'set' 'current_heating_setpoint' to 'chambre_jonathan_radiateur' failed: 'TypeError: Command 0xbc33acfffe63fb1e/1 manuSpecificTuya.setData({"status":0,"transid":58,"dp":2,"datatype":2,"length_hi":0,"length_lo":4,"data":[0,0,0,135]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Cannot read property 'length' of undefined)'
```

drzony, 2020-12-23:
For web frontend see #4266 
If you want to check with latest version, see [here](https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Kalypox, 2020-12-23:
Niiice, i switch to dev branch now !
Thanks for the tip !

Kalypox, 2020-12-25:
I have sometimes this message:
```
Zigbee2MQTT:error 2020-12-25 16:05:38: Publish 'set' 'local_temperature_calibration' to 'chambre_jonathan_radiateur' failed: 'TypeError: Command 0xbc33acfffe63fb1e/1 manuSpecificTuya.setData({"status":0,"transid":12,"dp":44,"datatype":2,"length_hi":0,"length_lo":4,"data":[0,0,0,0]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Cannot read property 'length' of undefined)'
```

But the valve works perfectly.

Should i close the issue ?

Kalypox, 2021-01-25:
Hi,
FYI, the "problem" is always here:
`Zigbee2MQTT:error 2021-01-25 09:22:32: Publish 'set' 'local_temperature_calibration' to 'chambre_jonathan_radiateur' failed: 'TypeError: Command 0xbc33acfffe63fb1e/1 manuSpecificTuya.setData({"status":0,"transid":70,"dp":44,"datatype":2,"length_hi":0,"length_lo":4,"data":[0,0,0,0]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Cannot read property 'length' of undefined)'`

insipiens, 2021-01-25:
I wonder if that friendly name is just too long?

drzony, 2021-01-25:
This may just be communication error with the device (bad reception, too many devices for coordinator etc)

closed 2021-03-05 by github-actions[bot]