Koenkk/zigbee2mqtt #5349: Tuya temperature sensor updating slowly or not at all

maxiedaniels, 2020-12-19:
## What happened

I'm using a Tuya temperature sensor with Home Assistant, zigbee enabled using zigbee2mqtt+armbian on a Pi Zero + CC2531 that I flashed. My Tuya temperature sensor is barely updating or not updating at all (and I have A/C running to intentionally keep the temperature changing). If I look at the graph in Home Assistant, it sometimes updates every 5 minutes, sometimes every 30 minutes, sometimes it doesn't update for hours. I've tried a second CC2531, no improvement. I've tried doing a custom config in my HA config, with force_update: true, and scan_interval: 20, but it doesn't help.

The only clue I have is that my Pi Zero+Armbian stops responding via ssh at times, not sure what causes it. But i've checked all the logs I can find and don't see anything that suggests a problem.

## What did you expect to happen
I expect the sensor to update normally.

## How to reproduce it (minimal and precise)
This happens no matter what I do.

## Debug info
Zigbee2MQTT version: 1.16.2
Adapter hardware: CC2531 (I've tried two different ones)
Adapter firmware version: Z-Stack Home 1.2 (default)

sjorge, 2020-12-19:
There is a huge thread, that I can't locate right now. It's a known 'issues' but the device does not support configuring reporting, so we can't make it send more frequant updates.

Koenkk, 2020-12-20:
Thanks @sjorge for linking #4202. I guess this can be closed.

closed 2020-12-20 by Koenkk