Koenkk/zigbee2mqtt #5350: TRADFRI remote control: no messages after pairing  (Adapter: Conbee/RaspBee)

sim-san, 2020-12-19:
## What happened
I was able to successfully connect the [IKEA TRADFRI remote control](https://www.zigbee2mqtt.io/devices/E1524_E1810.html). 
Log of the paring: https://hastebin.com/icomoraqox.apache
After that, whenever I press any of the buttons on the remote, I don't get any message at all. Neither in the logs of Zigbee2MQTT nor in HomeAssitant.

## What did you expect to happen
See some information in the log...
I wonder if it can be because I use the first generation Conbee stick ?  I have in my Zigbee network two other bulbs from Müller Licht and a sensor from Xiaomi. These work well except that the "linkquality" value is always at 0.

## How to reproduce it (minimal and precise)
/

## Debug info
Zigbee2MQTT version: 1.16.2-dev commit: 8e3dd2e
Adapter hardware: **Conbee I** (ConBee/RaspBee)
Adapter firmware version: 0x263a0500

sim-san, 2020-12-19:
I have enable the zigbee-herdsman logging and found this information after pressing a button on the remote

```
2020-12-19T19:55:39.979Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":15,"manufacturerCode":null,"commandIdentifier":2},"Payload":{}},"address":34567,"endpoint":1,"linkquality":0,"groupID":901}'
2020-12-19T19:55:39.979Z zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '34567', skipping...
```

sim-san, 2020-12-19:
I found a post from the ZHA (Zigbee Integration in HomeAssitant based on zigpy):

`Hrm, For whatever reason Conbee I is using wrong NWK address for Ikea Tradfri 5 button remote. Same remote works fine on ConBee II You however still should see some data from zigpy-deconz when you press buttons on the remote. Make sure to enable debug for both zigpy and zigpy-deconz components.`
Source: https://community.home-assistant.io/t/zha-zigbee-tested-devices-please-add-your-device-results/17718/1257

Seems that the Conbee I has an internal problem with the IKEA TRADFRI 5 button remote

Koenkk, 2020-12-20:
Can you try stopping z2m, update the networkadress for the remote in the data/database.db to `34567 ` and starting z2m after that?

sim-san, 2020-12-20:
Thx, now I get messages...

sim-san, 2020-12-20:
But after some clicks I do not get any message

sim-san, 2020-12-20:
@Koenkk Do you recommend the Conbee II or Electrolama zig-a-zig-ah! (zzh!) or slaesh’s CC2652RB stick ?

Koenkk, 2020-12-20:
Conbee II is still experimental so Electrolama zig-a-zig-ah! (zzh!) or slaesh’s CC2652RB stick (no big differences between those)

closed 2021-01-20 by sim-san