Koenkk/zigbee2mqtt #5351: Illuminance peaks aqara motion sensor

jgarmar, 2020-12-19:
[image]

Hi! Could someone tell me why I have those Illuminance peaks with the motion sensor aqara? It's always around 93, 92 luxes, I'm updated to the last version Thank you

kevincaradant, 2020-12-20:
Hi

Not sure about Z2M faults.

Its look like your motion sensor is sometimes 'broken' and reports weird values. Mine don't do this. I check right now

[image]

Maybe your battery is weak? Did you try to change it? 
Do you have a good / enought signal with the co-ordinator / routers?

sjorge, 2020-12-20:
When i still used mine I also had peaks, it was because one of the lamps would hit the sensor.
So if there was motion and I turned on the lights it would have a huge peak.

jgarmar, 2020-12-20:
Yes, but it is not the case, because the peaks occur when the room is dark and there is no movement ... I have asked for another sensor because it does not happen to me with the others that I have. thanks

closed 2020-12-20 by Koenkk