Koenkk/zigbee2mqtt #5352: YSR-MINI-01 : Transition / State

Kalypox, 2020-12-20:
## What happened
If i set a transition time in configuration, the device does not light up/light down anymore with ON and OFF.

## What did you expect to happen
Enable the ON/OFF fonctionnality + transition time

## How to reproduce it (minimal and precise)
- Add YSR-MINI-01
- Add transition time in configuration.yaml: 
 ````
   '0x00124b001f945a7e':
    friendly_name: led_1
    transition: 0.5
 ```
- Try to change state of the led with:
`zigbee2mqtt/led_1/set` & `{"state": "ON"}` or `{"state": "OFF"}`

Only 
`zigbee2mqtt/led_1/set` & `{"state": "TOGGLE"}` work

## Debug info
Zigbee2MQTT version:1.16.2
Adapter hardware: PiZiGate
Adapter firmware version: 3.1d

Koenkk, 2020-12-20:
Do transitions work when changing the brightness? e.g. publish to `zigbee2mqtt/led_1/set` payload `{"brightness": 20, "transition": 3}`

Kalypox, 2020-12-20:
If transition time **is configured** in configuration.yaml `transition: 1`
**`{"brightness": 20, "transition": 3}` work with 3sec transition**
`{"brightness": 20}` work with 1 sec transition

If transition time **is not configured** in configuration.yaml
**`{"brightness": 20, "transition": 3}` work with 3sec transition**
`{"brightness": 20}` work with 0 sec transition

EDIT:

If transition time **is configured** in configuration.yaml

I can power ON the led with service `light.turn_on` ! The led turn on immediatly without transition.
Buit i can't power OFF with the service `light.turn_off`

Koenkk, 2020-12-20:
When the device is on, does `{"brightness": 0, "transition": 3}` work?

Kalypox, 2020-12-20:
If transition time **is not configured** in configuration.yaml
`{"brightness": 0, "transition": 3}` doesn't works
`{"brightness": 0}` doesn't works too

If transition time **is configured** in configuration.yaml transition: 1
`{"brightness": 0, "transition": 3}` doesn't works
`{"brightness": 0}` doesn't works too

Koenkk, 2020-12-20:
I think there is a bug in the firmware of this device. Because for `{"brightness": 0, "transition": 3}` and `{"brightness": 20, "transition": 3}` the same command is used (which works fine on other devices).

Kalypox, 2020-12-21:
I have a second YSR-MINI-01, i'll try tomorrow.

Kalypox, 2020-12-21:
It's alsmot the same with my second.
My first: brightness=1 is a little bit lit
My second: brightness=1 is off

When you said "other device" what do you mean ? Others YSR-MINI or other led controller ?

Koenkk, 2020-12-21:
@Kalypox with other device I mean a different led controller/light (e.g. Philips Hue bulb)

Kalypox, 2020-12-21:
Ok, so i "just" have to create a new device following this tutorial : https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

I'll try !

Koenkk, 2020-12-21:
Not sure what you are trying to accomplish. What I mean is that it's not working due to a bug in the firmware of this device, we cannot fix it from Zigbee2MQTT. You can see this because the on/off transition works correctly with other bulbs.

Kalypox, 2020-12-21:
My idea was to apply the transition settings at "brightness", "color_temp", "color_xy" but not "state".
The YSR won't turn on/off when there is "transition" setting with "state".

And also try to map "brightness:0" to "state:OFF"

By default there is a transition on YSR when we turn on/off.

closed 2020-12-25 by Kalypox