Koenkk/zigbee2mqtt #5356: Touchlink not working wit CC2652P?

olili, 2020-12-20:
I'm facing problems by scanning for touchlink devices with my  CC2652P

What did you expect to happen
   Philips ambient light bar will be correctly detected as touchlink device 
   Distance between Light bar and CC2562 is less then 10cm; 
   With my old CC2530 this setupp is working without any problem. 

What happened
- either NO touchlink device is detected//found
   INFO: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/scan', payload '{"data":{"found":[]},"status":"ok"}'
   INFO: Start Touchlink scan
  INFO: Finished Touchlink sc5cm disan
  INFO: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/scan', payload '{"data":{"found":[]},"status":"ok"}'

-or error:
  INFO: Start Touchlink scan
  ERROR: Request 'zigbee2mqtt/bridge/request/touchlink/scan' failed with error: 'SRSP - AF - interPanCtl after 6000ms'
  INFO: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/scan', payload '{"data":{},"error":"SRSP - AF - interPanCtl after 6000ms","status":"error"}'

## How to reproduce it (minimal and precise)
start touchlink scan in  web frontend 

## Debug info
zigbee2mqtt version 1.16.2 commit: 0514204
coordinator type: CC2652P,  zStack3x0, coordinator revision 20201113
frontend version 0.2.100

Any Hints?

Koenkk, 2020-12-20:
Looks that it's quite a big device, I guess you have to put it at the right spot.

> SRSP - AF - interPanCtl after 6000ms

This is a known issue in the CC2652 (has to be fixed by texas instruments).

olili, 2020-12-21:
Thx.
Tried several positions. . Distance less than 5cm. No housing. No improvement, nothing recognized.
Once I got the SRSP-AF message I have to reset the CC2652P, otherwise it will not work anymore. 

Seems to be an bigger Issue. Tried also egony-FW. Same behavior. 

Best regards, p.

sjorge, 2020-12-29:
@Koenkk can confirm, zzhp-lite uses the same chip and I am also ahving the issue.
Touchlink doesn't seem to work at all, once I hit scan... the stick seems to die.

All commands (for example permit_join, or a set/get):
```
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{},"error":"Cannot execute command, in Inter-PAN mode","status":"error"}'
```

I have to physically remove the stick, count to about 10 and insert it before it works again..
- kill z2m
- unplug
- wait a bit
- plug
- start z2m

davidjb, 2020-12-30:
I'm seeing the same behaviour, albeit on a CC26X2R1 (zzh).  My Touchlink scans will either:
a) work + find the device, b) work + not find anything or c) error like @olili describes (`SRSP - AF - interPanCtl after 6000ms`) and then after erroring, zigbee2mqtt will crash as @sjorge describes, requiring the same unplug-wait-replug process to get zigbee2mqtt starting again.  

Without physically unplugging or seemingly a sufficient wait in the middle, zigbee2mqtt fails to connect:

```
Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
```

I'm running the latest dev (https://github.com/Koenkk/zigbee2mqtt/commit/d1842ab69ae62a8a59a0de1b0caa86f73d614763) version.  Here's the traceback of a Touchlink scan that crashes zigbee2mqtt.

```
Zigbee2MQTT:info  2020-12-31 08:13:22: Start Touchlink scan
Zigbee2MQTT:error 2020-12-31 08:13:39: Request 'zigbee2mqtt/bridge/request/touchlink/scan' failed with error: 'SRSP - AF - interPanCtl after 6000ms'
Zigbee2MQTT:info  2020-12-31 08:13:39: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/scan', payload '{"data":{},"error":"SRSP - AF - interPanCtl after 6000ms","status":"error"}'

/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:879
            throw new Error(`Cannot execute command, in Inter-PAN mode`);
                  ^
Error: Cannot execute command, in Inter-PAN mode
    at ZStackAdapter.checkInterpanLock (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:879:19)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:166:22)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:23:12)
    at Object.func (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:165:44)
    at Queue.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/queue.js:29:46)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/queue.js:8:71
```

davidjb, 2021-01-31:
Still an issue as of several days ago with latest dev version

davidjb, 2021-02-03:
Just upgraded to 1.17.1 and Touchlink is working again, but only once and not perfectly.  For note, Permit Join was enabled at this point before I started.  I actually noticed this below issue when Touchlink was working previously but didn't think too much of it -- if I run a Touchlink scan, I can find and successfully factory reset the device but an "unsupported device" (generic Zigbee icon and red exclamation mark icon) in the web UI.

Here's the logs up to that point:

```
Zigbee2MQTT:info  2021-02-03 19:20:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0xc0ffeec0ffeec0ffee"},"type":"pairing"}'
Zigbee2MQTT:info  2021-02-03 19:20:51: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xc0ffeec0ffeec0ffee","ieee_address":"0xc0ffeec0ffeec0ffee"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-02-03 19:20:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xc0ffeec0ffeec0ffee"},"type":"device_announced"}'
Zigbee2MQTT:info  2021-02-03 19:20:52: Successfully factory reset device through Touchlink
Zigbee2MQTT:info  2021-02-03 19:20:52: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/factory_reset', payload '{"data":{"channel":11,"ieee_address":"0xc0ffeec0ffeec0ffee"},"status":"ok","transaction":"s4cpy-5"}'
```

Then, if I remove the 'failed' device via the web UI, the device was immediately interviewed again, and joined the network successfully.

A subsequent scan after the first one leads to the same error as before `Cannot execute command, in Inter-PAN mode`, resulting in the same unplug-wait-replug process to get zigbee2mqtt started again.

After this point, zigbee2mqtt had crashed and I had to restart it, leaving me with `permit_join` set as false on startup.  Now, I tried a Touchlink scan again and this succeeded okay with the factory reset of my next device.  After this, I then enabled `permit_join` and the device joined the network fine - no issues with failed pairing.  At this point, trying to re-run a Touchlink scan crashes zigbee2mqtt again.

So, it seems that you only get one scan per invocation of zigbee2mqtt (or plug cycle of the dongle) -- maybe the Touchlink scan leaves the setup in some unclean state? - and there's some oddity when touchlink devices do try and join immediately after / during (?) the scan - perhaps a race condition.  Hopefully the debugging helps -- trial and error was enough to get my devices paired.

ipatalas, 2021-02-14:
@davidjb Thanks for your input. 
I've got double Philips Hue Play light bar and I'm struggling to pair it (had it working on my previous z2m adapter).
The first one I finally got working using the same technique as you mentioned. I had to unplug the adapter for a while, then I started it back, turned on z2m, set `permit_join` to true and then tried Touchlink. It detected one of the lights correctly. 
It was also not recognized at the beginning (same generic Zigbee icon and red exclamation mark icon) but after removing it it auto-joined the network and this time it got recognized correctly and everything works just fine.
The problem is with the other light. I've tried the same approach couple of times and no luck so far :(
The lights are 40-50cm apart. I moved the adapter close to the other light and surprisingly it still detects the first one and only this one (even though it's 40-50cm away).
I'm doing it over and over again last half an hour and nothing :( I even unplugged the first light so that only this one is on.
This is sick...

davidjb, 2021-03-17:
Still an issue; tested with 1.18.1

flavorplus, 2021-04-06:
Yep still have the same issue. I’m on 1.18.2

afleuryg, 2021-04-07:
Hi,
Sorry to add an other voice to the sauce.
Same issue here: using either touchlink scan or factory reset will land my zig-a-zig-ah in this strange unresponsive state, with a  "SRSP - AF - interPanCtl after 6000ms" error.
Zigbee2MQTT version 1.18.1 (commit #unknown)
zigbee-herdsman (0.13.71)
Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"}'
Zigbee channel used: 11
Zigbee network is empty (coordinator only).

@koenkk, you said it was a know issue with CC2652, but do you think there is a way to fail gracefully at least? (I didn't dwelve into the code yet)

(and as I am here, thanks for this wonderful piece of software!)

Koenkk, 2021-04-08:
@afleuryg the problem is that the whole adapter crashes, so you have to completely stop z2m and replug the adapter. I hope it will be fixed in the new FW update from Texas Instruments (should be released soon since they release every quarter).

877dev, 2021-04-09:
> @afleuryg the problem is that the whole adapter crashes, so you have to completely stop z2m and replug the adapter. I hope it will be fixed in the new FW update from Texas Instruments (should be released soon since they release every quarter).

Great to hear this, as I 'm having the same issue with ZZH when attempting to Touchlink reset my Hue bulb (strange as it used to work with zzh).

khmikkelsen, 2021-04-27:
> @davidjb Thanks for your input.
> I've got double Philips Hue Play light bar and I'm struggling to pair it (had it working on my previous z2m adapter)
> ...
> **Update2:**
> OMG. I've got it but it's a pain...
> It disappeared again and couldn't find it for couple of minutes. Then I got it again and figured out the timeout issue. For some (bizzare) reason only odd operations success so first, third and so on. At the beginning I was clicking 'identify' (succeeds), 'factory reset' (fails), 'identify' (succeeds again), 'factory reset' (fails again). Then I figured then when I click Identify twice the second one fails (same timeout). That got me thinking :)
> Finally clicked identify, identify and factory reset (it should also work when you click factory reset as the first one) and it worked...
> That's been hell of a ride.
> 
> Hope that helps someone in the future but you need a lot of patience.

This fix is such an IT thing, that it actually worked for me. Bless you and your soul.

877dev, 2021-05-04:
> @afleuryg the problem is that the whole adapter crashes, so you have to completely stop z2m and replug the adapter. I hope it will be fixed in the new FW update from Texas Instruments (should be released soon since they release every quarter).

@Koenkk Hello, I don't suppose the Texas firmware fix got released yet? 

Can't remember how to pair my Hue bulbs with my ZZH, touchlink crashes the adapter every time..

Koenkk, 2021-05-04:
@877dev TI updated their SDK, new firmware can be found here: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin but didn't check if this fixes the touchlink issue yet.

877dev, 2021-05-04:
@Koenkk it does not appear to have fixed the touchlink issue for me, using zzh:

```
Zigbee2MQTT:info  2021-05-04 19:48:29: Start Touchlink scan
Zigbee2MQTT:debug 2021-05-04 19:48:40: Saving state to file /app/data/state.json
Zigbee2MQTT:error 2021-05-04 19:48:49: Request 'zigbee2mqtt/bridge/request/touchlink/scan' failed with error: 'SRSP - AF - interPanCtl after 6000ms'
Zigbee2MQTT:info  2021-05-04 19:48:49: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/scan', payload '{"data":{},"error":"SRSP - AF - interPanCtl after 6000ms","status":"error","transaction":"xn776-2"}'
(node:8) UnhandledPromiseRejectionWarning: Error: Cannot execute command, in Inter-PAN mode
    at ZStackAdapter.checkInterpanLock (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:881:19)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:166:22)
    at Generator.next (<anonymous>)
    at /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:23:12)
    at Object.func (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:165:44)
    at Queue.<anonymous> (/app/node_modules/zigbee-herdsman/dist/utils/queue.js:29:46)
    at Generator.next (<anonymous>)
    at /app/node_modules/zigbee-herdsman/dist/utils/queue.js:8:71
(Use `node --trace-warnings ...` to show where the warning was created)
(node:8) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
(node:8) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

dbotman, 2021-05-07:
I just updated de CC2562R to firmware 20210430. I successfully did a factory reset for a Ikea bulb with touchlink. But with other Ikea lamps it fails several times.

JacobMillward, 2021-05-21:
I have a ZZH! (CC2652R) and this bug seems to have gotten my adapter into a state I can't get out of. I've reflashed, it, re-plugged it. Tried a few different combinations.

Can someone please give me some steps to get back to a working state, so I can never touch the touchlink button again?

dbotman, 2021-05-29:
> I have a ZZH! (CC2652R) and this bug seems to have gotten my adapter into a state I can't get out of. I've reflashed, it, re-plugged it. Tried a few different combinations.
> 
> Keep getting the following that @davidjb mentioned
> `Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)`
> 
> ~Can someone please give me some steps to get back to a working state, so I can never touch the touchlink button again?~
> Can confirm that I managed to get back into a working state by flashing the newer 20210430 coordinator firmware. For some reason the 20210128 firmware wouldn't let me continue.

@JacobMillward Does it always work for you with firmware 20210430? It works for me once in a while, but if it fails I have to restart everything.

MattWhite-personal, 2021-06-05:
Where are you finding the 20210430 firmware for the ZZH CC2642R - finding the same bug with the Touchlink process and struggling to get it up and running again without recreating the Z2M docker image and content.

JacobMillward, 2021-06-18:
@dbotman I haven't tried using Touchlink again since, so I wouldn't know. I'm a bit worried about bricking my ZZH (I don't have the debug adapter), so don't want to risk it. My lights need to keep working!

davidjb, 2021-07-19:
Still an issue, but able to be worked around if you've got Philips lights + have a dimmer switch to [trigger a factory reset](https://www.zigbee2mqtt.io/devices/9290011370.html#hue-dimmer-switch), avoiding needing Touchlink.

Jellyfrog, 2021-08-02:
Did someone try the latest firmware 20210708 yet?

sjorge, 2021-08-02:
Still broken I believe

877dev, 2021-08-03:
> Still broken I believe

Hi, did you try it?
I can test it over the next couple of days.

sjorge, 2021-08-03:
I tried it while rebuilding my network, I'm pretty sure I already upgraded to the latest firmware when I was doing that.

877dev, 2021-08-07:
@Koenkk @sjorge confirmed not working for me, 

```
Zigbee2MQTT:info  2021-08-07 14:49:21: Starting Zigbee2MQTT version 1.21.0 (commit #70891ee)
Zigbee2MQTT:info  2021-08-07 14:49:21: Starting zigbee-herdsman (0.13.124)
Zigbee2MQTT:debug 2021-08-07 14:49:21: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/ttyACM0","rtscts":false}}'
```

Start touchlink scan:

```
(node:7) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
Zigbee2MQTT:error 2021-08-07 14:43:35: Request 'zigbee2mqtt/bridge/request/touchlink/scan' failed with error: 'SRSP - AF - interPanCtl after 6000ms'
Zigbee2MQTT:info  2021-08-07 14:43:35: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/scan', payload '{"data":{},"error":"SRSP - AF - interPanCtl after 6000ms","status":"error","transaction":"iynz2-2"}'
Zigbee2MQTT:debug 2021-08-07 14:44:18: Received MQTT message on 'zigbee2mqtt/Radiator_main_bedroom/set/local_temperature_calibration' with data '-1'
Zigbee2MQTT:debug 2021-08-07 14:44:18: Publishing 'set' 'local_temperature_calibration' to 'Radiator_main_bedroom'
Zigbee2MQTT:error 2021-08-07 14:44:18: Publish 'set' 'local_temperature_calibration' to 'Radiator_main_bedroom' failed: 'Error: Command 0x847127fffe0eb922/1 manuSpecificTuya.setData({"status":0,"transid":0,"dp":44,"datatype":2,"length_hi":0,"length_lo":4,"data":[255,255,255,246]}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Cannot execute command, in Inter-PAN mode)'
Zigbee2MQTT:debug 2021-08-07 14:44:18: Error: Command 0x847127fffe0eb922/1 manuSpecificTuya.setData({"status":0,"transid":0,"dp":44,"datatype":2,"length_hi":0,"length_lo":4,"data":[255,255,255,246]}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Cannot execute command, in Inter-PAN mode)
    at ZStackAdapter.checkInterpanLock (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1022:19)
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:330:18)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:42)
    at /app/node_modules/zigbee-herdsman/src/utils/queue.ts:21:18
    at new Promise (<anonymous>)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:19:16)
    at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:329:27)
    at /app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:706:39
    at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:282:20)
    at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:705:39)
    ```

davidjb, 2021-09-07:
Still an issue on 1.21.1 and latest firmware 20210708

McMornan, 2021-09-19:
Hi, I have a slae.sh stick 2652rb - same thing. Not a bummer but ... unexpected...

klaernie, 2021-09-30:
I ran into this issue a couple times today (got my TubeZB POE gateway).

The workaround I found to make it more reliable is to restart Z2M and the gateway/stick. Then wait for it to settle, and disable joining. Right after that a Touchlink scan works, finds the bulb I'm trying to pair, so I can enable joining again, and factory reset the bulb.

This was with Z2M 1.19.1 (9bd46932) and a zStack3x0 coordinator with revision 20210708.

afleuryg, 2021-10-10:
> The workaround I found to make it more reliable is to restart Z2M and the gateway/stick. Then wait for it to settle, and disable joining. Right after that a Touchlink scan works, finds the bulb I'm trying to pair, so I can enable joining again, and factory reset the bulb.

This doesn't solve completely the problem, but it alleviates a bit the bad behaviour.
This is what I found working 3 times out of 4 for me (which is a big enhancement on the 0 times out of 15 I had before):
- plug the stick
- startup z2m
- wait 20 seconds
- permit join
- wait 4 second
- disable permit join
- wait 10 seconds
- send any touchlink command (scan, factory reset, identify)
- if the action doesn't fail in "interPanCtl", whatever the result of the previous action, wait 60 seconds (yes, it's slooooow)
- return to step 4 (permit join)

It's slow, but at least there is no need to disconnect/reconnect the stick and restart z2m.

SerhiiAksiutin, 2021-10-10:
I am surprised how this has been open for a year. Either I am missing on any other obvious way to connect Hue lights to `zigbee2mqtt` or it is absolutely a blocker and has to be treated with the highest priority. Please advise.

elamperti, 2021-11-10:
Not stale, problem still exists.

SerhiiAksiutin, 2021-11-10:
Not for my 8 hue lights of different kink. If I restart a light right before triggering reset. it works every time.

elamperti, 2021-11-10:
@SerhiiAksiutin I've already tried esoteric procedures like that one and it didn't work for me. In any case that's just a workaround, not a solution, and this issue must remain open.

sorentorp, 2021-11-11:
I'm new to zigbee2mqtt - but also a problem for me.

`zigbee2mqtt    | Zigbee2MQTT:info  2021-11-11 08:50:26: Start Touchlink scan
zigbee2mqtt    | (node:7) UnhandledPromiseRejectionWarning: Error: Cannot execute command, in Inter-PAN mode
zigbee2mqtt    |     at ZStackAdapter.checkInterpanLock (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1023:19)
zigbee2mqtt    |     at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:197:18)
zigbee2mqtt    |     at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:42)
zigbee2mqtt    |     at /app/node_modules/zigbee-herdsman/src/utils/queue.ts:21:18
zigbee2mqtt    |     at new Promise (<anonymous>)
zigbee2mqtt    |     at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:19:16)
zigbee2mqtt    |     at ZStackAdapter.permitJoin (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:196:26)
zigbee2mqtt    |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:226:36)
zigbee2mqtt    |     at listOnTimeout (internal/timers.js:557:17)
zigbee2mqtt    |     at processTimers (internal/timers.js:500:7)
zigbee2mqtt    | (node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 6)
zigbee2mqtt    | Zigbee2MQTT:info  2021-11-11 08:50:37: Finished Touchlink scan`

Koenkk, 2021-11-15:
FYI; new TI SDK did not fix the issue, created a bug report on the TI forum: https://e2e.ti.com/support/wireless-connectivity/zigbee-thread-group/zigbee-and-thread/f/zigbee-thread-forum/1053970/simplelink-cc13x2-26x2-sdk-znp-crashing-when-executing-interpan-commands

Azertooth, 2021-12-14:
I think that using touchlink, while trying to pair an hue ledstrip, bricked my coordinator. It does't receive zigbee message anymore, from none of my devices. Tried to delete and pairing again one of them without success and now all my lights are not working. I've tried to unplug/replug the stick (it's recognized by the hosts) and restart the z2m but still nothing, any help on how to recover from that?

juanimoni, 2021-12-20:
Hi @Koenkk, did you see the answer from TI Ryan Brown1 regarding your bug report from Nov. 15th ? 
Are you able to solve the error with this info ? 
best regards and thank you in advance for your incredible software !

Koenkk, 2021-12-21:
@juanimoni for updates check https://e2e.ti.com/support/wireless-connectivity/zigbee-thread-group/zigbee-and-thread/f/zigbee-thread-forum/1053970/simplelink-cc13x2-26x2-sdk-znp-crashing-when-executing-interpan-commands

uncle-fed, 2022-01-03:
Hello @Koenkk , there was a reply on 17 Dec suggesting a possible interim fix until new SDK is out. Could you kindly try the fix?  `nwk_setStateIdle( FALSE );`

Koenkk, 2022-01-03:
@uncle-fed thanks for pinging me, this indeed fix the issue.

New fw is available here: [`20220103`](https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin)

closed 2022-01-03 by Koenkk

davidjb, 2022-01-04:
Thanks @Koenkk, fantastic work - repeated Touchlink scans are now working! 🎉 

I have found that after doing a scan, then factory resetting a bulb (joining is permitted), the bulb fails to pair correctly (on z2m 1.22.2):

```
Zigbee2MQTT:error 2022-01-04 04:56:30: Failed to interview '0x...', device has not successfully been paired
```

But, if I then remove the unsupported device, it'll rejoin the network immediately and pair correctly. Trying this process again is the same - an Touchlink factory reset will see the bulb join the network but fail to pair correctly initially.

This isn't a worry for me as it's easy to work around, but just an FYI for anyone else.

famschouenberg, 2022-01-04:
I can confirm it is now working :-) Got my Zigstar last week and a Hue bulb to test today. Very lucky to find out the firmware for Touchlink has been fixed, since the bulb was previously connected to a Hue bridge

averter, 2024-05-26:
I'm using the SONOFF Zigbee 3.0 USB Dongle with Philips Hue light bulbs. Two bulbs are connected and working and two others that I bought second hand cannot join the network no matter what I try (probably have not been factory reset). I'm on version [1.37.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.37.1) commit: [ea39d86](https://github.com/Koenkk/zigbee2mqtt/commit/ea39d86). Below is the log for an attempt to join via one of the working bulbs (20 cm from the second hand one). 

```
info 2024-05-26 07:45:10Zigbee: allowing new devices to join via 0x0017880108b853a6.
info 2024-05-26 07:45:16Start Touchlink scan
info 2024-05-26 07:45:16Set InterPAN channel to '11'
warning 2024-05-26 07:45:17Scan request failed or was not answered: 'Error: Timeout - null - 254 - null - 4096 - 1 after 500ms'
info 2024-05-26 07:45:17Set InterPAN channel to '15'
warning 2024-05-26 07:45:17Scan request failed or was not answered: 'Error: Timeout - null - 254 - null - 4096 - 1 after 500ms'
info 2024-05-26 07:45:17Set InterPAN channel to '20'
warning 2024-05-26 07:45:18Scan request failed or was not answered: 'Error: Timeout - null - 254 - null - 4096 - 1 after 500ms'
info 2024-05-26 07:45:18Set InterPAN channel to '25'
warning 2024-05-26 07:45:18Scan request failed or was not answered: 'Error: Timeout - null - 254 - null - 4096 - 1 after 500ms'
info 2024-05-26 07:45:18Set InterPAN channel to '12'
warning 2024-05-26 07:45:19Scan request failed or was not answered: 'Error: Timeout - null - 254 - null - 4096 - 1 after 500ms'
info 2024-05-26 07:45:19Set InterPAN channel to '13'
warning 2024-05-26 07:45:19Scan request failed or was not answered: 'Error: Timeout - null - 254 - null - 4096 - 1 after 500ms'
info 2024-05-26 07:45:19Set InterPAN channel to '14'
info 2024-05-26 07:45:25Restore InterPAN channel
error 2024-05-26 07:45:31Request 'zigbee2mqtt/bridge/request/touchlink/scan' failed with error: 'SRSP - AF - interPanCtl after 6000ms'
error 2024-05-26 07:48:30Failed to keep permit join alive: Error: Cannot execute command, in Inter-PAN mode
```

SerhiiAksiutin, 2024-05-26:
@averter did you try to power off/on the light right before triggering reset. it works every time.

averter, 2024-05-26:
@SerhiiAksiutin yes I did, and it either output the above, or that touchlink is already in progress. I'll try to get the bulb close to the dongle and see if that helps, but in theory, it should be able to also pair via the other bulbs which are working (not just the coordinator) right?

averter, 2024-06-13:
I've managed to do a small progress. The two faulty bulbs are of a different kind: [philips hue LUX](https://www.pcmag.com/reviews/philips-hue-lux-starter-kit). I'm able to connect to their bridge using the [gen1 philips app](https://www.apkmirror.com/apk/philips-consumer-lifestyle/philips-hue-gen-1/philips-hue-gen-1-1-14-0-1-release/philips-hue-gen-1-1-14-0-1-android-apk-download/) and control them. However I cannot, for the life of me, find a factory reset button and touchlink on zigbee2mqtt still outputs `Scan request failed or was not answered: 'Error: Timeout - null - 254 - null - 4096 - 1 after 500ms'`. 
These lamps are [supported](https://www.zigbee2mqtt.io/devices/433714.html) and quite old, so I don't get it why touchlink does not work...

averter, 2024-07-01:
> @averter did you try to power off/on the light right before triggering reset. it works every time. Also make sure dongle is really close the bulb
> 
> cmd I used:
> 
> ```
> mosquitto_pub  -d -t zigbee2mqtt/bridge/request/touchlink/factory_reset -m ""
> mosquitto_pub  -d -t zigbee2mqtt/bridge/request/touchlink/scan -m ""
> ```

Amazing. Only today I was able to get a lamp to physically put the light bulb near the dongle, and it immediately worked. Thank you very much!