Koenkk/zigbee2mqtt #5357: Support for water endpoint for Hive SLR2 controller in utils.js please

roadsnail, 2020-12-20:
Hi,

I have been testing Hive SLR2/SLT2 support and making reasonable progress. The latest dev release does not contain an entry for 'water' in const endpointnames declaration in utils.js which threw me initially. Can this be added please? Thanks.

Thus:-

`const endpointNames = [
    'left', 'right', 'center', 'bottom_left', 'bottom_right', 'default',
    'top_left', 'top_right', 'white', 'rgb', 'cct', 'system', 'top', 'bottom', 'center_left', 'center_right',
    'ep1', 'ep2', 'row_1', 'row_2', 'row_3', 'row_4', 'relay',
    'l1', 'l2', 'l3', 'l4', 'l5', 'l6', 'l7', 'l8',
    'button_1', 'button_2', 'button_3', 'button_4', 'button_5',
    'button_6', 'button_7', 'button_8', 'button_9', 'button_10',
    'button_11', 'button_12', 'button_13', 'button_14', 'button_15',
    'button_16', 'button_17', 'button_18', 'button_19', 'button_20',
    'button_light', 'button_fan_high', 'button_fan_med', 'button_fan_low',
    'heat', 'cool',
];`

becomes:

`const endpointNames = [
    'left', 'right', 'center', 'bottom_left', 'bottom_right', 'default',
    'top_left', 'top_right', 'white', 'rgb', 'cct', 'system', 'top', 'bottom', 'center_left', 'center_right',
    'ep1', 'ep2', 'row_1', 'row_2', 'row_3', 'row_4', 'relay',
    'l1', 'l2', 'l3', 'l4', 'l5', 'l6', 'l7', 'l8',
    'button_1', 'button_2', 'button_3', 'button_4', 'button_5',
    'button_6', 'button_7', 'button_8', 'button_9', 'button_10',
    'button_11', 'button_12', 'button_13', 'button_14', 'button_15',
    'button_16', 'button_17', 'button_18', 'button_19', 'button_20',
    'button_light', 'button_fan_high', 'button_fan_med', 'button_fan_low',
    'heat', 'cool', 'water',
];`

Koenkk, 2020-12-20:
Added!

closed 2020-12-20 by Koenkk

robertalexa, 2021-08-02:
Hey @roadsnail 

I was reading this thread and noticed that you own a SLT2 remote control? I am currently trying to add battery support for all the Hive remotes but i would need some help with people testing before i call it done.

If you still have it, and you are willing to offer me 5 minutes of your time, i can give you some steps of what to do, and if the battery percent starts showing up then I will submit a PR. Thank you

roadsnail, 2021-08-03:
Hi, yes, I have the SLT Hive controller. I am busy at the moment, but let me know what needs doing and if not too involved and I find the time, then I'll give it a go... 

Cheers
Chris

robertalexa, 2021-08-03:
Hi @roadsnail Chris,

Thanks for coming back to me on this one. In the meantime, another user join my thread and offered to help with the SLT2. You are more than welcome to have a look too if you want :)

https://github.com/Koenkk/zigbee2mqtt/issues/8184

Rob