Koenkk/zigbee2mqtt #5359: Moes relay marked as unsupported.

nickynickukrnet, 2020-12-20:
Faced with problem when I tried to pair the device - got message:
"device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_zmy1waw6' is NOT supported" , 
But actualy this model is present in 'supported' list/devices.js

log file is attached.
[moes_.txt](https://github.com/Koenkk/zigbee2mqtt/files/5721387/moes_.txt)

## Debug info
Zigbee2MQTT version: 1.16.2.2
Adapter hardware: CC2531

Koenkk, 2020-12-20:
Can you provide a link to the device?

nickynickukrnet, 2020-12-20:
do You mean device model?

ali     ----------------------------------
https://aliexpress.ru/item/4001065019839.html?spm=2114.13010708.0.0.587533edoHBCBl&_ga=2.26379965.188328276.1608320421-1383686913.1608140534

in supported dev    ----------------------------------
https://www.zigbee2mqtt.io/devices/TS0121_switch.html

in devices.js     ----------------------------------
    // Moes
    {
        fingerprint: [
            {modelID: 'TS0121', manufacturerName: '_TYZB01_iuepbmpv'},
            {modelID: 'TS011F', manufacturerName: '_TZ3000_zmy1waw6'},
        ],
        model: 'MS-104Z',
        description: 'Smart light switch module (1 gang)',
        vendor: 'Moes',
        extend: generic.switch,
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            try {
                // Fails for some devices.
                // https://github.com/Koenkk/zigbee2mqtt/issues/4598
                await configureReporting.onOff(endpoint);
            } catch (e) {
                e;
            }
        },
    },

iac304es, 2020-12-21:
I have the same problem, it detects the device but marks it as not supported, how can I make them work is that I have 5 devices.

Koenkk, 2020-12-21:
Are you guys on the dev branch? (might be that it has been added support for very recent).  (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

nickynickukrnet, 2020-12-21:
Several peolpe I know  pair their relays without any problem.
One gay have the same problem - device seen as unsupported, he delete them from HA zigbee2mqttassistant device list and then device paired successfully.
But they have CC2538 Zigbee V3.0
They use same software that I have.

jvsystelab, 2020-12-28:
Hi everyone,

I have the same problem with this "supported" device.
In my case, I'm using a zzh adapter (CC2652R).

This is the log:
Successfully interviewed 'xxxx', device has successfully been paired
Device 'xxxx' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_zmy1waw6' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Is there any method to add this device? I have the installation pending to an answer.

Thanks.

mgg1010, 2020-12-28:
Doing exactly the same right now.  Same problem as everyone else.  I've tried "latest-dev" as a tag
on the image, restarted the container and ... same problem - unsupported device.

I see that in my container, the devices-js is dated Oct 26th, but the one from https://github.com/Koenkk/zigbee-herdsman-converters.git is Dec 28th.  Odd because the docker container was rebuilt 17 hours ago...

Koenkk, 2020-12-28:
`'TS011F' and manufacturer name '_TZ3000_zmy1waw6' ` is supported in the latest dev branch. @jvsystelab updating to the latest dev branch should make it supported.

@mgg1010 do you have devices.js overridden via a docker volume? (e.g. when adding support for a new device in the past).

mgg1010, 2020-12-28:
Hi Koenkk.  I don't.  My docker compose file is below.  I set this up in 30 minutes last night and it worked - nothing
special.  I used docker-compose pull / restart to restart the container.

  version: '3'
  services:
    zigbee2mqtt:
      container_name: zigbee2mqtt
      image: koenkk/zigbee2mqtt:latest-dev
      volumes:
        - ./data:/app/data
        - /run/udev:/run/udev:ro
      devices:
        - /dev/ttyUSB0:/dev/ttyACM0
      restart: always
      network_mode: host
      privileged: true
      environment:
        - TZ=Europe/London

jvsystelab, 2020-12-28:
Ok @Koenkk 

I can confirm, the dev branch detects the device and it works fine.

Thank you.

mgg1010, 2020-12-28:
I'm not an expert at this. Docker PS seems to show I'm still using the old one
Restarting fully

mgg1010, 2020-12-28:
OK - that fixed it - my fault, didn't update the image properly.

Thanks very much for this !!

Koenkk, 2020-12-28:
Since this device is supported in the dev branch I will close this.

closed 2020-12-28 by Koenkk