Koenkk/zigbee2mqtt #5362: Sonoff ZBmini; power source showing question mark

kireque, 2020-12-21:
## What happened
After adding multiple Sonoff zbmini devices I noticed that the power source is showing a question mark.

## What did you expect to happen
Power source showing a plug instead question mark

## How to reproduce it (minimal and precise)
Add Sonoff ZBmini

## Debug info
Zigbee2MQTT version: 1.16.2
Adapter hardware: ZZH
Adapter firmware version: 
{'meta': {'maintrel': 1, 'majorrel': 2, 'minorrel': 7, 'product': 1, 'revision': 20200925, 'transportrev': 2}, 'type': 'zStack3x0'}

Koenkk, 2020-12-21:
Can you provide the data/database.db entries of these devices?

kireque, 2020-12-22:
```
{
    "id": 42,
    "type": "Router",
    "ieeeAddr": "0x842e14fffe9e1296",
    "nwkAddr": 48943,
    "manufId": 4098,
    "manufName": "SONOFF",
    "powerSource": "Unknown",
    "modelId": "01MINIZB",
    "epList": [
        1,
        242
    ],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 256,
            "inClusterList": [
                0,
                3,
                4,
                5,
                6,
                4096
            ],
            "outClusterList": [
                4096
            ],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "modelId": "01MINIZB",
                        "manufacturerName": "SONOFF",
                        "powerSource": 0,
                        "zclVersion": 3,
                        "dateCode": "20200927"
                    }
                },
                "genOnOff": {
                    "attributes": {
                        "onOff": 0
                    }
                }
            },
            "binds": [
                {
                    "cluster": 6,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x00124b0021600460",
                    "endpointID": 1
                }
            ],
            "meta": {}
        },
        "242": {
            "profId": 41440,
            "epId": 242,
            "devId": 97,
            "inClusterList": [],
            "outClusterList": [
                33
            ],
            "clusters": {},
            "binds": [],
            "meta": {}
        }
    },
    "dateCode": "20200927",
    "zclVersion": 3,
    "interviewCompleted": true,
    "meta": {
        "reporting": 1
    },
    "lastSeen": 1608647836955
}
{
    "id": 43,
    "type": "Router",
    "ieeeAddr": "0x842e14fffe9e13c8",
    "nwkAddr": 31740,
    "manufId": 4098,
    "manufName": "SONOFF",
    "powerSource": "Unknown",
    "modelId": "01MINIZB",
    "epList": [
        1,
        242
    ],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 256,
            "inClusterList": [
                0,
                3,
                4,
                5,
                6,
                4096
            ],
            "outClusterList": [
                4096
            ],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "modelId": "01MINIZB",
                        "manufacturerName": "SONOFF",
                        "powerSource": 0,
                        "zclVersion": 3,
                        "dateCode": "20200927"
                    }
                },
                "genOnOff": {
                    "attributes": {
                        "onOff": 0
                    }
                }
            },
            "binds": [
                {
                    "cluster": 6,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x00124b0021600460",
                    "endpointID": 1
                }
            ],
            "meta": {}
        },
        "242": {
            "profId": 41440,
            "epId": 242,
            "devId": 97,
            "inClusterList": [],
            "outClusterList": [
                33
            ],
            "clusters": {},
            "binds": [],
            "meta": {}
        }
    },
    "dateCode": "20200927",
    "zclVersion": 3,
    "interviewCompleted": true,
    "meta": {
        "reporting": 1
    },
    "lastSeen": 1608647809780
}
```

Koenkk, 2020-12-22:
Fixed.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-22 by Koenkk

mojijafari, 2021-01-16:
> ```
> {
>     "id": 42,
>     "type": "Router",
>     "ieeeAddr": "0x842e14fffe9e1296",
>     "nwkAddr": 48943,
>     "manufId": 4098,
>     "manufName": "SONOFF",
>     "powerSource": "Unknown",
>     "modelId": [...]

Thanks, how did you find this data? 
could you please help me?

kireque, 2021-01-18:
> Thanks, how did you find this data?
> could you please help me?

As Koenkk requested:
Can you provide the **data/database.db** entries of these devices?

In the .db file it is one rule. I only formatted is as json so it looks nice.