Koenkk/zigbee2mqtt #5366: Smartthings

biddster, 2020-12-21:
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

Koenkk, 2020-12-31:
Can you provide the herdsman debug log of the interview?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

biddster, 2021-01-02:
Log file attached. Here's the exception at the end of the interview process:

```
zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x000d6f000575f9bd', attempt 6 +23s
  zigbee-herdsman:controller:device:log Interview - failed for device '0x000d6f000575f9bd' with error 'Error: Interview failed because can not get node descriptor ('0x000d6f000575f9bd')
  zigbee-herdsman:controller:device:log     at Device.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:350:23)
  zigbee-herdsman:controller:device:log     at Generator.throw (<anonymous>)
  zigbee-herdsman:controller:device:log     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:25:65)' +2ms
  zigbee-herdsman:controller:error Interview failed for '0x000d6f000575f9bd with error 'Error: Interview failed because can not get node descriptor ('0x000d6f000575f9bd')' +0ms
Zigbee2MQTT:error 2021-01-02 16:10:05: Failed to interview '0x000d6f000575f9bd', device has not successfully been paired
```

[zigbee2mqtt.log](https://github.com/Koenkk/zigbee2mqtt/files/5760502/zigbee2mqtt.log)

Koenkk, 2021-01-02:
Looks the device does not communicate at all, check https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails and try replacing the battery.

biddster, 2021-01-02:
Thanks for the reply.

I've tried brand new batteries and followed the advice in the link before raising the bug. I'll persist with repeatedly trying to pair them. 

Maybe there's something else I have to do to them before putting them in pairing mode.

closed 2021-02-10 by github-actions[bot]

biddster, 2021-02-21:
In case anyone finds themselves landing here, there are now working. 

There have been 2 changes since I raised the bug:

1. Several releases of zigbee2mqtt
2. I changed zigbee channel to 20 from the defaul

I don't know which of the changes fixed it, but these motion sensors pair immediately now and are very lucid in operation.

famewolf, 2026-09-12:
> In case anyone finds themselves landing here, these are now working.
> 
> There have been 2 changes since I raised the bug:
> 
>     1. Several releases of zigbee2mqtt (I'm now on 1.17.1)
> 
>     2. I changed zigbee channel to 20 from the default
> 
> 
> I don't know which of the changes fixed it, but these motion sensors pair immediately now and are very lucid in operation.

Did you ever nail down if it was the channel change?  I'm on latest z2m with one of those smartthing motions sensors.  Get the flashing blue light for pairing but see no activity in z2m from it.  I'm currently using channel 25 and was hoping you might have updated info since I'd have to re-pair all the existing devices with the new channel (or at least I did the last time I changed the channel).