Koenkk/zigbee2mqtt #5367: Neo NAS-AB02B0 - unsupported device  '_TZE200_d0yu2xgi'

vencakratky, 2020-12-21:
## What happened
Hello, I bought Zigbee Alarm NAS-AB02B0, exactly this type:
https://www.aliexpress.com/item/4000813903996.html?spm=a2g0s.9042311.0.0.347b4c4dj4R0jl 

## What did you expect to happen
Cannot pair with CC2652 and Zigbee2mqtt ver. 1.16.2.2 (the same problem with Zigbee2MQTT Edge)
## How to reproduce it (minimal and precise)
I tried 10x pairing near the coordinator and without successs. 
debug log:
`debug 2020-12-21 22:48:51: Received Zigbee message from '0x842e14fffedb8370', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,1,24],"type":"Buffer"},"datatype":2,"dp":105,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
warn  2020-12-21 22:48:51: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_d0yu2xgi'
warn  2020-12-21 22:48:51: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-12-21 22:48:51: Received Zigbee message from '0x842e14fffedb8370', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,1,25],"type":"Buffer"},"datatype":2,"dp":105,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
warn  2020-12-21 22:48:51: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_d0yu2xgi'
warn  2020-12-21 22:48:51: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-12-21 22:48:51: Received Zigbee message from '0x842e14fffedb8370', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,42],"type":"Buffer"},"datatype":2,"dp":106,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
warn  2020-12-21 22:48:51: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_d0yu2xgi'
warn  2020-12-21 22:48:51: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-12-21 22:48:52: Received Zigbee message from '0x842e14fffedb8370', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,1,26],"type":"Buffer"},"datatype":2,"dp":105,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
warn  2020-12-21 22:48:52: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_d0yu2xgi'
warn  2020-12-21 22:48:52: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-12-21 22:48:52: Received Zigbee message from '0x842e14fffedb8370', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,43],"type":"Buffer"},"datatype":2,"dp":106,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
warn  2020-12-21 22:48:52: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_d0yu2xgi'
warn  2020-12-21 22:48:52: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-12-21 22:48:52: Received Zigbee message from '0x842e14fffedb8370', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,42],"type":"Buffer"},"datatype":2,"dp":106,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
warn  2020-12-21 22:48:52: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_d0yu2xgi'
warn  2020-12-21 22:48:52: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-12-21 22:48:57: Received Zigbee message from '0x842e14fffedb8370', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,1,27],"type":"Buffer"},"datatype":2,"dp":105,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
warn  2020-12-21 22:48:57: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_d0yu2xgi'
warn  2020-12-21 22:48:57: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.`

## Debug info
Zigbee2MQTT version: 1.16.2.2.
Adapter hardware: CC2652 
Adapter firmware version: latest 

Thank you verry much :)

Koenkk, 2020-12-22:
Already supported in the latest dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-22 by Koenkk

877dev, 2021-02-05:
I have this same issue, thanks for adding to dev branch. I guess it will make it's way to release at some point..

Koenkk, 2021-02-05:
@877dev is in z2m 1.17.1

877dev, 2021-02-05:
@Koenkk i made a mistake, my device is different. 

I tried adding a unsupported device but got stuck. 

I raised an issue if you have any spare time to advise. 

Thanks

leroyloren, 2021-04-07:
Hi,

I have this siren, when pairing, it immediately pairs properly, but it cannot be controlled and sends only link quality, other options (and temperatures) are inactive.

[image]

[image]

zbuh, 2021-04-09:
I have exactly the same issue, only link quality is published!

zbuh, 2021-04-09:
@leroyloren Have you solved this? if not, we need to open a new issue.

leroyloren, 2021-04-09:
@zbuh 

did not solve, I only found that there may be an error in the new HW revision identification.

Probably the manufacturer changed and the integration for the Z2M is dependent on the previous one.

[https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4617](url)

zbuh, 2021-04-09:
`Zigbee2MQTT:debug 2021-04-09 21:40:51: Received Zigbee message from '0x804b50fffe048344', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,10,17,0,27,68],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-04-09 21:40:51: No converter available for 'NAS-AB02B0' with cluster 'manuSpecificTuya' and type 'raw' and data '{"data":[9,10,17,0,27,68],"type":"Buffer"}'`

leroyloren, 2021-04-10:
https://github.com/Koenkk/zigbee-herdsman-converters/issues/1932