Koenkk/zigbee2mqtt #5368: innr OSL130C not working (No converter available)

joermke, 2020-12-21:
# Bug Report
## What happened
- innr OSL130C paired (which introduced 4 devices for some reason)
- Pulled latest-dev
- Started zigbee2mqtt, no issues
- Saw linkquality messages, so I assume that the pairing is still good
- Other zigbee devices (Osram SmartPlugs, Philips Hue) still working nicely
- Sending `zigbee2mqtt/OSL130C/set` with payload `{"state": "OFF"}`
- Received `No converter available for 'state' (OFF)`

## What did you expect to happen
I expect the OSL130C to turn off

## How to reproduce it (minimal and precise)
See above, always reproducible

## Debug Info
Zigbee2MQTT version: 1.16.2-dev (commit #8e87cbd), zigbee-herdsman (0.13.41)
Adapter hardware: CC2531
Adapter firmware version: 3.2.6, rev. 20190608

Koenkk, 2020-12-22:
Can you provide the debug logging of this?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

joermke, 2020-12-22:
Sure! This is what shows up if I physically turn on the innr spots. As I said, after pairing, I have 4 devices related to the innr. Thus I labeled the one broadcasting the link quality OSL130C and added _a, _b, _c to the other ones. As You can see, OSL130C_b announced itself, too. However, the OSL130C still reports the link quality.

`debug 2020-12-22 23:00:08: Device 'OSL130C_b' announced itself
info  2020-12-22 23:00:08: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"OSL130C_b","ieee_address":"0x00158d0002a313d4"},"type":"device_announce"}'
info  2020-12-22 23:00:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"OSL130C_b"},"type":"device_announced"}'
debug 2020-12-22 23:00:33: Received MQTT message on 'zigbee2mqtt/fhem/state' with data 'online'
debug 2020-12-22 23:00:41: Received Zigbee message from 'OSL130C', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
info  2020-12-22 23:00:41: MQTT publish: topic 'zigbee2mqtt/OSL130C', payload '{"led":false,"linkquality":5}'`

Now trying to set the state of eiter one yields:

`debug 2020-12-22 23:06:08: Received MQTT message on 'zigbee2mqtt/OSL130C/set' with data '{"state":"OFF"}'
error 2020-12-22 23:06:08: No converter available for 'state' (OFF)`

and, respectively,

`info  2020-12-22 23:06:41: MQTT publish: topic 'zigbee2mqtt/OSL130C', payload '{"led":false,"linkquality":47}'
debug 2020-12-22 23:06:50: Received MQTT message on 'zigbee2mqtt/OSL130C_b/set' with data '{"state":"OFF"}'
debug 2020-12-22 23:06:50: Publishing 'set' 'state' to 'OSL130C_b'
debug 2020-12-22 23:07:03: Received MQTT message on 'zigbee2mqtt/fhem/state' with data 'online'
error 2020-12-22 23:07:10: Publish 'set' 'state' to 'OSL130C_b' failed: 'Error: Command 0x00158d0002a313d4/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'No network route' (205))'
debug 2020-12-22 23:07:10: Error: Command 0x00158d0002a313d4/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with 
error: 'No network route' (205))
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:339:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:24:58)
info  2020-12-22 23:07:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'OSL130C_b' failed: 'Error: Command 0x00158d0002a313d4/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"OSL130C_b"},"type":"zigbee_publish_error"}'`

Note: Devices _a, _b and _c behave identically. So why do I see 4 devices after pairing in the first place? The spots cannot be controlled individually.

Thanks for having a look at this!

Koenkk, 2020-12-23:
> So why do I see 4 devices after pairing in the first place?

Also wondering this, can you provide your complete data/databse.db?

joermke, 2020-12-23:
Here is the database:
[database.db](https://github.com/Koenkk/zigbee2mqtt/files/5737011/database.txt)

And these are the devices in the configuration.yaml:
`  '0x00178801060ca70c':
    friendly_name: 'OSL130_c'
  '0x0017880104aaed30':
    friendly_name: 'OSL130C_a'
  '0x00158d0002a313d4':
    friendly_name: 'OSL130C_b'
  '0x00124b0014d9ab18':
    friendly_name: 'OSL130C'
`

Koenkk, 2020-12-24:
I think you are confused, these 4 devices are not the same device:
- `0x00178801060ca70c` (friendly_name: OSL130_c) is a philips hue go
- `0x0017880104aaed30` (friendly_name: OSL130C_a) is a philips hue white ambiance bulb
- `0x00158d0002a313d4` (friendly_nmae: OSL130C_b) this is the only Innr OSL130C
- `0x00124b0014d9ab18` (friendly_name: OSL130C) this is a CC2530 router, so not strange you cannot set the state of it.

joermke, 2020-12-26:
Yes, indeed...
Thanks for clarifying this. I did not think, however, that the two Philips devices can be paired to the zigbee2mqtt since they are up and running with a Hue gateway.
So I will focus on the true OSL130C and check the logs again.

joermke, 2020-12-26:
Thanks for narrowing this down!
With Your help I was able to find the true source of the issue. I placed the control unit in the shelter for my electric sheep. Even though it is pretty close to the router it seems to shield just enough to allow for the device to register but not to accept any commands.
Thus, I will close this non-issue. Thanks for the support, I really love this project!

closed 2020-12-26 by joermke